MelbourneDeveloper/Device.Net #19: USB - Interfaces Other Than Default

MelbourneDeveloper, 2019-01-04:
Any given USB device can have multiple different interfaces, and pipes within those interfaces. There should be a way to select which interface is used before InitializeAsync is called.

schotime, 2019-01-16:
I'm assuming that will fix being able to connect like: usbDevice.OpenEndpointReader(ReadEndpointID.Ep02);
?

MelbourneDeveloper, 2019-01-16:
@schotime , I believe the code you have posted is from LibUsbDotNet. I've had problems with that library, but yes, the plan is to allow for a similar pattern to be able to select a different interface.

There is some work involved because the way to do this is different on each platform. If this is causing you grief, I could perhaps fast track this issue. Are you currently trying to use Usb.Net and hitting this problem?

schotime, 2019-01-16:
No probs. Cheers

MelbourneDeveloper, 2019-01-16:
@schotime , are you part of an indoor golf simulator business? Do you have a website?

schotime, 2019-01-16:
na. integrating with Foresight GC2 launch monitor
[image]

MelbourneDeveloper, 2019-01-16:
Cool!

MelbourneDeveloper, 2019-01-17:
@schotime so this is how the code is looking. I will put out an alpha release soon methinks...

```cs
            var devices = await DeviceManager.Current.GetDevices(_DeviceDefinitions);
            TrezorDevice = devices.FirstOrDefault();
            await TrezorDevice.InitializeAsync();

            var windowsDevice = (WindowsUsbDevice)TrezorDevice;
            var firstInterface = windowsDevice.UsbInterfaces[0];
            var writePipe = firstInterface.UsbInterfacePipes.FirstOrDefault(p => p.IsWrite);
            var readPipe = firstInterface.UsbInterfacePipes.FirstOrDefault(p => p.IsRead);
            firstInterface.ReadPipe = readPipe;
            firstInterface.WritePipe = writePipe;
            windowsDevice.WriteUsbInterface = firstInterface;
            windowsDevice.ReadUsbInterface = firstInterface;
```

MelbourneDeveloper, 2019-01-17:
Look right?

MelbourneDeveloper, 2019-01-17:
Would you use the word Pipe? Or Endpoint?

MelbourneDeveloper, 2019-01-17:
Actually, I think Endpoint sounds better.

MelbourneDeveloper, 2019-01-17:
@schotime, the NuGet is here and should be listed soon. It's alpha so you have to include prerelease:

https://www.nuget.org/packages/Usb.Net/2.3.1-alpha

The code is as above. You have to first get the device, and then call InitializeAsync. This _should_ get all the available interfaces. The will be in the UsbInterfaces collection. However, it won't give you a textual name of the interfaces. It will just set up the handle to the interface for writing. You can set the read/write endpoints as well, but that probably won't be necessary.

If you have any trouble, I strongly recommend cloning the repo, turning on break on all errors and send me any information about what goes wrong.

MelbourneDeveloper, 2019-01-17:
This is the code branch where the fix has been made:

https://github.com/MelbourneDeveloper/Device.Net/tree/%2319

MelbourneDeveloper, 2019-02-19:
@schotime , this issue is going to go back on the low priority list. Let me know if it's still important.

schotime, 2019-02-19:
Oh mate. Apologies. I've been swamped and must have missed your update. Appreciate the effort to get something for me. 

Low priority is fine. It's probably what mine is to swap libraries at the moment as it is currently working with libusb.

I'll keep you updated. Sorry again.

MelbourneDeveloper, 2019-02-19:
@schotime no problem. There is now available wrapper for libusb. If you use it the code will be compatible with other platforms and you will get device listening. It's been tested on windows and Mac. 

Sample
https://github.com/MelbourneDeveloper/Device.Net/tree/master/src/Device.Net.LibUsbSample

Wrapper
https://github.com/MelbourneDeveloper/Device.Net/blob/master/src/Device.Net.LibUsb/LibUsbDevice.cs

mxjones, 2019-07-01:
Hi @MelbourneDeveloper! I also have a similiar requirement that I think this ticket resolves (access to all endpoints for an interface - my specific sceneario has 2 read endpoints)

I see the help wanted tag on this & I took a quick look at the changes you made on that branch linked previously (adding this for Windows USB) - what else do you think is needed here before it is ready? I can try and take a look at this. Would you want to add something to the generic interface to allow this (without casting) or do you think it might be to hard to do this for all the different targets? (Android/UWP etc) 

Cheers!

MelbourneDeveloper, 2019-07-01:
@mxjones it's been a while since I looked at this. I think the direction I was going in the branch is roughly what I wanted to do. the problem was it didn't end up being a clean solution. 

I'm happy to look at a pull request if you want to continue with it. But, seeing that it is an issue for multiple people, I will make sure I get back to this one soon.

MelbourneDeveloper, 2019-07-01:
I labelled it high priority.

mxjones, 2019-07-02:
Thanks @MelbourneDeveloper - I am happy to take a look into this a little bit and work towards a PR or at least get to know the codebase a bit more (no obligation to merge, could just help towards the final implementation/float some ideas)

Just to get a bit more information, I am personally keen to get this in for both Windows & Android - Would you want this done in a similiar way as the branch was progressing (done per specific target) or something that is exposed within the interface (IDevice)?

Cheers :)

MelbourneDeveloper, 2019-07-02:
@mxjones awesome!

I'll help you along the way. I've dusted of the 19 branch and merged develop in to it. I've roughly sketched out the design for IUsbDevice and its related interfaces. This design should forward to Android when we get to this point. 

Any help would be appreciated. I recommend small incremental PRs without going too far down any one path. I think you can see where I'm headed with the current code in 19.

MelbourneDeveloper, 2019-07-02:
@mxjones I've gone over this again. I think the design is pretty close now. It's mostly set up for DI so that it can be fully unit testable. The design should transfer right over to Android when we get to it. Feel free to give me some feedback on the design. It certainly will change, but I don't think it will change too much.

MelbourneDeveloper, 2019-07-02:
@mxjones I tested the new code with a Usb device on windows and it actually works, so I think it's pretty close.

MelbourneDeveloper, 2019-07-02:
@schotime still interested in this?

https://github.com/MelbourneDeveloper/Device.Net/tree/%2319

schotime, 2019-07-03:
I am, but how would i specify a particular read endpoint like `ReadEndpointID.Ep02`?

MelbourneDeveloper, 2019-07-03:
@schotime after you've initialized the device, there will be a list of interfaces. You can set the read and write interfaces.

MelbourneDeveloper, 2019-07-03:
@schotime  with this setter: https://github.com/MelbourneDeveloper/Device.Net/blob/a9316b415de9d5c570fa1c9f941a0328ab6c7991/src/Usb.Net/Windows/WindowsUsbDevice.cs#L32

MelbourneDeveloper, 2019-07-03:
@schotime @mxjones actually there is more work to be done. I should be able to look at this soon. @schotime is right. These changes don't yet let the endpoint to be set.

MelbourneDeveloper, 2019-07-03:
@schotime it's good to go on Windows.

This is one way:
```cs
windowsUsbDevice.WriteUsbInterface.ReadEndpoint = windowsUsbDevice.UsbInterfaces[0].UsbInterfaceEndpoints[0];
```

or

```cs
windowsUsbDevice.WriteUsbInterface.ReadEndpoint = windowsUsbDevice.WriteUsbInterface.ReadEndpoint;
```

mxjones, 2019-07-03:
@MelbourneDeveloper Wow, that was quick! I will spend some time this evening looking over the changes, had a glance at them now and design looks good to me, will pull it down and try to give it a test for my use case here

mxjones, 2019-07-03:
I had a little play with this, the ability to choose the read pipe is working nicely for me (I am also hitting up against #67 but I think that can just be solved by manually keep reading for now eg. by default it is reading 64 bytes but I'm getting more, I can probably work out how many I need to read and keep reading that way. In my test, I just added a setter to the ReadBuffer and set it to 5000 and that is working OK currently)

Thanks for your work on this! I will carry on looking and playing with this (I am mostly just playing around / hobbying / prototyping at the moment) - I only have some very basic comms done at the moment in the test so looking forward to getting more in and testing that out aswell.

mxjones, 2019-07-03:
Got a bit further, was also looking at being able to read from an InterruptEndpoint...can see some initial work in the fork: https://github.com/mxjones/Device.Net/tree/%2319

Was able to see this working but ran out of time, will continue playing later or tomorrow. I think this is out of scope for this issue though! :D

MelbourneDeveloper, 2019-07-03:
@mxjones anything you can see that is wrong that would stop me from releasing these changes?

MelbourneDeveloper, 2019-07-03:
@mxjones this needs more work. I started looking at an implementation for Android but I realized the design is not right.

Feel free to offer thoughts, but I'm going to go down a different path in #19-Android

mxjones, 2019-07-03:
To answer the first question, I couldn't see anything that was wrong regarding the technical aspect (it worked well being able to choose a different endpoint) that would stop you from releasing.

Happy to see which direction you are taking the design and perform testing aswell and provide any help where possible - it is quite a difficult thing to get in cleanly so was expecting a couple of design iterations! :D I will fork the new brand tomorrow and adapt my test code and carry on testing - I still have more 'advanced things to implement on my side (testing) - I will also look at trying to get something in so I can test the android component now that you have already made some headwaybl there. 

I will take a look tomorrow, thanks for this!

MelbourneDeveloper, 2019-07-03:
@mxjones thanks! I'm pretty going to sit on this for a while. 

I want to make sure I get the design right. The design issue is the inheritance hierarchy. WindowsUsbDevice inherits from WindowsDeviceBase. But, it doesn't inherit from a hypothetical UsbDeviceBase so it can't share any code between the AndroidUsbDevice and WindowsUsbDevice. To make that happen, C# would need multiple inheritance which would be a bad design anyway.

So, I'm considering a redesign where all of the platform specific stuff is dependency injected in. I will let you know how the progress goes.

MelbourneDeveloper, 2019-07-03:
@mxjones have a look at the #19-Android branch now. I've completely separated the platform specific implementation from the usb interface handling. This means that the handling can be reused on Android. 

Thoughts are welcome...

mxjones, 2019-07-04:
@mxjones didn't get as much time to look at this as I planned - but I think it looks really good. Design does feel nicer (I personally like composition over inheritance most of the time anyway)

I did make some progress with getting more advanced stuff in, and had the interrupt endpoint working (+ timeout) - shall I create a new issue for adding the interrupt endpoint? or is that something you were not thinking of adding?

I plan on spending some time getting an android test up and running on the weekend, so will have a chance then to test out the new android code you committed.

MelbourneDeveloper, 2019-07-04:
@mxjones yes please create a new issue with as much background info as possible. I don't know what an interupt endpoint is so you'll have to explain it. 

I won't be able to look at this until the new design is done. I can only focus on one thing at a time. 

> I think it looks really good. Design does feel nicer (I personally like composition over inheritance most of the time anyway)

Awesome. I think this style will eventually get rolled out to all the devices . It will solve a lot of the code sharing issues that exist.

mxjones, 2019-07-04:
Sure will do, and no problem

MelbourneDeveloper, 2019-07-06:
@mxjones I've gone further. I think the general design is pretty close now. I've tested Windows but not Android. If you feel like testing Android, that would be great. I will hopefully finish up tomorrow.

mxjones, 2019-07-06:
Hi @MelbourneDeveloper - I created a super simple Android project for testing your branch and with a couple of minor changes all of the basics are working great (Connecting, Writing some bytes, Reading response, Disconnecting) - **awesome!**

Couple of minor things noted when testing Android - they are minor things that just may need slight refactoring after the design changes.

- I needed to move opening the Android device higher up in AndroidUsbDeviceHandler, because it is used in the block above it (passing to endpoint ctor & claiming interface)

https://github.com/MelbourneDeveloper/Device.Net/blob/9ca01d88dada7cc574bcd9a291c59ec9b6a3ef8f/src/Usb.Net.Android/AndroidUsbDeviceHandler.cs#L197-L202

to

https://github.com/MelbourneDeveloper/Device.Net/blob/9ca01d88dada7cc574bcd9a291c59ec9b6a3ef8f/src/Usb.Net.Android/AndroidUsbDeviceHandler.cs#L145

- The default way of finding the endpoints didn't work for me, I suspect it is because it is looking for interrupt endpoints:

https://github.com/MelbourneDeveloper/Device.Net/blob/9ca01d88dada7cc574bcd9a291c59ec9b6a3ef8f/src/Usb.Net.Android/AndroidUsbDeviceHandler.cs#L161-L162

On my local, I just changed this to look for the bulk endpoint type and then read/write depending on direction In/Out

https://github.com/mxjones/Device.Net/blob/a896060ebb360ec57a101ae899dd12be594c5297/src/Usb.Net.Android/AndroidUsbDeviceHandler.cs#L167-L168

Another related minor tweak is that it is checking if the endpoint is null - this is always false because the getter will set it to the first compatible endpoint if it is null - I just changed this to check the associated interface instead. (The end result of this was that the Interfaces never get assigned, by default - but manually assigning these works OK)

https://github.com/mxjones/Device.Net/blob/a896060ebb360ec57a101ae899dd12be594c5297/src/Usb.Net.Android/AndroidUsbDeviceHandler.cs#L173-L183

- The DeviceListener wouldn't find my Device correctly, I'm not sure why - didn't look into it too much as I haven't tried it in develop branch yet & probably unrelated to these changes - I just handled the finding/permission/receiver directly

No notes when testing Windows, require no changes & minimum test app is working (Connecting, Writing some bytes, Reading response, Disconnecting)

Sorry for the long post - just a few notes whilst I was testing that may prove useful. This is great work though, good job so far!

MelbourneDeveloper, 2019-07-06:
@mxjones all good points here. I will digest these slowly.

Question: IntializeAsync is needed to get the interfaces. But, then, we may need to "claim" the interface. Currently, this is done in one step on Android, but that doesn't make sense. We should enumerate the interfaces and "claim" only one when we have the list. So, I can't see how initialization can be done without two steps... This creates problems for the rest of the design of the system...

Any thoughts?

MelbourneDeveloper, 2019-07-07:
@mxjones I did a PR on your behalf and I've merged your code in to this branch: https://github.com/MelbourneDeveloper/Device.Net/tree/%2377-AndroidAndInterrupt . I hope you don't mind. This branch is up to date with the latest from develop. Lots of refactoring.

 I think what you've done with endpoint defaulting there is much better. I only copied that from some library so there's a good chance the original code only worked with the Trezor.

I've added a stub here for interrupts, so feel free to flesh this out so I understand what it is about.
#77

MelbourneDeveloper, 2019-07-07:
@mxjones I will leave the interrupt and timeout stuff for 3.1. It's definitely on the right track, but I've gotta fix the UWP and LibUsb versions which is going to take more time.

mxjones, 2019-07-08:
@MelbourneDeveloper Thanks for this! Theres a bit of cleaning up to do on that no doubt, so will see what is needed (eg. I made changes to the Sample project for my local testing, I will revert those etc)

> 
> 
> @mxjones all good points here. I will digest these slowly.
> 
> Question: IntializeAsync is needed to get the interfaces. But, then, we may need to "claim" the interface. Currently, this is done in one step on Android, but that doesn't make sense. We should enumerate the interfaces and "claim" only one when we have the list. So, I can't see how initialization can be done without two steps... This creates problems for the rest of the design of the system...
> 
> Any thoughts?
> 
> Maybe the intended interface number can be specified in the constructor, and if it is specified, it will try to claim that one?

I see, for default initialisation perhaps it only grabs the first interface and claims that. I think passing the interface number in the ctor is a good option, to override the default behaviour (eg. It will default to 0, but can be set in ctor if a different interface is needed). 
Could also move claiming to the UsbInterface instead, adding a 'Claim' method so if someone is setting it up/choosing the interface manually themselves then they can issue the call as needed. 

So by default, it would claim Interface[0] - the one that is sets as the default interface in for the Device - but with your added work exposing multiple interfaces, someone could find the one they want and then just call UsbInterface.Claim() (not 100% sure on this - it could just be a call on the device - Claim(int interfaceId) ) if needed. 

> 
> 
> @mxjones I will leave the interrupt and timeout stuff for 3.1. It's definitely on the right track, but I've gotta fix the UWP and LibUsb versions which is going to take more time.

No problem, I think it makes sense to leave that for 3.1 as these changes are already quite large, Thanks for the Merging that, as mentioned I will look at cleaning my bits up a little & implementing the Android Interrupt read method (yet to start testing the Android interrupt, next on my list! :) )

I will reply to that specific issue with some more information.

MelbourneDeveloper, 2019-07-08:
> I think passing the interface number in the ctor is a good option, to override the default behaviour (eg. It will default to 0, but can be set in ctor if a different interface is needed).

Yep. This is looking light the right option I think. I might take a look at some other libraries and see what they are doing.

MelbourneDeveloper, 2019-07-14:
@mxjones quick update.

The refactoring of UWP didn't according to plan, but it did elucidate some things. It turns out that my reference device for USB Trezor doesn't use bulk or control transfer. It uses Interrupt. This is what has caused so much confusion and why the code was all wrong on Android, and UWP.

You probably already saw this, but [this](https://github.com/MelbourneDeveloper/Device.Net/blob/dee398c6ef03cd4283c50eaa7a6b7d64fa16cbaa/src/Usb.Net.UWP/UWPUsbDevice.cs#L63) is what I had to do to get the Trezor working on UWP. It is listening to interrupt events. One good thing about this is that it tells me that I am on the right track with the event idea... On UWP, they've implemented events for when interrupt data is received. So, that makes me more confident that events are the right choice.

Anyway, all the code I've written is hack code to make the Trezor work. So, I now need a new device to test with that is more normal. Do you know of any cheap devices I can order that require sending/receiving?

The latest code is in 78-UWPRefactor if you want to have a look.

MelbourneDeveloper, 2019-07-14:
@mxjones , anyway, I'm getting there. All this should have never been necessary. It's just ended up this way because of the design of the Trezor.

closed 2019-07-17 by MelbourneDeveloper

mxjones, 2019-07-17:
@MelbourneDeveloper Sorry been busy but have been checking in on this

Aaah, I see where the confusion came from! That is odd for that device to be using the interrupt for standard communication, some devices don't support bulk transfer - but I would think it should use control over interrupt for that if that was the case - not sure!

I hadn't seen that, but makes sense - It sounds like you are in a good place now though and the design sounds promising - I especially like the event idea for interrupt data, so that the user doesn't need to explicitly poll themselves.

I will take a look at that branch / latest code tomorrow and pull it down / point my test applications at it and have a play.

As far as cheap devices, I'm not actually too sure. I am currently using DSLR Cameras (not cheap!) and don't have much experience with any other devices to be honest - I would say a mouse/keyboard would *most likely* use interrupt endpoint for their transfer (key press -> event model sounds reasonable to me) but apart from that not actually too sure - so sorry I can't be of more help on that one, but I am happy to test with the devices I have and feed that back.

MelbourneDeveloper, 2019-07-21:
@mxjones @schotime 

Version 3.0 of Usb.Net is up: https://www.nuget.org/packages/Usb.Net/3.0.0-beta

I would be really stoked if you could test and report issues. Thanks.

mxjones, 2019-08-01:
I've been super busy and not had as much time to test as I wanted but can confirm it is working OK for me in my Windows test code, I will look porting over my Android test code and testing that also - will let you know, or raise any issues if anything comes up.

Thanks for your great work on this!

MelbourneDeveloper, 2019-08-03:
@mxjones thanks! That's awesome. Beta 2 is up now. If you could give that a bash, it would be great.

MelbourneDeveloper, 2019-08-03:
@schotime Version 3 Beta 2 is up. The main feature change from 2->3 has been the ability to select Usb Interfaces and Endpoints on all platforms. The UsbDevice class now has a property called UsbInterfaceManager. This allows you to set the interface after initialization, and then you can select endpoint on the IUsbInterface. Documentation is coming on this.