MelbourneDeveloper/Device.Net #27: USB - Raw COM Port Access

MelbourneDeveloper, 2019-01-18:
It is possible to scan through COM ports and listen for data. While I haven't got a need for this, one use case is for GPS devices. Would anyone find this useful? Anyone willing to help? I have code for it already, but I'd just need for it to be integrated in to Device.Net in a useful way.

hbopuri, 2019-01-22:
Can you please share the code I can test it and see if I can join the team

MelbourneDeveloper, 2019-01-22:
@hbopuri , if you would like to write a pull request, here is some code that works well. I've used it before for GPS reading:

https://stackoverflow.com/a/9092519/1878141

However, there are a lot of complications with this. My library is designed for enumerating through devices, while serial port reading is different. The pattern isn't to enumerate through the devices, it's to listen on every port until you find a useful message. This doesn't really fit with Device.Net, but it's not impossible either.

I will implement it at some point, but this is not a high priority right now.

MelbourneDeveloper, 2019-08-29:
@hbopuri there is now a library for raw COM/Serial Port access. It's released here:

[SerialPort.Net](https://www.nuget.org/packages/SerialPort.Net/3.0.0)

The code is [here](https://github.com/MelbourneDeveloper/Device.Net/tree/master/src/SerialPort.Net)

[WindowsSerialPortDevice](https://github.com/MelbourneDeveloper/Device.Net/blob/master/src/SerialPort.Net/WindowsSerialPortDevice.cs)

You can see unit tests here. One of these tests will expect a device to be on COM3

https://github.com/MelbourneDeveloper/Device.Net/blob/be795b38f713076a5234eb46054ae19f1314ede2/src/Device.Net.UnitTests/IntegrationTests.cs#L19

But this one will scan the COM ports and try to connect to all:
https://github.com/MelbourneDeveloper/Device.Net/blob/be795b38f713076a5234eb46054ae19f1314ede2/src/Device.Net.UnitTests/IntegrationTests.cs#L74

I expect the tests to have issues on your machine because you're going to have different devices connected on different ports.

Please let me know how you go with this. I will be closing this issue soon if I don't hear anything.

MelbourneDeveloper, 2019-08-29:
@hbopuri I will document this soon. There is an [issue stub](https://github.com/MelbourneDeveloper/Device.Net/issues/110) for documentation. For now, the unit tests will give you what you need. Also, the windows sample will list out COM ports that are available with a device.

closed 2019-12-09 by MelbourneDeveloper