MelbourneDeveloper/Device.Net #38: The "ResolveLibraryProjectImports" task failed- problem

yanivhanya, 2019-02-03:
I downloaded the sample. but I can't run it on my visual studio. I get error message-
The "ResolveLibraryProjectImports" task failed unexpectedly.System.IO.FileNotFoundException: Could not load assembly 'Device.Net, Version=0.0.0.0, Culture=neutral, PublicKeyToken='. Perhaps it doesn't exist in the Mono for Android profile?File name: 'Device.Net.dll'   at Java.Interop.Tools.Cecil.DirectoryAssemblyResolver.Resolve(AssemblyNameReference reference, ReaderParameters parameters)   at Java.Interop.Tools.Cecil.DirectoryAssemblyResolver.Resolve(String fullName)   at Xamarin.Android.Tasks.ResolveLibraryProjectImports.Extract(DirectoryAssemblyResolver res, ICollection`1 jars, ICollection`1 resolvedResourceDirectories, ICollection`1 resolvedAssetDirectories, ICollection`1 resolvedEnvironments)   at Xamarin.Android.Tasks.ResolveLibraryProjectImports.Execute()   at Microsoft.Build.BackEnd.TaskExecutionHost.Microsoft.Build.BackEnd.ITaskExecutionHost.Execute()   at Microsoft.Build.BackEnd.TaskBuilder.<ExecuteInstantiatedTask>d__26.MoveNext()

I tried a lot of "google" solutions. including rebuild and so on. with no success. 
do you have any idea how to run the project?

MelbourneDeveloper, 2019-02-03:
Hi. I've not seen this one before. Is this just a clean sample, or have you changed it in some way? If so, please send me a copy through GitHub or some other link...

This is what I recommend trying first:

- Update Visual Studio 2017 to the latest version
- Clear your NuGet cache
- Do a Git clean on the repo
-Try again

If you have any troubles with the above, let me know.

@yanivhanya

MelbourneDeveloper, 2019-02-03:
Also, have you tried right clicking on the Device.Net project and hitting "Build". Does that succeed?

yanivhanya, 2019-02-04:
Hello, thank you very much for your reply.
it is the sample, without any changes 'till now.

I tried to clean the cashe, rebuild, and I also moved the folder to a "clean" location, I got the same error.

when I tried to build only the "device.Net" project, I got different error-
"Severity	Code	Description	Project	File	Line	Suppression State
Error		Cannot import the following key file: Device.Net.pfx. The key file may be password protected. To correct this, try to import the certificate again or manually install the certificate to the Strong Name CSP with the following key container name: VS_KEY_18A88A9FC8AF5E92	Device.Net.UWP			
".

Again, thanks for the help, I really need this project to my bigger project.

MelbourneDeveloper, 2019-02-04:
@yanivhanya yeah that's what I thought. Try the develop branch. That issue was fixed.

yanivhanya, 2019-02-04:
I"m sorry, I just downloaded it, cleaned cache, and ran it. I got the same error messages.

MelbourneDeveloper, 2019-02-04:
Are you still getting the key signing error? It should not happen in debug mode. I fixed this in the csproj file yesterday. In release mode it will sign the package but in debug it won't.

If you're really stuck just go in to the properties of the project and turn off signing.

yanivhanya, 2019-02-04:
I tried all the process again, downloaded, clean every thing, change path. but I got the same error.
I include screen shot, if you'll  be kind enough to check?
 
[image]

MelbourneDeveloper, 2019-02-04:
Turn off signing

[image]

@yanivhanya

MelbourneDeveloper, 2019-02-04:
@yanivhanya I've now fixed all three projects so that as long as you compile in debug, it should not try to sign the assemblies.

MelbourneDeveloper, 2019-02-04:
@yanivhanya I actually got the same "ResolveLibraryProjectImports" error for the first time today. My issue was that all the NuGet packages had not downloaded properly. But, I am still pretty sure your issue is coming up because the code signing is still turned on for several libraries.

yanivhanya, 2019-02-05:
it is working for me now after removing the sign

MelbourneDeveloper, 2019-02-05:
@yanivhanya awesome! Can I close this ticket?

MelbourneDeveloper, 2019-02-05:
@yanivhanya I am closing the ticket but please @ me if you have other issues.

closed 2019-02-05 by MelbourneDeveloper

yanivhanya, 2019-02-07:
Hello. Thanks for help. I was OOO for two days. 

I don't get any error messages for now. but I couldn't make it work. I develop in visual studio for more them 15 years, but I guess I ignorant in USB use or may be your project is not intend for my purpose. 

I need to connect my android tablet to a "controller" that get ascii code command via USB port. I did it successfully with my windows, but couldn't do it for android.   

does your project fit for this purpose? or I missing something? 
if yes, do you any example of how to it? just to send the string/ ascii  "1" to the port?

Thanks  a lot in advance.

MelbourneDeveloper, 2019-02-07:
The basic answer is that the code should be exactly the same between windows and Android. It should be no problem. The only difference is that you have to register the AndroidUsbDeviceFactory. Have a look here: https://github.com/MelbourneDeveloper/Device.Net/blob/master/src/Usb.Net.AndroidSample/MainActivity.cs

The best thing to do is put your code on GitHub and I will investigate.

yanivhanya, 2019-02-07:
well, for now the code I use is the same you share with me in the link. I run the app. disconnect it from the PC (so I can connect the usb controller). and press the button. I just get "waiting for device" message.

the code I use for the windows is-
```
 Private Sub btnConnect_Click(ByVal sender As System.Object, ByVal e As System.EventArgs) Handles btnConnect.Click
        SerialPort1.PortName = cmbPort.Text
        SerialPort1.BaudRate = cmbBaud.Text
        SerialPort1.Parity = IO.Ports.Parity.None
        SerialPort1.StopBits = IO.Ports.StopBits.One
        SerialPort1.DataBits = 8
        SerialPort1.Open()
        btnConnect.Enabled = False
        btnDisconnect.Enabled = True
    End Sub
``` 

and then-

```
    Private Sub btnSend_Click(ByVal sender As System.Object, ByVal e As System.EventArgs) Handles btnSend.Click
        SerialPort1.Write("1")
    End Sub
```

MelbourneDeveloper, 2019-02-08:
@yanivhanya  that is Windows specific code and I doubt it will ever work on Android.

yanivhanya, 2019-02-08:
Thanks a lot. does it means there no way to send messages to USB connected device from Xamarin? I don't have to use exactly my windows code, just need to send "1" or "2"... to the device.

MelbourneDeveloper, 2019-02-08:
No. Here is the Android sample. The code you have posted has nothing to do with this repo. 

https://github.com/MelbourneDeveloper/Device.Net/tree/master/src/Usb.Net.AndroidSample