MelbourneDeveloper/Device.Net #40: UWP Limitations for Specific Collections

freeblazin, 2019-02-11:
I am attempting to enumerate an HID device in a UWP app on Win 10 IOT platform.  The device is a touchscreen monitor and utilizes the Consumer Usage Page collection.  (https://www.chalk-elec.com/?p=2060)

I have added the device capabilities to the Package.appxmanifest, but have been unsuccessfull at enumerating the device, as the output of DeviceManager.Current.GetDevicesAsync method always returns null with no error.

The Official UWP HID API documents the blocking of device usages under the the Consumer Usage Page collection.  Does this limitation still apply to this library?

MelbourneDeveloper, 2019-02-11:
Hi. Can you upload your repo somewhere so I can have a look?

MelbourneDeveloper, 2019-02-12:
@freeblazin I recommend cloning the repo and putting a breakpoint on this line:
https://github.com/MelbourneDeveloper/Device.Net/blob/d516d87a6e7dc279874e19334a5f2d3ba30ae71b/src/Device.Net.UWP/UWPDeviceFactoryBase.cs#L47

> The Official UWP HID API documents the blocking of device usages under the the Consumer Usage Page collection. Does this limitation still apply to this library?

This library is a wrapper for the standard UWP Hid API so if it doesn't work with the standard UWP API, it won't work here. I also recommend writing a console app and attempting to enumerate the devices in .NET Core to check that it's not something specific to UWP. It's very hard to get the package manifest right.

MelbourneDeveloper, 2019-02-19:
Closed due to inactivity

closed 2019-02-19 by MelbourneDeveloper