MelbourneDeveloper/Device.Net #44: About Writing to HID Device,WriteAsync should set reportID instead of 0 to bytes

Halry, 2019-02-23:
In WIndowsHIDDevice.cs,Line 180,bytes[0] should be set to the reportID from HID device Out definition.Otherwise,write to device will cause exception"IO operation will not work. Most likely the file will become too long or the handle was not opened to support synchronous IO operations."

MelbourneDeveloper, 2019-02-23:
@Halry thanks for reporting this. It seems about right. Can you submit a pull request for this?

Where do we get the report id from?

Is this for the case where the buffer size is 65 instead of 64?

MelbourneDeveloper, 2019-02-23:
@Halry do you mean this method 

HidD_GetInputReport?

https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/hidsdi/nf-hidsdi-hidd_getinputreport

MelbourneDeveloper, 2019-02-23:
@Halry please see this page:
https://github.com/MelbourneDeveloper/Device.Net/wiki/Hid---DataHasExtraByte---64-byte-Buffer-Vs.-65-Byte-Buffer

MelbourneDeveloper, 2019-02-23:
@Halry can you please send me the documentation for the device you are trying to work with? 

I need to understand a few things. Will the Report ID change on each write? Or, is it always going to be the same? I just need to know what the use case is.

MelbourneDeveloper, 2019-02-24:
@Halry , can you please try https://www.nuget.org/packages/Hid.Net/2.6.0-beta

I've now added two new methods WriteReportAsync(https://github.com/MelbourneDeveloper/Device.Net/blob/4b6b32930f6b58415d30c082389edd863c81a1ed/src/Hid.Net/Windows/WindowsHIDDevice.cs#L171), and ReadReportAsync(https://github.com/MelbourneDeveloper/Device.Net/blob/4b6b32930f6b58415d30c082389edd863c81a1ed/src/Hid.Net/Windows/WindowsHIDDevice.cs#L136). These methods should allow you to specify or receive a ReportId.

Let me know if this fixes your issue.

Halry, 2019-02-26:
Yes,it works.

closed 2019-02-26 by Halry

MelbourneDeveloper, 2019-02-26:
Awesome! Thanks for testing