MelbourneDeveloper/Device.Net #48: Exception will be thrown on some computers.

Halry, 2019-02-26:
I have tested my program on my PC and it works fine(except read,Idon't know why my input report won't be picked up via WriteAndReadAsync()),but at least it won't throw an exception.
On my laptop,otherwise,it will throw a lot of throw,after starting the listener.
//exception below
An exception of type 'System.Exception' occurred in test.exe but was not handled in user code
Could not get device interface detail. Error code: 122
//==============
and call stack lead me to WindowsDeviceFactoryBase.cs,line:76

It seems my laptop has some wired hid devices made the SetupDiGetDeviceInterfaceDetail() failed,because the program can listen to the device and initialize device.
Maybe listener should try to get all hid devices on the computer?I don't know.

MelbourneDeveloper, 2019-02-27:
@Halry can we please organise a TeamViewer session so I can have a look at your computer and debug the problem? What is your time zone?

Also, if you could turn on all exceptions, and take some screenshots of the line where the code breaks, with locals etc. that would help me.

Halry, 2019-02-28:
My timezone is GMT+8.
I will try to turn on all exceptions ASAP.

MelbourneDeveloper, 2019-02-28:
Cool . That's only three hours behind me. I might be free tomorrow night or Saturday. Please join the Slack group and you can message me directly. I should be home about 3pm tomorrow. @halry

Halry, 2019-03-01:
After commented "WindowsDeviceBase.HandleError" in WindowsDeviceFactoryBase.cs at line:76.It seems works fine,Although it won't throw an exception,it works.But maybe it won't be safe.
And when my device unplugged,it will throw a lot of errors about System.IO can't find device,Is there a way to optimize it?

closed 2019-03-02 by MelbourneDeveloper