MelbourneDeveloper/Device.Net #51: GetConnectedDeviceDefinitionsAsync never finishes

nosmokingbandit, 2019-03-12:
Take this basic example:

    using Device.Net;
    using Hid.Net.Windows;
    using System;

    namespace ConsoleApp
    {
        class Program
        {
            static void Main(string[] args)
            {
                Test.Find();
                Console.ReadLine();
            }
        }

        class Test
        {
            public static async void Find()
            {
                WindowsHidDeviceFactory.Register();
                var devices = await DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(new FilterDeviceDefinition { VendorId = 0x04D9, ProductId = 0x0130 });
                Console.WriteLine("this is never printed to console");
                foreach (var device in devices)
                {
                    Console.WriteLine(device.DeviceId);
                }
            }
        }
    }

The application hangs before `Console.WriteLine("this is never printed to console");` is reached.

Windows 10 x64 with netcore 2.2

MelbourneDeveloper, 2019-03-12:
@nosmokingbandit thanks for reporting! Investigating. Are you interested in letting me TeamViewer on to your computer?

MelbourneDeveloper, 2019-03-12:
If you have time to help me debug, please clone the repo and step in to that method so we can figure out exactly which line fails. 

Is this a new problem? Or it never worked for you?

MelbourneDeveloper, 2019-03-12:
[image]

[image]

@nosmokingbandit it works fine on my computer. I have to go to work soon. If you get back to me quickly, I can remote on to your computer.

nosmokingbandit, 2019-03-12:
I have to admit that my Visual Studio debugging skills aren't quite as good as I'd like, but part of the problem seems to be that `Device.Net.APICalls.SetupDiEnumDeviceInterfaces` always results in a `NO_MORE_ITEMS` error on the first call. I'm going to see if forcing windows to use libusb instead changes anything for me.

MelbourneDeveloper, 2019-03-12:
@nosmokingbandit so you're not interested in helping me to diagnose the problem?

frajasalo, 2019-03-27:
Hi,

I tested the code and it prints on the console, but found that when one (or many) device(s) matches the filter an exception occures in Device.Net.dll and the returned list is empty.

Win 10 (1803, Build 17134.590), 64Bit, .NET 4.6.1

MelbourneDeveloper, 2019-03-27:
@frajasalo can I please remote on to your computer to debug the problem? I can't recreate it. 

If not, can you please turn on break on all exceptions and take screenshots of the exception details? Also post a ToString of the exception?

MelbourneDeveloper, 2019-03-28:
@frajasalo you can directly paste screenshot images in to GitHub comments and I will see them.

frajasalo, 2019-03-28:
Sorry have been busy today.  I'll try providing the screenshots tomorrow or on weekend.

MelbourneDeveloper, 2019-03-28:
@frajasalo thanks!

MelbourneDeveloper, 2019-03-28:
@frajasalo very keen to support people using the library and correct bugs.

MelbourneDeveloper, 2019-03-28:
@frajasalo 

> I tested the code and it prints on the console, but found that when one (or many) device(s) matches the filter an exception occures in Device.Net.dll and the returned list is empty.
> 
> Win 10 (1803, Build 17134.590), 64Bit, .NET 4.6.1

Actually, this could be happening for a different reason... Some devices don't allow information to be obtained from them. So, they fail. What should happen is that the exception gets stopped and logged if you're using a logger. And you're correct, the list will contain no elements. That might be what's happening in your case. But, I still would like to get more information so I can diagnose exactly.

MelbourneDeveloper, 2019-03-28:
@nosmokingbandit, @frajasalo, I have fixed a bug in the develop branch. It may be related to the problems you've been having. I will do further investigation, test and get it in to a release soon.

MelbourneDeveloper, 2019-03-30:
@nosmokingbandit , @frajasalo the separate known issue is fixed. It may have been related, but I'm guessing not... 

I will do some further investigation on Device.Net.APICalls.SetupDiEnumDeviceInterfaces soon.

frajasalo, 2019-04-08:
Helllo,

and sorry for being late with my answer.

Its strange but now it seems to work as expected.  Maybe I had to restart Visual Studio or the whole system after installing the NuGet package !?

Updated to version 2.7.0 and it's working also.

Thanks!

MelbourneDeveloper, 2019-04-11:
@nosmokingbandit closing this, but let me know if you want to take another stab at it and I'll try to fix any issues you are having.

closed 2019-04-11 by MelbourneDeveloper