MelbourneDeveloper/Device.Net #58: Working with Composite Devices on Windows 10

davidehnis, 2019-04-11:
When I attempt to use Device.Net (2.7) with my iPhone use an instance of FilterDeviceDefinition, the library doesn't find it.  When looking for it via Device Manager (Windows 10), the device is shown under a composite usb manifest.  

When I look for it via code using the Management object searcher, it finds the device without issue.

`ManagementObjectCollection collection;
 using (var searcher = new ManagementObjectSearcher
 ($"Select * From Win32_PnPEntity WHERE DeviceID Like '%{deviceId}&MI_%'"))
  collection = searcher.Get();`

when I use device.net to accomplish the same thing, using the code below, GetDevices() doesn't return the iPhone:

`var deviceDefinitions = new List<FilterDeviceDefinition>
 {
            new FilterDeviceDefinition{ DeviceType= DeviceType.Usb, VendorId= 0x05AC, 
                   ProductId=0x12A8, Label="Apple iPhone" },
 };

 var devices = await DeviceManager.Current.GetDevices(deviceDefinitions);`

MelbourneDeveloper, 2019-04-11:
@davidehnis interesting. I will test with an iPhone myself and see what happens. Thanks.

MelbourneDeveloper, 2019-04-12:
@davidehnis out of curiosity, what are you trying to do with the iPhone? The iPhone might not be responding because it doens't have a custom device type. Generally, iPhones only allow transferring files backward and forward. For that I would recommend standard file transfer instead of Device.Net. 

Is there some transfer mode on a iPhone that I don't know about?

davidehnis, 2019-04-13:
@MelbourneDeveloper , mainly I using it as a testing vehicle.  I'm creating a solution that will need to connect to a custom device.  I was just making sure that device.net could find any device and that's it.  Just using the iPhone to plan/test for the actual.

MelbourneDeveloper, 2019-04-13:
@davidehnis that makes sense. However, it might be the case that Apple have not registered any custom Hid or USB devices for the iPhone. That would mean that they may show up in a scan, but Device.Net won't include them in the returned list because it can't get enough information about the device. I still plan to test that hypothesis for myself, but I'd recommend testing with some device that you know has a custom interface.

davidehnis, 2019-04-15:
@MelbourneDeveloper , that's fair.  So it's like I'm getting a false negative.

MelbourneDeveloper, 2019-04-15:
@davidehnis I'm guessing so. I will see if I can get my hands on an iPhone.

MelbourneDeveloper, 2019-04-19:
Does anyone have an iPhone? I probably won't have access for a couple of weeks.

AlmaJensen, 2019-05-02:
I don't know if this is relevant here or not but I thought I would mention it incase it's helpful but I have some external scanning sleds that I use with ipods and before any of my apps can make use of the sled I had to add this to the info plist.

	<key>UISupportedExternalAccessoryProtocols</key>
	<array>
		<string>com.datecs.linea.pro.msr</string>
		<string>com.datecs.linea.pro.bar</string>
	</array>

MelbourneDeveloper, 2019-05-02:
@AlmaJensen that's interesting. Are you talking about the plist for an iPod app? Or a Mac app?

MelbourneDeveloper, 2019-06-23:
@davidehnis I have pushed out a new release that has more error logging - especially around enumerating connected devices. I have also updated the wiki on how to enable logging. You might want to try it. It may give you information about why the iPhone doesn't show up.

https://github.com/MelbourneDeveloper/Device.Net/wiki/Enumerating-Connected-Devices

closed 2019-06-28 by MelbourneDeveloper