MelbourneDeveloper/Device.Net #60: How to enumerate all connected HID devices?

pc8181, 2019-05-20:
m newbie here ...
Can i get all connected HID devices with it's VID and PID ?

can i get sample codes ?.

Thanks...

pc8181, 2019-05-20:
`
          Imports Device.Net
           Imports Hid.Net.Windows

    Public Class Form1
       Private Sub Form1_Load(sender As Object, e As EventArgs) Handles MyBase.Load
        Try
            GetHID()
        Catch ex As Exception
        End Try
    End Sub

    Public Async Sub GetHID()
        WindowsHidDeviceFactory.Register()
        Dim dfnt As New FilterDeviceDefinition With {.DeviceType = DeviceType.Hid, .VendorId = &H5, .ProductId = &H5}
        Dim dvs = Await DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(dfnt)
    End Sub
     End Class

`

this above code could not compile because KasperSky Free version blocked 

[image]

MelbourneDeveloper, 2019-05-20:
> Can i get all connected HID devices with it's VID and PID ?

@pc8181

First, register the Hid factory. On Windows:

```cs
WindowsUsbDeviceFactory.Register(null);
```
Then, call:
```cs
var devices = await DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(null);
```
Please run this Windows sample:
https://github.com/MelbourneDeveloper/Device.Net/tree/master/src/Usb.Net.WindowsSample

You will see Hid devices like this:
[image]

It looks like you have malware on your computer. You will need to fix your computer.

MelbourneDeveloper, 2019-05-20:
@pc8181 your VB sample looks correct to me. You just need to fix your malware problem.

pc8181, 2019-05-21:
thanks for your detailed reply sir sure i will try to scan and fix my computer

Thanks Sir :)

closed 2019-05-23 by MelbourneDeveloper

heartacker, 2019-10-15:
@MelbourneDeveloper  i have register and program   All the way round in "await"
```
            var devices = await DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(null);
```  
i am sure my hid devices are connected       
```
            WindowsHidDeviceFactory.Register(logger, tracer);
            WindowsUsbDeviceFactory.Register(logger, tracer);
            //WindowsSerialPortDeviceFactory.Register(Logger, Tracer);
            //deviceListener = new DeviceListener(df, 3000) { Logger = new DebugLogger() };
            //deviceListener.DeviceInitialized += DeviceListener_DeviceInitialized;
#if false

            //Define the types of devices to search for.
            //var df = new List<FilterDeviceDefinition>
            //{
            //       new FilterDeviceDefinition()
            //    {
            //        DeviceType = DeviceType.Hid,
            //        VendorId = 0x0d28,
            //        ProductId =0x2024
            //    }
            //};
            //var task = await DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(
            //                       new FilterDeviceDefinition()
            //                       {
            //                           DeviceType = DeviceType.Hid,
            //                           VendorId = 0x0d28,
            //                           ProductId = 0x2024
            //                       });  
#endif

            var devices = await DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(null);
            Console.WriteLine("Currently connected devices: ");

            foreach (var device in devices)
            {
                Console.WriteLine(device.DeviceId);
            }
```
what is wrong, help me please
thx

MelbourneDeveloper, 2019-10-15:
@heartacker please the log output

heartacker, 2019-10-16:
@MelbourneDeveloper I create a DLL to enumerate USB devices，and  i usd a console exe to invoke the DLL,then DLL work well. but i usd  a netcore 3.0 WPF to invoke the DLL ,the DLL does not work. :confounded:
I can deal with it,how  can  I do,thanks

MelbourneDeveloper, 2019-10-16:
@heartacker I need to see the log output. Please post it.

MelbourneDeveloper, 2019-10-16:
@heartacker please read:

https://github.com/MelbourneDeveloper/Device.Net/wiki/Debugging,-Logging,-and-Tracing