MelbourneDeveloper/Device.Net #61: How use Device.Net for real time monitoring data from Hid device?

UporotayaPanda, 2019-06-05:
How I can use library for real time monitoring data from HID device?

MelbourneDeveloper, 2019-06-05:
@UporotayaPanda connec to the device using the sample code, and then call [ReadAsync](https://github.com/MelbourneDeveloper/Device.Net/blob/3a7324746e01a0e252d2a4d1b630ed4b623f3903/src/Device.Net/IDevice.cs#L16) in a loop.

UporotayaPanda, 2019-06-05:
@MelbourneDeveloper , what about device listener? 
I'm trying to use it
Maybe you have an example of how to implement real-time data reading?

MelbourneDeveloper, 2019-06-05:
@UporotayaPanda 
One step at a time. Connect to a device first and make sure you can do that. Then call ReadAsync in a loop. Please post your code so I can see what you've done. There are plenty of examples for you to look at.

UporotayaPanda, 2019-06-05:
@MelbourneDeveloper 
It's info about my mouse device.

[image]

```
        private static async Task Read()
        {
            var definitions = new List<FilterDeviceDefinition>
            {
                new FilterDeviceDefinition
                {
                    DeviceType = DeviceType.Hid, 
                    VendorId = 0x046D, 
                    ProductId = 0xC31C
                },
                new FilterDeviceDefinition
                {
                    DeviceType = DeviceType.Hid, 
                    VendorId = 0x09DA, 
                    ProductId = 0x1985
                }
            };

            var devices = await DeviceManager.Current.GetDevicesAsync(definitions);

            if (devices.Any())
            {
                //read   
            }
        }
```

I try get my mouse and keyboard devices. But I always have null...

MelbourneDeveloper, 2019-06-06:
@UporotayaPanda, I'm not sure that the mouse will show up. Mice are usually not custom USB devices. They are usually handled by the OS at the kernel level. 

Please see this sample and debug it. It will list all devices that can be connected to

https://github.com/MelbourneDeveloper/Device.Net/blob/3a7324746e01a0e252d2a4d1b630ed4b623f3903/src/Usb.Net.WindowsSample/Program.cs#L105

UporotayaPanda, 2019-06-07:
@MelbourneDeveloper  It work. Thanks.
Is it possible to track the flow of data without using a loop?
Using event?

MelbourneDeveloper, 2019-06-07:
@UporotayaPanda excellent! Unfortunately not with the existing libraries. But, if you want to look at the low level code, there would be a way to do it.

closed 2019-06-09 by MelbourneDeveloper