MelbourneDeveloper/Device.Net #62: Trouble discovering USB mass storage device

charliez23, 2019-06-10:
Your Window's sample discovers Trezor devices. I modified the sample to discover a plugged in USB storage device but was unsuccessful (0 devices discovered).

Your sample code modified to discover the USB storage device: 

new DeviceDefinition{ DeviceType= DeviceType.Usb, VendorId= 0x0781, ProductId=0x5575, ReadBufferSize=64, WriteBufferSize=64, Label="SanDisk Cruzer Glide USB Device"}.

I tried both device types = Usb and Hid

I got the Vendor ID and Product ID from looking at properties in the Device Manager i.e. USB\VID_0781&PID_5575\4C530001300730112171

Is there a way to discover ALL devices without knowing the Vendor ID and Product ID? I noticed they are nullable types and set both to null but still got 0 devices discovered.

I tried an alternate route also using you libraries from NuGet. I get the devices using and read the property DeviceID using (var searcher = new ManagementObjectSearcher(@"Select * From Win32_PnPEntity"))	collection = searcher.Get(); string Device_ID = (string)device.GetPropertyValue("DeviceID");

Device_ID: SWD\WPDBUSENUM_??_USBSTOR#DISK&VEN_SANDISK&PROD_CRUZER_GLIDE&REV_1.00#4C530001300730112171&0#{53F56307-B6BF-11D0-94F2-00A0C91EFB8B}

Using your USB.NET library: Main {	WindowsUsbDeviceFactory.Register(); var device = new WindowsUsbDevice(Device_ID); var ia = device.InitializeAsync(); ia.Wait(); Console.WriteLine(device.IsInitialized); }

ia.Wait() throws an exception "Device handle no good". Any idea what device ID is expected?

MelbourneDeveloper, 2019-06-10:
@charliez23
Hi. This question has been coming up a bit lately. I will write a more detailed response soon, but meantime, to answer your question, simply don't specify any filters. Put null in there. that's how you get all devices.

https://github.com/MelbourneDeveloper/Device.Net/issues/61

MelbourneDeveloper, 2019-06-10:
@charliez23 I have updated this [document on enumerating all devices](https://github.com/MelbourneDeveloper/Device.Net/wiki/Enumerating-Connected-Devices) . There is sample code there enumerate all devices.

"Device handle no good" probably means that what you are passing is not a valid custom USB device Id. You can see the line of code where it is failing [here](https://github.com/MelbourneDeveloper/Device.Net/blob/3a7324746e01a0e252d2a4d1b630ed4b623f3903/src/Usb.Net/Windows/WindowsUsbDevice.cs#L53). Not all devices will connect even when the device id is registered. To be honest, I doubt that a USB stick is a custom USB device. Normally, you'd just copy files across like a normal hard drive. I can't see much point in trying to talk to it via a custom protocol. But, if there's some special reason, I'd really like to hear about it so I can put support in for this scenario.

MelbourneDeveloper, 2019-06-11:
@charliez23 I recommend that you get the latest on the Develop branch and retry the Windows sample. I have made improvements to error handling in the libraries and updated the documentation on enumerating devices. The important thing is to use logging. Please have a read here: https://github.com/MelbourneDeveloper/Device.Net/wiki/Enumerating-Connected-Devices

I will do some testing of the new code and eventually release a new NuGet package 2.8 soonish.

MelbourneDeveloper, 2019-06-11:
@charliez23 if you think you have actually uncovered a bug, please let me know.

MelbourneDeveloper, 2019-06-17:
@charliez23 I'm going to need to close this soon. Anything I can help you with?

MelbourneDeveloper, 2019-06-23:
@charliez23 I have pushed out a new release that has more error logging - especially around enumerating connected devices. I have also updated the wiki on how to enable logging. You might want to try it. It may give you information about why the mass storage device cannot be connected to.

https://github.com/MelbourneDeveloper/Device.Net/wiki/Enumerating-Connected-Devices

closed 2019-06-28 by MelbourneDeveloper