MelbourneDeveloper/Device.Net #63: ArgumentOutOfRangeException calling InitializeAsync

bryans69, 2019-06-17:
Simple test project at this point in time, but get System.ArgumentOutOfRange when calling InitializeAsync

Device has been found OK

  var deviceDefinitions = new List<FilterDeviceDefinition>
    {
        new FilterDeviceDefinition{ DeviceType= DeviceType.Hid, VendorId= 0x043D, ProductId=0x0235 }
    };
            var devices = await DeviceManager.Current.GetDevicesAsync(deviceDefinitions);
            var device =new WindowsHidDevice( devices.FirstOrDefault().DeviceId); <- device points to expected USB device
            await device.InitializeAsync(); <- Exception here

Running this in a simple windows form at this point in time

System.ArgumentOutOfRangeException
  HResult=0x80131502
  Message=Positive number required.
Parameter name: bufferSize
  Source=mscorlib
  StackTrace:
   at System.IO.FileStream..ctor(SafeFileHandle handle, FileAccess access, Int32 bufferSize, Boolean isAsync)
   at Hid.Net.Windows.WindowsHidDevice.Initialize()
   at Hid.Net.Windows.WindowsHidDevice.<InitializeAsync>b__24_0()
   at System.Threading.Tasks.Task`1.InnerInvoke()
   at System.Threading.Tasks.Task.Execute()
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at Hid.Net.Windows.WindowsHidDevice.<InitializeAsync>d__24.MoveNext()
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter.GetResult()
   at WindowsFormsApp1.Form1.<button2_Click>d__4.MoveNext() in C:\Users\bryans\source\repos\WindowsFormsApp1\WindowsFormsApp1\Form1.cs:line 94

MelbourneDeveloper, 2019-06-17:
Thanks for reporting!

I'll go over the code and see if we can find something wrong there. It looks like that device isn't allowing some kind of access though... Maybe there are two device IDs you can connect to?

Any chance you can turn on first chance exceptions , debug and send me a screenshot?

MelbourneDeveloper, 2019-06-17:
@bryans69 it looks like the issue is [here](https://github.com/MelbourneDeveloper/Device.Net/blob/3a7324746e01a0e252d2a4d1b630ed4b623f3903/src/Hid.Net/Windows/WindowsHIDDevice.cs#L71). It is expecting that the ConnectedDeviceDefinition will have a ReadBufferSize, and a WriteBufferSize. However, it looks like those things are not being obtained for some reason. 

You could modify the code so that you can specify those two values by adding setters [here](https://github.com/MelbourneDeveloper/Device.Net/blob/3a7324746e01a0e252d2a4d1b630ed4b623f3903/src/Hid.Net/Windows/WindowsHIDDevice.cs#L32).

If this solves your problem, I'd really like to hear about it. If it does, I could make these values writable so that others can solve the same problem.

bryans69, 2019-06-17:
Hi Christian

Thanks for the quick response. Not sure if I’ve enabled the option correctly, but looks like the same error

[cid:image003.png@01D52561.04EB90E0]

As far as I can tell, it is the only device on this address

[cid:image002.png@01D52560.912E33B0]

It’s a USB printer, and I know that it supports being written to (just trying to work out how to). At this point I’m just trying to initialize, not actually write to the device, that is step 2.

Hope this helps, please let me know if you need anything else

Regards

Bryan

Bryan Stone
Email bryan_stone@bits.uk.com
Business I.T. Systems Ltd
104A Roebuck Road
Chessington, Surrey, KT9 1EU
Tel: 020 3258 4020
Fax: 020 3258 4030
Email: info@bits.uk.com
Website: www.bits.uk.com

DISCLAIMER:
The information in this E-Mail message is confidential and may be legally privileged. It is intended solely for the addressee and access to this message by anyone else is unauthorised. If you have received the message in error, please notify us immediately and delete it. If you are not the intended recipient, any disclosure, copying, distribution or any other use is prohibited and may be unlawful. Business I.T. Systems Ltd is a limited company registered in England and Wales. Registered number: 2563646. Registered office: 104A Roebuck Road, Chessington, Surrey, KT9 1EU
Please consider the environment. Is it necessary to print this email?

MelbourneDeveloper, 2019-06-17:
@bryans69 I can't see your screenshots there. You just need to copy and paste the image in.

bryans69, 2019-06-17:
[image]

bryans69, 2019-06-17:
{
  "polygonList": [
    "0,0,1371,0,1371,692,0,692,0,0"
  ]
}

bryans69, 2019-06-17:
[image]

MelbourneDeveloper, 2019-06-17:
@bryans69 yep, I understand. As mentioned above, the issue is because the buffer sizes are not being set when the device is initialized. I recommend that you clone the repo, and modify the code so that you can set the buffer sizes. 

If this solves your issue, I will put that in a release and push out the NuGet package.

MelbourneDeveloper, 2019-06-17:
@bryans69 cloning the repo and adding the project to your solution will help with debugging a lot.

bryans69, 2019-06-17:
OK, will do that tomorrow. Any suggestions on what the sizes should be?

MelbourneDeveloper, 2019-06-17:
@bryans69 they are usually 64. But, sometimes they can be 5000 I think.

Something is going wrong with the device though, because [this call](https://github.com/MelbourneDeveloper/Device.Net/blob/3a7324746e01a0e252d2a4d1b630ed4b623f3903/src/Hid.Net/Windows/HidAPICalls.cs#L60) used [here](https://github.com/MelbourneDeveloper/Device.Net/blob/3a7324746e01a0e252d2a4d1b630ed4b623f3903/src/Hid.Net/Windows/WindowsHidDeviceFactory.cs#L51) should obtain those values.

MelbourneDeveloper, 2019-06-17:
@bryans69 please stay in contact and I'll do my best to help you through this.

bryans69, 2019-06-17:
OK, I’ll check those calls tomorrow and let you know what happens

MelbourneDeveloper, 2019-06-17:
@bryans69 

> Any suggestions on what the sizes should be?

You will need to check with the manufacturer exactly the device is expecting to receive. These values are the size of amount of data that is written to/from the device. Only they will be able to tell you what that is.

bryans69, 2019-06-18:
I manually changed the WriteBufferSize just to test, and as you suspected, InitializeAsync then worked. However, I cannot write using WriteAndReadAsync. I get no errors, but using this https://freeusbanalyzer.com/ I can see that nothing is being sent to the device (I have an application from the manufacturer that does communicate fine, and can see that traffic)

When I check WMI for USBHub, I can see that I actually have two devices for that ID, one with MI_01

Device ID: USB\VID_043D&PID_0235&MI_01\6&2F016DBC&0&0001, PNP Device ID: USB\VID_043D&PID_0235&MI_01\6&2F016DBC&0&0001, Description: USB Printing Support
Device ID: USB\ROOT_HUB30\4&525CFB&0&0, PNP Device ID: USB\ROOT_HUB30\4&525CFB&0&0, Description: USB Root Hub (USB 3.0)
Device ID: USB\VID_043D&PID_0235\75274494501H1, PNP Device ID: USB\VID_043D&PID_0235\75274494501H1, Description: USB Composite Device
Device ID: USB\VID_04B8&PID_0007\4C43504B3032383441, PNP Device ID: USB\VID_04B8&PID_0007\4C43504B3032383441, Description: USB Printing Support
Device ID: USB\VID_05E3&PID_0608\5&31A63465&0&10, PNP Device ID: USB\VID_05E3&PID_0608\5&31A63465&0&10, Description: Generic USB Hub

Could it be I’m trying to talk to the wrong one? As you can see from the earlier screenshot, DeviceManager.Current.GetDevicesAsync only returns one device.

I’m communicating with the manufacturer as well, but not sure at this point if its an API issue, or me not understanding enough about USB

MelbourneDeveloper, 2019-06-18:
@bryans69 

Cool. Any chance you can fork my repo, your changes in, and send me the link? I won't be able to look for a couple of days unfortunately. But I will certainly do my best to diagnose.

bryans69, 2019-06-18:
I just changed this where you suggested. I could get the setter to work, and as I was just trying to prove a point, hard coded - WindowsHIDDevice,cs
     public override ushort WriteBufferSize => 64;// ConnectedDeviceDefinition == null ? (ushort)0 : (ushort)ConnectedDeviceDefinition.WriteBufferSize.Value;

Can't help thinking I'm connecting to the wrong device instance (if that is the correct terminology for USB devices), hence it effectively being read only

MelbourneDeveloper, 2019-06-18:
@bryans69  Yeah, but you have written some code to connect to your device and send data. Can you post that?

bryans69, 2019-06-18:
very quick and dirty
 private async void button2_Click(object sender, EventArgs e)
        {
            var deviceDefinitions = new List<FilterDeviceDefinition>
    {
        new FilterDeviceDefinition{ DeviceType= DeviceType.Hid, VendorId= 0x043D, ProductId=0x0235 }
    };
            var devices = await DeviceManager.Current.GetDevicesAsync(deviceDefinitions);
            var device =new WindowsHidDevice( devices.FirstOrDefault().DeviceId);
            await device.InitializeAsync();
            //Create a buffer with 3 bytes (initialize)
            var buffer = new byte[10];
            buffer[0] = 0xA5;
            buffer[1] = 0x00;
            buffer[2] = 0x07;
            buffer[3] = 0x50;
            buffer[4] = 0xE0;
            buffer[5] = 0x21;
            buffer[6] = 0x01;
            buffer[7] = 0x00;
            buffer[8] = 0x00;
            buffer[9] = 0x00;
            //Write and read the data to the device
            var readBuffer = await device.WriteAndReadAsync(buffer);
            if (readBuffer.Length>0)
            {
                Console.WriteLine(readBuffer.Length);
            }
        }

bryans69, 2019-06-18:
This is actually the command I'm trying to send / get a response from

MelbourneDeveloper, 2019-06-18:
@bryans69 I can't see anything funny there. Maybe the device has a Hid and USB device registered. Have you tried seeing if you can send data using the USB Guid?

Try registering the USB factory and then enumerating all devices on the machine. you might find that it shows up as a USB device.

bryans69, 2019-06-18:
Strangely, that still only returns one device. So not seeing the USB ones at all
[image]

MelbourneDeveloper, 2019-06-18:
@bryans69 do you know for sure that freeusbanalyzer can sniff Hid traffic?

MelbourneDeveloper, 2019-06-18:
@bryans69 I mean you can step right in to the code with F11 to see the literal API calls that are sending the data...

bryans69, 2019-06-18:
This is what I see from the code I posted
[image]
and this is what I see from the working application
[image]

Hope that makes sense

I stepped into your code and  await _WriteFileStream.WriteAsync(bytes, 0, bytes.Length); in WindowsHIDDevices appears to work, ie it doesn't error, but I don't see it in the above, and never get a reply

bryans69, 2019-06-18:
Just to clarify, the last entry in the top screenshot is the InitializeAsync call

MelbourneDeveloper, 2019-06-18:
@bryans69 Odd.

I'll spend some time thinking about it and send you some more things to try. This looks like a good tool, so I'll try it out and see what the other devices I test with show. Sorry, just gonna have to bare with me over the next couple of days.

bryans69, 2019-06-18:
OK thanks. Will let you know if I have any more success

MelbourneDeveloper, 2019-06-23:
@bryans69 I have pushed out a new release. The main focus here was error handling and logging around connecting to and enumerating devices. However, I've added a constructor for Windows Hid Devices to accept read/write buffer sizes. I hope this helps you.

MelbourneDeveloper, 2019-06-23:
@bryans69 , I have made tracing the data IO the highest priority for the next release. 

https://github.com/MelbourneDeveloper/Device.Net/projects/8

I hope to get this done soon.

MelbourneDeveloper, 2019-06-23:
@bryans69 I've made a branch where I've fixed up tracing. If you grab this, I'd say that tracing is pretty reliably here:

https://github.com/MelbourneDeveloper/Device.Net/tree/%237-Tracing

As long as you pass a tracer in to the constructor, it will trace IO for you. This should pretty much guarantee that the data is reaching the device. If [FileStream.WriteAsync()](https://github.com/MelbourneDeveloper/Device.Net/blob/75ddf460238cf5b8d79868ef9e73b162bfdbbed7/src/Hid.Net/Windows/WindowsHIDDevice.cs#L240) is not sending the data, and not throwing an exception, I'd be extremely surprised.

bryans69, 2019-06-25:
Thanks

I’ll pull this down and take a look, although won’t be until next week now

One thing I have discovered is that there appears to be two pipes, one for reading and one for writing

\\?\usb#vid_043d&pid_0235&mi_01#6&2f016dbc&0&0001#{28d78fad-5a12-11d1-ae5b-0000f803a8c2}\PIPE00
usb#vid_043d&pid_0235&mi_01#6&2f016dbc&0&0001#{28d78fad-5a12-11d1-ae5b-0000f803a8c2}\PIPE01

However, using WindowsHidDevice or WindowsUSBDevice constructors gives 0 values for both read and write buffers

https://github.com/MelbourneDeveloper/Device.Net/tree/%237-Tracing

MelbourneDeveloper, 2019-06-28:
> One thing I have discovered is that there appears to be two pipes, one for reading and one for writing

@bryans69 you're definitely dealing with a USB device there and not a Hid device.

I only changed the Hid equivalent because I thought you were dealing with a Hid device. So, the problem is [here](https://github.com/MelbourneDeveloper/Device.Net/blob/a63ed3781b16f18dbefed13c6f0c9215377cceaa/src/Usb.Net/Windows/WindowsUsbDevice.cs#L23) and [here](https://github.com/MelbourneDeveloper/Device.Net/blob/a63ed3781b16f18dbefed13c6f0c9215377cceaa/src/Usb.Net/Windows/WindowsUsbDevice.cs#L176)

MelbourneDeveloper, 2019-06-30:
@bryans69 I have added an overload in the Develop branch for USB:

https://github.com/MelbourneDeveloper/Device.Net/blob/f75d0856eccb8043b875ad5ec62f5373732e053e/src/Usb.Net/Windows/WindowsUsbDevice.cs#L35

MelbourneDeveloper, 2019-07-02:
@bryans69 

> One thing I have discovered is that there appears to be two pipes, one for reading and one for writing

Perhaps you're experiencing this problem? Would that be right? Someone else commented on this and maybe it needs a fix.
https://github.com/MelbourneDeveloper/Device.Net/issues/19

bryans69, 2019-07-02:
Hi

Looks like it could be. Not going to get a chance to look at it this week, but will try and pull down and investigate next week

#19<https://github.com/MelbourneDeveloper/Device.Net/issues/19>

MelbourneDeveloper, 2019-07-03:
@bryans69 there's lots of work been done around the pipes problem here: https://github.com/MelbourneDeveloper/Device.Net/issues/19 . I think that branch should fix the issue and I will prep a nuget for it soon.

MelbourneDeveloper, 2019-07-21:
@bryans69 I'm closing this particular issue because I believe that all your issues will be fixed in Version 3.0, or you will at least be able to work around them. I will release 3.0 beta today. It would be great if you could grab it and test.

closed 2019-07-21 by MelbourneDeveloper