MelbourneDeveloper/Device.Net #64: Can enumerate HID, but not USB

vixez, 2019-06-19:
Hello,

I'm trying to enumerate all USB devices as a test. The end goals is to detect a WiFi stick being (dis)connected. I can enumerate all HID devices, but no USB devices are showing.

I'm using the following code:

` WindowsHidDeviceFactory.Register();
            WindowsUsbDeviceFactory.Register();
            var devices = await DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(null);
            Console.WriteLine("Currently connected devices: ");
            foreach (var device in devices)
            {
                Console.WriteLine(device.DeviceId);
            }
            Console.WriteLine();`

Am I missing something?

Thanks

MelbourneDeveloper, 2019-06-19:
@vixez 

That code looks correct to me. A lot of people have been reporting issues with enumerating devices. I have put some fixes in the Develop branch to capture some logging around what is going wrong. I will do a release of this soon. Meantime, I recommend cloning the Develop branch and running the Windows sample. You should see the device there.

This is the Windows sample.
https://github.com/MelbourneDeveloper/Device.Net/blob/develop/src/Usb.Net.WindowsSample/Program.cs

[Here](https://github.com/MelbourneDeveloper/Device.Net/blob/2d48dc2c7802fa0555d7b0cbd88aaac43344fa58/src/Usb.Net.WindowsSample/Program.cs#L30), the factories are registered with logging. This means that you will see information about what is going wrong in the debug output window. However, the easiest way to find out what is going wrong is to debug with first chance exceptions turned on.

MelbourneDeveloper, 2019-06-23:
@vixez 

I have pushed out a new release that has more error logging - especially around enumerating connected devices. I have also updated the wiki on how to enable logging. You might want to try it. It may give you information about why the device cannot be connected to. One thing that could be going wrong is that your USB device might not be initializing properly. If that's the case, then the logger will at least log an error message telling you that.

Also, see this for logging. Your code currently does not register any loggers.
https://github.com/MelbourneDeveloper/Device.Net/wiki/Enumerating-Connected-Devices

closed 2019-06-28 by MelbourneDeveloper