MelbourneDeveloper/Device.Net #67: Implement Reading and Writing N Bytes

realivanjx, 2019-06-30:
Hi,

Is it possible to implement reading and writing n bytes from/to a device more like a `Stream`? Something like this:
```
Read(byte[] buff, int offset, int read)
Write(byte[] buff, int offset, int len)
```

Thanks.

MelbourneDeveloper, 2019-06-30:
Can you give an example of the situation? 

I think the answer is that you have to break data up in to 64 bit chunks yourself but you can create a helper method to do that.

realivanjx, 2019-06-30:
Hmm so it is always 64 bytes? Im trying to transfer data from a stream (file, network, etc) to a device.

realivanjx, 2019-06-30:
For that kind of task i usually make a variable to hold the data from src stream like `byte[] readBuff = new byte[buffSize]` then `int read = stream.Read()` returns amount of data it receives (sometimes not == buffSize) so to make the code efficient i can just pass `read` to your api like this: `IDevice.Write(readBuff, read)`.

MelbourneDeveloper, 2019-06-30:
@realivanjx it's usually 64 bytes. 

You need to break the chunks up with your own code. You can see how I've done it with Trezor.net or Ledger.net

realivanjx, 2019-06-30:
Ok. Currently im doing it like this with linq:

```
                int written = 0;
                int writeBuffLen = device.ConnectedDeviceDefinition.WriteBufferSize ?? 64;

                while (written < read)
                {
                    toRead = Math.Min(writeBuffLen, read - written);
                    await device.WriteAsync(readBuff.Skip(written).Take(toRead).ToArray());
                    written += toRead;
                }
```

Just one problem. It is too slow. Im usually writing 100s of MB of data to a device (firmware flashing). With LibUsbDotNet i can write whatever amount of buffer i want (was 1 MB buffer size) and it flashes really quickly but then i got AccessViolationException and doesnt even lists devices on some pc. Any idea? Thanks.

MelbourneDeveloper, 2019-06-30:
@realivanjx we can't control how much data can be sent in one hit. My understanding is that most devices accept 64 bytes in one hit and some accept 5000. If the other library accepts more, it's because it is breaking up the chunks for you. 

What kind of device is it? Maybe I can get one and try to do the same thing. I can optimise the code if I can figure out what is slow.

realivanjx, 2019-06-30:
motorola device. im implementing fastboot into c# based on python-adb project on github (they use libusb)

MelbourneDeveloper, 2019-06-30:
Cool! 

Do you think you could share the source code?

realivanjx, 2019-06-30:
maybe when im done testing i could share it. it is in a big project that i couldnt share but i can still share the fastboot helper class.

MelbourneDeveloper, 2019-06-30:
Cool. I'd like to try to optimise if possible but I can't really do much unless I can see the code. 

Alternatively, you could modify the code and submit a PR

realivanjx, 2019-06-30:
Just one question. When i read from device and the device sends less than 64 bytes response will i get n bytes of array it responds or 64 bytes of array with zeroes to fill it?

MelbourneDeveloper, 2019-06-30:
@realivanjx it just returns whatever the USB call reads in to the buffer. You can see the code here:

https://github.com/MelbourneDeveloper/Device.Net/blob/a63ed3781b16f18dbefed13c6f0c9215377cceaa/src/Usb.Net/Windows/WindowsUsbDevice.cs#L109
It's calling WinUsb_ReadPipe

The write call is here:

https://github.com/MelbourneDeveloper/Device.Net/blob/a63ed3781b16f18dbefed13c6f0c9215377cceaa/src/Usb.Net/Windows/WindowsUsbDevice.cs#L126

WinUsb_WritePipe

There's no magic. It just makes those calls. If anything looks wrong, feel free to send a correction.

realivanjx, 2019-06-30:
Ok i will try it later. Thanks...

NikolayUvarov, 2019-06-30:
> 
> 
> Can you give an example of the situation?
> 
> I think the answer is that you have to break data up in to 64 bit chunks yourself but you can create a helper method to do that.

My device can work with 32 and 16 bytes lenths, so I think it is a good question.

realivanjx, 2019-06-30:
I think it is possible to call `WinUsb_WritePipe` with bytes larger than `WriteBufferSize`. Mind trying it? Thanks.

realivanjx, 2019-06-30:
I think we can put buffer size more than `WriteBufferSize` property.
https://docs.microsoft.com/en-us/windows/desktop/api/winusb/nf-winusb-winusb_writepipe#remarks

MelbourneDeveloper, 2019-06-30:
@realivanjx good call! Thanks for pointing to the documentation on this. I will address in the next release. The doco says:

> - The buffer length must be a multiple of the maximum endpoint packet size.
> - The length must be less than or equal to the value of MAXIMUM_TRANSFER_SIZE retrieved by WinUsb_GetPipePolicy.

> There are no restrictions on the size of the buffer if RAW_IO is not set as the pipe's policy type. If the size of the buffer is greater than the maximum transfer length reported by MAXIMUM_TRANSFER_SIZE, WinUSB divides the request into smaller requests and submits them serially to the host controller. 

This class is probably not obtaining the correct max transfer value. I will go over the code and check where it is getting the value for MAXIMUM_TRANSFER_SIZE. But, I'm not sure that this will necessarily improve performance. The call will automatically divide up the buffer anyway. 

I will certainly investigate and fix any issues I find. Good work!

realivanjx, 2019-06-30:
Just asking what is the current pipe policy for this library?

MelbourneDeveloper, 2019-06-30:
@realivanjx I'm not sure if the code does call WinUsb_GetPipePolicy . It's currently getting the buffer sizes from the descriptor with [WinUsb_GetDescriptor](https://github.com/MelbourneDeveloper/Device.Net/blob/a63ed3781b16f18dbefed13c6f0c9215377cceaa/src/Usb.Net/Windows/WindowsUsbDevice.cs#L176). Maybe this is just wrong, but I need to investigate.

It would be really awesome if you could fix the problem and submit a pull request.

realivanjx, 2019-07-01:
@MelbourneDeveloper I have found a solution to my problem. I removed your `WriteBufferSize` check in `WriteAsync()` inside `WindowsUsbDevice.cs` and i just give 1 MB of byte array as buffer, it now flashes as fast as it can. The api either doesnt care abt the size or it splits the packet directly in `winusb.dll` so it is much faster now. Mind if i make a PR about this? If so how can i do that? I never make any PR before. Thanks :)

MelbourneDeveloper, 2019-07-01:
Fantastic! 

The code is wrong, so I will need to adjust quite a few things. I can't guarantee that the PR will be accepted exactly as is, but I can guarantee that the idea behind your PR will be included in the next release. 

Thanks! Please send the PR.

realivanjx, 2019-07-01:
ok then. for now i will just use the debug version of it not really a big deal.

realivanjx, 2019-07-01:
i checked out to a new branch and trying to push to here but i got permission denied

MelbourneDeveloper, 2019-07-02:
https://github.com/MelbourneDeveloper/Device.Net/blob/dee398c6ef03cd4283c50eaa7a6b7d64fa16cbaa/src/Usb.Net/Windows/WindowsUsbDevice.cs#L35

@realivanjx does the above solve your problem?

mxjones, 2019-07-04:
@MelbourneDeveloper Hey, I remembered reading this whilst I was playing with the other work in #19 - as I was testing I found something in relation to this...

> @realivanjx I'm not sure if the code does call WinUsb_GetPipePolicy . It's currently getting the buffer sizes from the descriptor with [...]

The Read & Write buffer size for my device from the descriptor is 64, which didn't seem right to me (same as reported here) - after looking a bit more it seems like the correct values are being returned for each specific endpoint by the WinUsb_QueryPipe call (WINUSB_PIPE_INFORMATION.MaximumPacketSize) - Not sure if this is only for the device I'm using though.

MelbourneDeveloper, 2019-07-04:
@mxjones yes this does look like a bug but it can be worked around right now by manually specifying the lengths. It's just not getting the correct default value.

mxjones, 2019-07-04:
Yeah as you said, this is not affecting me as I'm just manually specifying the lengths :)

MelbourneDeveloper, 2019-07-07:
Please follow #72, and #75

closed 2019-07-07 by MelbourneDeveloper