MelbourneDeveloper/Device.Net #68: Typo on documentation

NikolayUvarov, 2019-06-30:
In wiki code example:

public async Task InitializeTrezorAsync()
        {
            //Get the first available device and connect to it
            var devices = await DeviceManager.Current.GetDevices(_DeviceDefinitions);
            TrezorDevice = devices.FirstOrDefault();
            await TrezorDevice.InitializeAsync();
        }

Function GetDevices is inimplemented. There is function GetDevicesAsync

MelbourneDeveloper, 2019-06-30:
@NikolayUvarov thanks for pointing that out. Which page is that on?

NikolayUvarov, 2019-06-30:
Sorry, my fault. Here it is:
https://github.com/MelbourneDeveloper/Device.Net/wiki/Device-Listener

MelbourneDeveloper, 2019-06-30:
@NikolayUvarov I've fixed this up in a a few places. Thanks a lot.

closed 2019-06-30 by MelbourneDeveloper