MelbourneDeveloper/Device.Net #69: My code example with treads and questions

NikolayUvarov, 2019-06-30:
I want to say that is longwaited-by-me library. And works great. 

Here is my example of using the library. I have to threads, one for sending controls to my device, and the other for reading. All work fine except I do not understand how to 
handle connect and disconnect events and get user-friendly error message in exceptions.

It would be handy if we could:
- register device pid and vid eg
var myDefinition = new FilterDeviceDefinition{ DeviceType= DeviceType.Hid, VendorId= 0x0468, ProductId=0xf242 };
var device = await DeviceManager.Current.GetDevicesAsync(deviceDefinitions);
- be able to add listener or callback to call onConnect(), onDisconnect() events on every device.
- when using list of devices with different pid&vid be able to get a structuture with pid, vid, etc of connectded device, like we specify in device filter, but with additional info. Now property myDev.deviceId contains unseparated string with desctription.
- need more clarifictation how to work with more than one connected device (with equial pid&vid or several devices, that match to filters).
- 'nullification' of disconnectd device. Now I do a workaround:
 catch(Exception e) {
                        readText = "*** Error while reading device! \n ";// + e.InnerException.ToString();
                        myHidDevice = null;
                    }
- when getting errors like "error while reading, error while writing, error device not connected" via exception be able to get code and/or textual defenition of the error.

namespace DeviceTerminal
{
    public partial class Form1 : Form
    {
        public Form1()
        {
            InitializeComponent();
            deviceReaderThread.Start();
            deviceSenderThread.Start();
        }
        ~Form1()
        {
            deviceSenderThread.Abort();
        }

        static UInt64 recv_c = 0;
        static UInt64 sent_c = 0;

        //var myHidDevice;

        public static void DeviceSenderThread()
        {
            while (true)
            {
                Thread.Sleep(200);

                var buffer = new byte[64];
                //buffer[0] = Convert.ToByte('S'); buffer[1] = Convert.ToByte('\n'); buffer[2] = 0;
                buffer[0] = Convert.ToByte('S'); buffer[1] = Convert.ToByte('1'); buffer[2] = Convert.ToByte('\n'); buffer[3] = 0;

                if (myHidDevice != null) {
                    sent_c++;
                    myHidDevice.WriteAsync(buffer); } 
                else {
                    Thread.Sleep(100);
                }
            }
        }
        public static async void DeviceReaderThread()
        {
            string readText = "";
            while (true)
            {
                if (myHidDevice != null) { 
                    recv_c++;
                    try
                    {
                        var readBuffer = await myHidDevice.ReadAsync();
                        readText = Encoding.UTF8.GetString(readBuffer, 0, readBuffer.Length);
                    }
                    catch(Exception e) {
                        readText = "*** Error while reading device! \n ";// + e.InnerException.ToString();
                        myHidDevice = null;
                    }
                    Program.form1.textBox1_log.Invoke(new Action(() => Program.form1.textBox1_log.AppendText(readText)));
                }
                else
                {
                    Thread.Sleep(100);
                }
            }
        }
        Thread deviceSenderThread = new Thread(new ThreadStart(DeviceSenderThread));
        Thread deviceReaderThread = new Thread(new ThreadStart(DeviceReaderThread));

        UInt64 prev_recv_c = 0;
        UInt64 prev_sent_c = 0;
        private static IDevice myHidDevice;

        private void timer1sec_Tick(object sender, EventArgs e)
        {
            label_read_count.Text = Convert.ToString(recv_c - prev_recv_c); prev_recv_c = recv_c; //recv_c = 0;
            label_sent_count.Text = Convert.ToString(sent_c - prev_sent_c); prev_sent_c = sent_c; //sent_c = 0;
            label_total_recv.Text = Convert.ToString(recv_c);
            label_total_sent.Text = Convert.ToString(sent_c);

            if (myHidDevice != null)
                label_deviceid.Text = myHidDevice.DeviceId;
            else
                label_deviceid.Text = "None";
        }

        private void button2_Click(object sender, EventArgs e)
        {
            if (!deviceReaderThread.IsAlive) { deviceReaderThread.Start(); }
            if (!deviceSenderThread.IsAlive) { deviceSenderThread.Start(); }
        }

        private void Form1_FormClosing(object sender, FormClosingEventArgs e)
        {
            if (deviceSenderThread != null && deviceSenderThread.IsAlive)
            {
                deviceSenderThread.Abort();
            }
            if (deviceReaderThread != null && deviceReaderThread.IsAlive)
            {
                deviceReaderThread.Abort();
            }
        }
        private static async Task InitializeMyHid()
        {
            WindowsHidDeviceFactory.Register();

            //Define the types of devices to search for. This particular device can be connected to via USB, or Hid
            var deviceDefinitions = new List<FilterDeviceDefinition>
            {
                new FilterDeviceDefinition{ DeviceType= DeviceType.Hid, VendorId= 0x0468, ProductId=0xf242 },
            };

            //Get the first available device and connect to it
            var devices = await DeviceManager.Current.GetDevicesAsync(deviceDefinitions);
            myHidDevice = devices.FirstOrDefault();
            await myHidDevice.InitializeAsync();

        }
        private async void button1_Click_1Async(object sender, EventArgs e)
        {
            try
            {
                await InitializeMyHid();
            }
            catch (Exception exa) {
                Program.form1.textBox1_log.Invoke(new Action(() => Program.form1.textBox1_log.AppendText("*** Error initializing device! \n")));
            }
        }

        private void button2_Click_1(object sender, EventArgs e)
        {
            deviceReaderThread.Suspend();
            deviceSenderThread.Suspend();
        }

        private void Form1_Load(object sender, EventArgs e)
        {

        }
    }
 

}

MelbourneDeveloper, 2019-06-30:
@NikolayUvarov thanks for posting your code. So, if I understand you correctly, you would like me to review your code and answer your questions, is that right? I an do that. But, I need to investigate a bit first.

NikolayUvarov, 2019-07-01:
I have put code for reference to my questions. If you found thease ideas
worth to implement please let me know.

MelbourneDeveloper, 2019-07-02:
@NikolayUvarov without going too deep in to the code. Here are some answers to your questions:

> register device pid and vid 

I think what you want to use is a DeviceListener. Here is an example of using the DeviceListener

Registering pid and vid:
https://github.com/MelbourneDeveloper/Device.Net/blob/34bfa581dac17fd70381b5ef5a710bb508cfb6f2/src/Usb.Net.UWP.Sample/TrezorExample.cs#L18

Creating the listener
https://github.com/MelbourneDeveloper/Device.Net/blob/34bfa581dac17fd70381b5ef5a710bb508cfb6f2/src/Usb.Net.UWP.Sample/TrezorExample.cs#L40

Start the listener
https://github.com/MelbourneDeveloper/Device.Net/blob/34bfa581dac17fd70381b5ef5a710bb508cfb6f2/src/Usb.Net.UWP.Sample/TrezorExample.cs#L64

> be able to add listener or callback to call onConnect(), onDisconnect() events on every device.

This event occurs after the listener initializes the device:
https://github.com/MelbourneDeveloper/Device.Net/blob/34bfa581dac17fd70381b5ef5a710bb508cfb6f2/src/Usb.Net.UWP.Sample/TrezorExample.cs#L45

This occurs after the listener realises it's been disconnected:
https://github.com/MelbourneDeveloper/Device.Net/blob/34bfa581dac17fd70381b5ef5a710bb508cfb6f2/src/Usb.Net.UWP.Sample/TrezorExample.cs#L51

> when using list of devices with different pid&vid be able to get a structuture with pid, vid, etc of connectded device, like we specify in device filter, but with additional info. Now property myDev.deviceId contains unseparated string with desctription.

You need to keep track of the devices that have been connected in the events above.

> need more clarifictation how to work with more than one connected device (with equial pid&vid or several devices, that match to filters).

Save as above. Use the events.

> 'nullification' of disconnectd device. Now I do a workaround:

> when getting errors like "error while reading, error while writing, error device not connected

Exceptions and messages are a bit of a problem in Device.Net.  Please see this issue:
https://github.com/MelbourneDeveloper/Device.Net/issues/11

I will make it higher priority.

MelbourneDeveloper, 2020-11-21:
@NikolayUvarov

Hi. Version 4 is coming out soon. Is this issue still relevant?

MelbourneDeveloper, 2020-12-13:
@NikolayUvarov Hi. I'm closing this issue.

If you still have issues with version 4.0.2-alpha, please report them here.

closed 2020-12-13 by MelbourneDeveloper