MelbourneDeveloper/Device.Net #71: Code, which works in Win7 gets exception in Win10 (standart app)

NikolayUvarov, 2019-07-01:
I have simple test form with a textfield and button, visual studio community 2017.
This code works perfect in windows7, but the same code in windows10 device is no.
I am getting error in line 
line             await myHidDevice.WriteAsync(buffer);
showing System.NullReverenceException in
    DeviceTerminal.Form1.<InitializeMyHid>d__4.MoveNext() 
This code is copy from the example.

`

using System;
using System.Collections.Generic;
using System.ComponentModel;
using System.Data;
using System.Drawing;
using System.Linq;
using System.Text;
```
using System.Threading;
using System.Threading.Tasks;
using System.Windows.Forms;
using Device.Net;
using Hid;
using Hid.Net;
using Hid.Net.Windows;

namespace DeviceTerminal
{
    
    public partial class Form1 : Form
    {
        public Form1()
        {
            InitializeComponent();
        }

        private static IDevice myHidDevice;

        private static async void InitializeMyHid()
        {
            WindowsHidDeviceFactory.Register();

            //Define the types of devices to search for. This particular device can be connected to via USB, or Hid
            var deviceDefinitions = new List<FilterDeviceDefinition>
            {
                new FilterDeviceDefinition{ DeviceType= DeviceType.Hid, VendorId= 0x0418, ProductId=0xf212 },
            };

            //Get the first available device and connect to it
            var devices = await DeviceManager.Current.GetDevicesAsync(deviceDefinitions);
            myHidDevice = devices.FirstOrDefault();
            await myHidDevice.InitializeAsync();
            
            //Create a buffer with 3 bytes (initialize)
            var buffer = new byte[64];
            buffer[0] = Convert.ToByte('R');
            buffer[1] = Convert.ToByte('\n');
            buffer[2] = 0;

            //Write the data to the device
            await myHidDevice.WriteAsync(buffer);

            //Read the response
            var readBuffer = await myHidDevice.ReadAsync();

            string readText = Encoding.UTF8.GetString(readBuffer, 0, readBuffer.Length);

            Program.form1.textBox1_log.Invoke(new Action(() => Program.form1.textBox1_log.AppendText("Got string: " + readText)));
        }
        private async void button1_Click_1Async(object sender, EventArgs e)
        {
            InitializeMyHid();
        }

    }
}
`

MelbourneDeveloper, 2019-07-01:
@NikolayUvarov thanks for posting. Can you please turn on break on all exceptions, and show me a screenshot of where the code is breaking? Also the stack trace.

NikolayUvarov, 2019-07-03:
Here the small project with test code that works in win7 but fails in win10.
I see that after call
var devices = await DeviceManager.Current.GetDevicesAsync(deviceDefinitions);
device|count = 0.

So the problem is not in exception itself, but in silently getting no devices in win10.

System.NullReferenceException
  HResult=0x80004003
  Message = Null reference.
  Source = DeviceTerminal
  Stack trace:
   в DeviceTerminal.Form1.<InitializeMyHid>d__4.MoveNext() in C:\common_virtual\oscillograph\DeviceTerminal\Form1.cs:line 49

[hid_dbg_project.zip](https://github.com/MelbourneDeveloper/Device.Net/files/3353278/hid_dbg_project.zip)

NikolayUvarov, 2019-07-03:
WindowsHidDeviceFactory.Register();
I have just found that code below does not show my device in win10. I think that is the problem. 

            var devices1 = await DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(null);
            Console.WriteLine("Currently connected devices: ");
            foreach (var device in devices1)
            {
                //Console.WriteLine(device.DeviceId);
                Program.form1.textBox1_log.Invoke(new Action(() => Program.form1.textBox1_log.AppendText("\r\nGot string:\r\n" + device.DeviceId)));

            }

NikolayUvarov, 2019-07-03:
BTW, I've just checked that is not security issue of win10 - device hid lister shows about 15 devices, but the code above only two.

MelbourneDeveloper, 2019-07-03:
@NikolayUvarov please register a Logger in the WindowsHidDeviceFactory.Register() method.

Watch the debug output window and see what turns up.

Also please include the screenshots etc.

MelbourneDeveloper, 2019-07-03:
@NikolayUvarov this is how you register a logger:

https://github.com/MelbourneDeveloper/Device.Net/blob/dee398c6ef03cd4283c50eaa7a6b7d64fa16cbaa/src/Usb.Net.WindowsSample/Program.cs#L30

MelbourneDeveloper, 2019-07-07:
@NikolayUvarov any progress?

Have you tried cloning this repo, and debugging the libraries here?

NikolayUvarov, 2019-07-08:
Currently I have added logger but I do not see any additional error
messages. I have noticed, that example which displaying list of usb devises
shows only two of them. I fuguered out that is not security issue because I
can access device via an vb app on win10.

My c# app gets no errors except it simply does not see a range of usb hid
devices under win10. And on win7 all work just perfect.

MelbourneDeveloper, 2019-07-08:
@NikolayUvarov it's probably different API behaviour between windows 7 and 10. 

You will need to debug. Please clone the repo , turn on break on all exceptions and debug to see what happens.

MelbourneDeveloper, 2019-07-14:
@NikolayUvarov , sorry about the short reply last time. I want to help you get this working. I am just very busy trying to get the changes done for [3.0](https://github.com/MelbourneDeveloper/Device.Net/projects/8) . Here is what I need you to do so I can diagnose the problem:

- Use Windows 10
- Clone this repo
- In your project, drop all the NuGet package references for this repo
- Add the projects from this repo that you need one by one to your solution (instead of the NuGet packages)
- Turn on break on all exceptions. I have written a guide just for you [here](https://christianfindlay.com/2019/07/14/visual-studio-break-on-all-exceptions/).
- Run the app
- Take screenshots, and copy the ToString() of any Exceptions you see

NikolayUvarov, 2019-07-15:
Thank you very much! Curriently I am abroad, will be able to check all
stuff this weekend (or maximum beginning of the next week).

aurirjr, 2019-07-17:
I am also getting Device count 0. Doing as @MelbourneDeveloper said:

```
System.Exception
  HResult=0x80131500
  Message=Could not get Hid Attributes (Call HidD_GetAttributes). Error code: 6
  Source=Device.Net
  StackTrace:
   at Device.Net.Windows.WindowsDeviceBase.HandleError(Boolean isSuccess, String message) in S:\temp_me_delete\Device.Net-master\src\Device.Net\Windows\WindowsDeviceBase.cs:line 36
```

MelbourneDeveloper, 2019-07-17:
@aurirjr some of the information is chopped off there. Please follow the steps for breaking on all exceptions here: https://christianfindlay.com/2019/07/14/visual-studio-break-on-all-exceptions/ and then wind back the call stack by one frame when you hit the exception. Then take a screenshot and post it here.

However, this is a pretty common API error. It just means that the library tried to get the details for a device, but the API call failed. The device probably cannot be queried.

aurirjr, 2019-07-18:
Sry It was chopped off. I will redo the procedure. 

I was analyzing yesterday and APICalls.CreateFile returned a SafeFileHandle just fine, with a valid Handle. The error code 87 was returned by HidD_GetManufacturerString. 

The device is an HID-compliant mouse, and It probably cannot be queried in Windows 10 for security reasons. With HIDSharp, TryOpen also doesn't work, as this case https://forum.zer7.com/topic/10122/ . I don't know if there is a workaround.

NikolayUvarov, 2019-07-18:
I noticed, that in the example provided by MD I see the full list of usb
devices.
When I use it in my app I am getting just empty list.

But for some reason I cannot get the full device list in my App.

Our goal to understand why such a difference.

I do not think that our issue is not security issue because I can easily
interact with same device via another application.

MelbourneDeveloper, 2019-07-20:
> I do not think that our issue is not security issue because I can easily
> interact with same device via another application.

It's possible that there are multiple device ids in the system. For example, there may be a USB Id, and a Hid Id. It's possible that the Hid Id fails while the USB one works fine. 

Do you see any USB device Ids? 

Usb.Net has some issues with USB right now. I'm working hard to fix those and version 3.0 should be much more stable with USB.

MelbourneDeveloper, 2019-07-20:
@aurirjr some devices just fail for some reason. If you find any library that is able to connect to it, please let me know.

MelbourneDeveloper, 2019-08-19:
@NikolayUvarov this issue will be closed soon. Please write back if you'd like to continue the discussion.

NikolayUvarov, 2019-08-20:
so I think we should wait for version 3 and then check in on win7/10

MelbourneDeveloper, 2019-08-20:
@nikolayuvarov V3 has been released. Are you still having the same issue?

Please see this article to help diagnose the issue
https://github.com/MelbourneDeveloper/Device.Net/wiki/Debugging,-Logging,-and-Tracing

NikolayUvarov, 2019-08-20:
I did not knew that, thanks!
I will report if any and if not also.
Thanks again!

NikolayUvarov, 2019-08-30:
Version 3.0.0 works fine in Win10! Thanks! The issue when I could see my
device in win7 and can't in win10 is now resolved.

вт, 20 авг. 2019 г. в 09:33, Nikolay Uvarov <nikolay.uvarov@gmail.com>:

MelbourneDeveloper, 2019-08-30:
@NikolayUvarov Awesome!

closed 2019-08-30 by MelbourneDeveloper