MelbourneDeveloper/Device.Net #80: Quick start example doesn't work (Hid.Net.Windows.WindowsHidException: WriteBufferSize must be specified)

KolesnichenkoDS, 2019-07-14:
I'm trying to rewrite a quick start example to read data from my USB mouse:

```csharp
WindowsUsbDeviceFactory.Register();
WindowsHidDeviceFactory.Register();

var definitions = new List<FilterDeviceDefinition>
{
    new FilterDeviceDefinition
    {
        DeviceType = DeviceType.Hid,
        VendorId = 0x18F8,
        ProductId = 0x0F99,
    }
};

var devices = await DeviceManager.Current.GetDevicesAsync(definitions);
var device = devices.FirstOrDefault();
if (device == null)
{
    Console.WriteLine("No connected devices");
    return;
}
await device.InitializeAsync();
```

The last line (`InitializeAsync`) throws the following exception:

```
Unhandled Exception: Hid.Net.Windows.WindowsHidException: WriteBufferSize must be specified. HidD_GetAttributes may have failed or returned an OutputReportByteLength of 0. Please specify this argument in the constructor. Note: Hid devices are always opened in write
 mode. If you need to open in read mode, please log an issue here: https://github.com/MelbourneDeveloper/Device.Net/issues
```

Not sure where I should specialize `WriteBufferSize` as I don't call `WindowsHidDevice` constructor explicitly anywhere.

KolesnichenkoDS, 2019-07-14:
Actually when I connect my mouse I see three devices:

```
\\?\hid#vid_18f8&pid_0f99&mi_01&col02#7&20bdfb95&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
\\?\hid#vid_18f8&pid_0f99&mi_01&col03#7&20bdfb95&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
\\?\hid#vid_18f8&pid_0f99&mi_01&col04#7&20bdfb95&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}
```

I've tried to initialize all of them, the last one fails with different exception:

```
Unhandled Exception: Hid.Net.Windows.WindowsHidException: ReadBufferSize must be specified. HidD_GetAttributes may have failed or returned an InputReportByteLength of 0. Please specify this argument in the constructor
```

MelbourneDeveloper, 2019-07-14:
@kolesnichenkods thanks for reporting. 

Sometimes buffer sizes are not picked up correctly. In those cases you need to construct the device manually . That will allow you to specify the buffer sizes with the device id. 

As for opening in read mode, I can see why that would be necessary so I will tackle that in 3.1 or 3.2. 

If you have any ideas on how the factory can construct the device without getting the buffer size automatically, let me know... Not really sure how that could work...

MelbourneDeveloper, 2019-07-21:
@KolesnichenkoDS the bug for automatically setting the buffer size is here: https://github.com/MelbourneDeveloper/Device.Net/issues/72

It will be fixed in 3.1. I'm releasing 3.0 beta today.

MelbourneDeveloper, 2019-08-19:
@AlcoholicParanoid there has been a lot of work around this in the V3 Beta 2 version. You can now manually specify the buffer size. Does this fix your issue?