MelbourneDeveloper/Device.Net #82: There were no devices

LuisAndres, 2019-07-18:
I added my USB device \\?\usb#vid_0e8d&pid_201d&mi_01#6&205a1a14&0&0001#{f72fe0d4-cbcb-407d-8814-9ed673d0dd6b} to the FilterDeviceDefinition list but in this part of code:
 var devices = await DeviceManager.Current.GetDevicesAsync(_DeviceDefinitions);            TrezorDevice = devices.FirstOrDefault();
Throw me "There were no devices found".

MelbourneDeveloper, 2019-07-21:
@LuisAndres Hi. Thanks for reporting. 

This this code

```cs
WindowsUsbDeviceFactory.Register(Logger);
WindowsHidDeviceFactory.Register(Logger);

var devices = await DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(null);
Console.WriteLine("Currently connected devices: ");
foreach (var device in devices)
{
    Console.WriteLine(device.DeviceId);
}
Console.WriteLine();
```
From here:
https://github.com/MelbourneDeveloper/Device.Net/wiki/Enumerating-Connected-Devices

Also, please post your _DeviceDefinitions code. Also, 3.0 is about to be released and this will have a lot better logging. Once, it's released, I recommend watching the log to see what is coming out. If the filtered device is not being found, and there is no log, I will consider that a bug and fix it.

schmitch, 2019-07-26:
Actually I'm using 3.0-beta and tried to enumerate over all devices via:
```
#if Windows
            WindowsUsbDeviceFactory.Register(null, null);
            WindowsHidDeviceFactory.Register(null, null);
#else
            LibUsbUsbDeviceFactory.Register(null, null);
#endif

            var devices = await DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(null);
            Console.WriteLine("Currently connected devices: ");
            foreach (var device in devices)
            {
                Console.WriteLine($"{device.DeviceId} - {device.Label} - {device.Manufacturer} -  {device.ProductName}");
            }
            Console.WriteLine();
```

Unfortunatly nothing is printed at all. 
However Windows has detected a single Device (TANDBERG RDX USB) which is a RDX USB drive. I've connected it trough Vmware ESXi `USBSTOR\Disk&Ven_TANDBERG&Prod_RDX&Rev_0205\007360421908&0 `

MelbourneDeveloper, 2019-07-26:
@schmitch can I TeamViewer on to your computer in an hour or two?

I can't do support for libusb though...

schmitch, 2019-07-26:
sadly I'm leaving early today.
But I can setup a test environment on monday (or some day next week)

srganeshram, 2019-07-27:
> Actually I'm using 3.0-beta and tried to enumerate over all devices via:
> #if Windows
>             WindowsUsbDeviceFactory.Register(null, null);
>             WindowsHidDeviceFactory.Register(null, [...]

I have the same problem.

MelbourneDeveloper, 2019-07-27:
@srganeshram I noticed that nothing is being passed on to the register methods. Try putting in loggers and tracers and post the results here. 

Also, please turn on break on all exceptions and send a screenshot. 

https://christianfindlay.com/2019/07/14/visual-studio-break-on-all-exceptions/

MelbourneDeveloper, 2019-07-28:
@LuisAndres @schmitch can you please log a comment like this one:
https://github.com/MelbourneDeveloper/Device.Net/issues/88#issuecomment-515727257
@srganeshram

It gives me the detail I need to diagnose the problem.

schmitch, 2019-07-29:
@MelbourneDeveloper unfortunatly I'm on Mac and just using a vm without VS to do a build and then I deploy to a Windows Server.
But I created a Logger that I used:
```
class ConsoleLogger : ILogger
    {
        public void Log(string message, string region, Exception ex, LogLevel logLevel)
        {
            Console.WriteLine($"LOG: {region}-{logLevel}: {message} - {ex}");
        }
    }
```

```
C:\Users\Administrator\Desktop\publish>dotnet UsbDeviceCheck.dll
LOG: WindowsHidDeviceFactory-Error: GetDeviceDefinition error. Device Id: \\?\hid#vid_0e0f&pid_0003&mi_00#8&211f5361&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} - Device.Net.Exceptions.ApiException: Could not get Hid Attributes (Call HidD_GetAttributes). Error code: 6
   at Device.Net.Windows.WindowsDeviceBase.HandleError(Boolean isSuccess, String message)
   at Hid.Net.Windows.HidAPICalls.GetHidAttributes(SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId, SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId)
LOG: WindowsHidDeviceFactory-Warning: Device with path \\?\hid#vid_0e0f&pid_0003&mi_00#8&211f5361&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} was skipped. See previous logs. -
LOG: WindowsHidDeviceFactory-Error: GetDeviceDefinition error. Device Id: \\?\hid#vid_0e0f&pid_0003&mi_01#8&95bc71c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} - Device.Net.Exceptions.ApiException: Could not get Hid Attributes (Call HidD_GetAttributes). Error code: 6
   at Device.Net.Windows.WindowsDeviceBase.HandleError(Boolean isSuccess, String message)
   at Hid.Net.Windows.HidAPICalls.GetHidAttributes(SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId, SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId)
LOG: WindowsHidDeviceFactory-Warning: Device with path \\?\hid#vid_0e0f&pid_0003&mi_01#8&95bc71c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} was skipped. See previous logs. -
Currently connected devices:
```

MelbourneDeveloper, 2019-07-29:
@schmitch I think you are hitting this problem.

MelbourneDeveloper, 2019-07-29:
@schmitch I think you are hitting this problem: https://github.com/MelbourneDeveloper/Device.Net/issues/88#issuecomment-515735551

MelbourneDeveloper, 2019-07-29:
@schmitch Essentially, the library is finding the Device Id, and trying to connect, but the access permissions being requested are being rejected so the handle cannot be obtained. This is a sort of known issue... 

I recommend cloning the code and playing around with the access permissions until you can connect. If you can get it working, I will merge your code in through a pull request.

MelbourneDeveloper, 2019-07-29:
@schmitch the issue is logged here and will be fixed in 3.1
https://github.com/MelbourneDeveloper/Device.Net/issues/89

MelbourneDeveloper, 2019-07-29:
@schmitch please clone the develop branch.

MelbourneDeveloper, 2019-07-29:
@schmitch , sorry I should point out that it may have nothing to do with permissions. It could just be a problematic device. Sometimes there is both a USB Id, and a Hid Id. Have you tried connected to the same device with a USB prefix?

schmitch, 2019-07-29:
what do you mean with "Have you tried connected to the same device with a USB prefix?"?

actually I just "connected" the device and added made mounted it in windows (so that It appears in explorer as B:)

MelbourneDeveloper, 2019-07-29:
@schmitch  run the Windows  sample unaltered - exactly as is without changing anything, and please post a screenshot.

Lotharyx, 2019-08-06:
I'm having a similar problem, but I'm using the latest packages available via NuGet, which is 2.9.0 (because I really don't like having dependencies that VS can't automatically resolve; makes the solution much less portable).  I can create two different problems.  My code is a WPF application.  Here's some code:

```
   public MainWindow() {
         InitializeComponent();
         WindowsUsbDeviceFactory.Register(Logger);
         WindowsHidDeviceFactory.Register(Logger);
         var devices = DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(null);
         devices.Wait(); // This hangs forever
         foreach(var device in devices.Result) {
            lstHIDs.Items.Add(device.ProductName);
         }
      }

      private void btnEnumerateClick(Object sender, RoutedEventArgs args) {
         Enumerate();
      }

      private async Task Enumerate() {
         var devices = await DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(null);
         foreach(var device in devices) {
            lstHIDs.Items.Add(device.ProductName);
         }
      }
```
As one can see, there are two approaches to calling `DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(null)`.  In the first case, during the constructor of `MainWindow`, execution hangs forever on `devices.Wait()` (I cannot use `await` here because I cannot change the return type of a constructor).  In the second case, in the method `Enumerate`, a breakpoint set at the foreach is hit and `devices` is an empty list.  If I use the same approach as in the constructor (using the `Task.Wait()` method instead of the `await` call), again the `Wait()` never returns.  So I'm not sure that there isn't just something wrong with the way I'm approaching the asynchronous programming (this is a completely new paradigm for me).

Some things to note:
- There is no overload of `Register` that accepts two arguments in 2.9.0
- There are no messages from the logger in the output window
- I have break-on-exceptions turned on for all exceptions; none are thrown
- I can clone the latest, run the example modified to call `GetConnectedDeviceDefinitionsAsync` immediately inside the `Go` method, and it works fine (but I am reluctant to break the NuGet connection).

So is this something that is definitely broken in 2.9.0?  If my only recourse is to use a newer package than what is in NuGet, I will do it, but with a polite suggestion that a newer package be published to NuGet soon :-)

Lotharyx, 2019-08-06:
Quick update:
I removed the references to the NuGet-installed Device.Net, Hid.Net, and Usb.Net packages, and added references to the ones built in the Device.Net solution I cloned.  Now I do see messages in the debug output, and every device detected immediately throws `Device.Net.Exceptions.ApiException` with a message of "Could not get Hid string.  Error code 87."  
From the debug output:
```
--------------------------------------
Message: Found device \\?\hid#vid_046d&pid_c069#6&483d65f&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Time: 8/6/2019 3:45:09 PM
Section: WindowsHidDeviceFactory
Error: 
--------------------------------------
Exception thrown: 'Device.Net.Exceptions.ApiException' in Device.Net.dll
Could not get Hid string. Error code: 87

--------------------------------------
Message: GetDeviceDefinition error. Device Id: \\?\hid#vid_046d&pid_c069#6&483d65f&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Time: 8/6/2019 3:45:09 PM
Section: WindowsHidDeviceFactory
Error: Device.Net.Exceptions.ApiException: Could not get Hid string. Error code: 87
   at Device.Net.Windows.WindowsDeviceBase.HandleError(Boolean isSuccess, String message)
   at Hid.Net.Windows.HidAPICalls.GetHidString(SafeFileHandle safeFileHandle, GetString getString)
   at Hid.Net.Windows.HidAPICalls.GetSerialNumber(SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId, SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId)
--------------------------------------

```

Ultimately the list returned by `GetConnectedDeviceDefinitionsAsync` is still empty.

Since this seems to have diverged from the originally-reported problem, should I open a new issue?  Or seek help another way?  Thanks.

MelbourneDeveloper, 2019-08-06:
@Lotharyx thanks for posting so much detail. You've uncovered a bug. I've seen this one multiple times, but ti didn't click that it was an issue. The code is trying to get the manufacturer, serial, and product, but perhaps the device doesn't supply these. The code was throwing and error, but I believe I have fixed it in the Develop branch now. 

I logged the issue [here](https://github.com/MelbourneDeveloper/Device.Net/issues/97)

I believe [this fix](https://github.com/MelbourneDeveloper/Device.Net/commit/9a57cb0f5999c1fcd13a3c6d1b92536920035e62) will solve the issue.

Could you please try again with the develop branch? If it solves your problem, I will merge this in to a Beta 3 release.

Thanks for reporting. This should help a lot of people.

Lotharyx, 2019-08-06:
I will have to test tomorrow; our power went out at work and I don't have the necessary project files at home.  Thank you for responding so quickly.

MelbourneDeveloper, 2019-08-06:
@schmitch have you tried connecting with the latest code in the Develop branch? You might have more luck, and there will be better logging.

MelbourneDeveloper, 2019-08-06:
@Lotharyx happy to help. Every small fix increases the chance that the next person will get it working.

MelbourneDeveloper, 2019-08-06:
@Lotharyx , I'm very interested to see if this solves the issue. I would have thought that if the device doesn't implement those strings, it's likely to be a problematic device... But, I have no idea. We will see.

Lotharyx, 2019-08-07:
The latest develop branch corrects the empty-list issue.

The strange thing is that the sample code from master always worked correctly, so I don't know what the difference is between it and my application (other than sample is console while mine is WPF).

MelbourneDeveloper, 2019-08-07:
@lotharyx I think you've alluded to another problem. The sample probably works because it's .net core while your WPF app is .net framework. Someone else was saying that they had an issue with framework. 

As for USB, are you sure that the device is registered as a winusb device? 

Did you turn on break on all exceptions to try to catch any errors?

MelbourneDeveloper, 2019-08-19:
@LuisAndres this will be closed soon. Please let me know if you're experiencing the issue with the latest code.

closed 2019-08-29 by MelbourneDeveloper