MelbourneDeveloper/Device.Net #85: Somehow return actual number of bytes read?

ejluttmann, 2019-07-23:
When calling ReadAsync(), it would be nice to know how many bytes were actually read.  In some cases, especially with Bulk transfers, its possible to receive less bytes than the requested number of bytes from the USB device based on the protocol being used. Would it would be possible to add an optional parameter that would return the number of bytes actually read? Right now it looks to just be zero filled.

MelbourneDeveloper, 2019-07-24:
Good call on this. 

Should the array be truncated to the size returned?

Should the result be a list and that list be truncated?

Should the result be a struct with the count?

For maintainability I would say truncate, but for performance struct. Generally I prefer maintainability over performance...

Thoughts?

ejluttmann, 2019-07-24:
My initial thought was truncating the size of the byte array returned, but for larger data transfers that could be a significant performance hit.  Really, just need the ability to obtain the number of actual bytes read - this could just be an 'int' defined in IDevice.  Same thing could be added for WriteAsync(), to get the actual number of bytes written (although probably not as necessary).

MelbourneDeveloper, 2019-07-29:
@ejluttmann please have a look at this branch and tell me what you think:

https://github.com/MelbourneDeveloper/Device.Net/tree/%2385-ReadResult

MelbourneDeveloper, 2019-07-29:
@ejluttmann I've only really solved the problem in Windows, but I've refactored so that it can be done for any platform.

MelbourneDeveloper, 2019-07-29:
@ejluttmann this is the main change for Windows:

[image]

ejluttmann, 2019-07-29:
That looks like it would work.

MelbourneDeveloper, 2019-07-30:
@ejluttmann OK. It's in the Develop branch now. It will go in to 3.0.

closed 2019-07-30 by MelbourneDeveloper

MelbourneDeveloper, 2019-08-03:
@ejluttmann this has been included in V3.0 Beta 2