MelbourneDeveloper/Device.Net #87: Issues on Raspberry Pi

PrasadKurdiMatta, 2019-07-24:
I am trying to build .Net Core application that runs on Raspberry Pi. I am doing this by publishing the .Net Core application as a Standalone package and built for Linux-arm.

But I am getting MonoException on Raspberry Pi and on Windows, my code is not recognizing my USB device.  Suggestion Please? I am guessing my Filterdefenition is not matching and is there any way of knowing my Filterdefinietion by code?

Here is my sample code:

Main:
    LibUsbUsbDeviceFactory.Register();
    USBServer server = new USBServer();
      var success = server.InitializeTrezorAsync().Result;
     server.StartListening();

USBServer:
    public class USBServer : IDisposable
    {
        #region Fields
        //Define the types of devices to search for. This particular device can be connected to via USB, or Hid
        private readonly List<FilterDeviceDefinition> _DeviceDefinitions = new List<FilterDeviceDefinition>
        {
          //  new FilterDeviceDefinition{ DeviceType= DeviceType.Hid, VendorId= 0x534C, ProductId=0x0001, Label="Trezor One Firmware 1.6.x", UsagePage=65280 },
            new FilterDeviceDefinition{ DeviceType= DeviceType.Hid},
            new FilterDeviceDefinition{ DeviceType= DeviceType.Usb}
           // new FilterDeviceDefinition{ DeviceType= DeviceType.Usb, VendorId= 0x534C, ProductId=0x0001, Label="Trezor One Firmware 1.6.x (Android Only)" },
         //   new FilterDeviceDefinition{ DeviceType= DeviceType.Usb, VendorId= 0x046D, ProductId=0xC534, Label="Toshiba" },
          //  new FilterDeviceDefinition{ DeviceType= DeviceType.Usb, VendorId= 0x0781, ProductId=0xC534, Label="Toshiba" }
        };

        #endregion

        #region Events

        public event EventHandler TrezorInitialized;
        public event EventHandler TrezorDisconnected;

        #endregion

        #region Public Properties
        public IDevice TrezorDevice { get; private set; }
        public DeviceListener DeviceListener { get; private set; }
        #endregion

        #region Event Handlers

        private void DevicePoller_DeviceInitialized(object sender, DeviceEventArgs e)
        {
            TrezorDevice = e.Device;
            TrezorInitialized?.Invoke(this, new EventArgs());
        }

        private void DevicePoller_DeviceDisconnected(object sender, DeviceEventArgs e)
        {
            TrezorDevice = null;
            TrezorDisconnected?.Invoke(this, new EventArgs());
        }

        #endregion

        #region Public Methods
        public void StartListening()
        {
            TrezorDevice?.Dispose();
            DeviceListener = new DeviceListener(_DeviceDefinitions, 3000);
            DeviceListener.DeviceDisconnected += DevicePoller_DeviceDisconnected;
            DeviceListener.DeviceInitialized += DevicePoller_DeviceInitialized;

            DeviceListener.Start();
        }

        public async Task<bool> InitializeTrezorAsync()
        {
            //Get the first available device and connect to it
            var devices = await DeviceManager.Current.GetDevicesAsync(_DeviceDefinitions);
            TrezorDevice = devices.FirstOrDefault();

            if (TrezorDevice != null)
            {
                await TrezorDevice?.InitializeAsync();
            }
            return true;
        }

        public async Task<byte[]> WriteAndReadFromDeviceAsync()
        {
            //Create a buffer with 3 bytes (initialize)
            var writeBuffer = new byte[64];
            writeBuffer[0] = 0x3f;
            writeBuffer[1] = 0x23;
            writeBuffer[2] = 0x23;

            //Write the data to the device
            return await TrezorDevice.WriteAndReadAsync(writeBuffer);
        }

        public void Dispose()
        {
            TrezorDevice?.Dispose();
        }

        #endregion
    }

MelbourneDeveloper, 2019-07-24:
@PrasadKurdiMatta 

Please log the ToString of the exception.

PrasadKurdiMatta, 2019-07-24:
Thanks for the quick reply, Here is the exception on Raspberry Pi

Unhandled Exception: System.AggregateException: One or more errors occurred. (li                             busb-1.0 library not found.  This is often an indication that libusb-1.0 was ins                             talled to '/usr/local/lib' and mono.net is not looking for it there. To resolve                              this, add the path '/usr/local/lib' to '/etc/ld.so.conf' and run 'ldconfig' as r                             oot. (http://www.mono-project.com/DllNotFoundException)) ---> System.DllNotFound                             Exception: libusb-1.0 library not found.  This is often an indication that libus                             b-1.0 was installed to '/usr/local/lib' and mono.net is not looking for it there                             . To resolve this, add the path '/usr/local/lib' to '/etc/ld.so.conf' and run 'l                             dconfig' as root. (http://www.mono-project.com/DllNotFoundException) ---> System                             .DllNotFoundException: Unable to load shared library 'libusb-1.0' or one of its                              dependencies. In order to help diagnose loading problems, consider setting the L                             D_DEBUG environment variable: liblibusb-1.0: cannot open shared object file: No                              such file or directory
   at MonoLibUsb.MonoUsbApi.Init(IntPtr& pContext)
   at MonoLibUsb.MonoUsbSessionHandle..ctor()
   --- End of inner exception stack trace ---
   at MonoLibUsb.MonoUsbSessionHandle..ctor()
   at MonoLibUsb.MonoUsbEventHandler.Init(UnixNativeTimeval unixNativeTimeval)
   at MonoLibUsb.MonoUsbApi.InitAndStart()
   at LibUsbDotNet.LudnMonoLibUsb.MonoUsbDevice.get_MonoUsbDeviceList()
   at LibUsbDotNet.Main.LegacyUsbRegistry.get_DeviceList()
   at LibUsbDotNet.UsbDevice.get_AllLibUsbDevices()
   at LibUsbDotNet.UsbDevice.get_AllDevices()
   at Device.Net.LibUsb.LibUsbDeviceFactoryBase.<>c__DisplayClass6_0.<GetConnect                             edDeviceDefinitionsAsync>b__0()
   at System.Threading.Tasks.Task`1.InnerInvoke()
   at System.Threading.Tasks.Task.<>c.<.cctor>b__278_1(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionCo                             ntext, ContextCallback callback, Object state)
--- End of stack trace from previous location where exception was thrown ---
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionCo                             ntext, ContextCallback callback, Object state)
   at System.Threading.Tasks.Task.ExecuteWithThreadLocal(Task& currentTaskSlot)
--- End of stack trace from previous location where exception was thrown ---
   at Device.Net.LibUsb.LibUsbDeviceFactoryBase.GetConnectedDeviceDefinitionsAsy                             nc(FilterDeviceDefinition deviceDefinition)
   at Device.Net.DeviceManager.GetDevicesAsync(IList`1 deviceDefinitions)
   at TestUSB.USBServer.InitializeTrezorAsync() in C:\Users\prasa\source\repos\T                             estUSB\USBServer.cs:line 67
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotifica                             tion)
   at System.Threading.Tasks.Task`1.get_Result()
   at TestUSB.Program.Main(String[] args) in C:\Users\prasa\source\repos\TestUSB

MelbourneDeveloper, 2019-07-24:
@PrasadKurdiMatta if you run the windows sample with no modifications whatsoever on windows, do you see the device listed?

PrasadKurdiMatta, 2019-07-24:
On Windows, I don't see any devices when I call the function

var devices = await DeviceManager.Current.GetDevicesAsync(_DeviceDefinitions);

and when I plug in a new USB, DeviceListener is also not called.
I am not sure whether I am using correct FilterDefenitions.

Thanks

MelbourneDeveloper, 2019-07-24:
@PrasadKurdiMatta that's an easy one. You need to install the libusb library on the device running the code. 

It says

> Unable to load shared library 'libusb-1.0' or one of its dependencies

If you look at the libusb sample for windows you will see that the library is included in the sample project. That only works on windows.

MelbourneDeveloper, 2019-07-24:
@PrasadKurdiMatta run the windows sample completely unaltered. 

Take a screen shot of the console app and paste it in to a git comment. Paste with ctrl v.

MelbourneDeveloper, 2019-07-24:
@PrasadKurdiMatta I can TeamViewer on to your computer right now for ten minutes. Want to do that?

MelbourneDeveloper, 2019-07-24:
Hit me up on twitter DM or email me through my website.

MelbourneDeveloper, 2019-07-24:
@PrasadKurdiMatta last point is that talking to a USB stick is probably not going to do anything. USB sticks are recognised as standard devices so you can read/write data by copying files with the normal file system. This library is designed for custom communication where you send request/response packets to the devices with a specially designed protocol.

PrasadKurdiMatta, 2019-07-24:
Thanks. But without knowing Vendor ID and Product Id, is there any way to recognize and read the USB drive?

MelbourneDeveloper, 2019-07-24:
I don't think so

MelbourneDeveloper, 2019-07-24:
My question would be : why can't you just copy files to/from the device like a normal drive?

PrasadKurdiMatta, 2019-07-24:
My requirement is the user can plug USB device(Any USB device) anytime, my app should sense the USB plugging in Event and read from that USB. The problem I have is we don't know whats the VID and PID that user is plugging in. 

Thanks

MelbourneDeveloper, 2019-07-24:
I see. Yes. That's a common problem. But , there will be specific APIs in each operating system to tell you when a drive is connected. I doubt this is the right library for the job, but if you ever find the right API for that, I would love to hear about it t.

MelbourneDeveloper, 2019-07-25:
@PrasadKurdiMatta can I close this issue for now?

PrasadKurdiMatta, 2019-07-25:
sure

closed 2019-07-25 by PrasadKurdiMatta