MelbourneDeveloper/Device.Net #88: HID Device not detecting

srganeshram, 2019-07-27:
Hi,

I am trying to read data from a HID device, the device type is ([USB RFID Id Contactless Proximity Smart Card Reader 125KHZ Em4001 Em4100](https://www.amazon.in/gp/product/B072TH16ZZ/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1)). The device type is not detected and no events like connected and disconnected event triggered.

Please suggest.

I have provided the sample code for your reference 
`     class Program
        {
            #region Fields
             private static readonly TrezorExample _DeviceConnectionExample = new TrezorExample();
            /// <summary>
            /// TODO: Test these!
            /// </summary>
        private static readonly DebugLogger Logger = new DebugLogger();
        private static readonly DebugTracer Tracer = new DebugTracer();
        #endregion

        static void Main(string[] args)
        {
            Console.WriteLine("Hello USB!");

            WindowsUsbDeviceFactory.Register(Logger, Tracer);
            WindowsHidDeviceFactory.Register(Logger, Tracer);

            var devices =  DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(null);
            Console.WriteLine("Currently connected devices: ");
            foreach (var device in devices.Result)
            {
                Console.WriteLine($"{device.DeviceId} - {device.Label} - {device.Manufacturer} -  {device.ProductName}");
            }
            Console.WriteLine();
            Console.WriteLine("Connected device completed.");

            _DeviceConnectionExample.TrezorInitialized += _DeviceConnectionExample_TrezorInitialized;
            _DeviceConnectionExample.TrezorDisconnected += _DeviceConnectionExample_TrezorDisconnected;
            Go();
            new ManualResetEvent(false).WaitOne();
        }

        private static async Task Go()
        {
            var menuOption = await Menu();

            switch (menuOption)
            {
                case 1:
                    try
                    {
                        await _DeviceConnectionExample.InitializeTrezorAsync();
                        await DisplayDataAsync();
                        _DeviceConnectionExample.Dispose();

                        GC.Collect();

                        await Task.Delay(10000);
                    }
                    catch (Exception ex)
                    {
                        Console.Clear();
                        Console.WriteLine(ex.ToString());
                    }
                    Console.ReadKey();
                    break;
                case 2:
                    Console.Clear();
                    DisplayWaitMessage();
                    _DeviceConnectionExample.StartListening();
                    break;
            }
        }

        #region Event Handlers
        private static void _DeviceConnectionExample_TrezorDisconnected(object sender, EventArgs e)
        {
            Console.Clear();
            Console.WriteLine("Disconnected.");
            DisplayWaitMessage();
        }

        private static async void _DeviceConnectionExample_TrezorInitialized(object sender, EventArgs e)
        {
            try
            {
                Console.Clear();
                await DisplayDataAsync();
            }
            catch (Exception ex)
            {
                Console.Clear();
                Console.WriteLine(ex.ToString());
            }
        }
        #endregion

        #region Private Methods
        private async static Task<int> Menu()
        {
            while (true)
            {
                Console.Clear();

                var devices = await DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(null);
                Console.WriteLine("Currently connected devices: ");
                foreach (var device in devices)
                {
                    Console.WriteLine(device.DeviceId);
                }
                Console.WriteLine();

                Console.WriteLine("Console sample. This sample demonstrates either writing to the first found connected device, or listening for a device and then writing to it. If you listen for the device, you will be able to connect and disconnect multiple times. This represents how users may actually use the device.");
                Console.WriteLine();
                Console.WriteLine("1. Write To Connected Device");
                Console.WriteLine("2. Listen For Device");
                var consoleKey = Console.ReadKey();
                if (consoleKey.KeyChar == '1') return 1;
                if (consoleKey.KeyChar == '2') return 2;
            }
        }

        private static async Task DisplayDataAsync()
        {
            var bytes = await _DeviceConnectionExample.WriteAndReadFromDeviceAsync();
            Console.Clear();
            Console.WriteLine("Device connected. Output:");
            DisplayData(bytes);
        }

        private static void DisplayData(byte[] readBuffer)
        {
            Console.WriteLine(string.Join(' ', readBuffer));
            Console.ReadKey();
        }

        private static void DisplayWaitMessage()
        {
            Console.WriteLine("Waiting for device to be plugged in...");
        }
        #endregion
    }`

MelbourneDeveloper, 2019-07-27:
@srganeshram please post the logging output from the output window.

Also, please turn on break on all exceptions and post a screen shot of any exceptions. 

https://christianfindlay.com/2019/07/14/visual-studio-break-on-all-exceptions/

srganeshram, 2019-07-28:
@MelbourneDeveloper 
Thanks for your quick response, below are the details you wanted.

**Console Output:**
[image]

**Exception Details and breakdown**

[image]

[image]

I got the error @ WindowsBaseDevice.cs 

**Exception details**

Device.Net.Exceptions.ApiException
  HResult=0x80131500
  Message=Could not get Hid Attributes (Call HidD_GetAttributes). Error code: 6
  Source=Device.Net
  StackTrace:
   at Device.Net.Windows.WindowsDeviceBase.HandleError(Boolean isSuccess, String message) in D:\SourceCode\Device.Net\src\Device.Net\Windows\WindowsDeviceBase.cs:line 37

MelbourneDeveloper, 2019-07-28:
@srganeshram OK, I understand the problem. Thankyou very much for your detailed description.

The problem is first occurring here:
https://github.com/MelbourneDeveloper/Device.Net/blob/190819464818a92b34c10a2907d86274eadec258/src/Hid.Net/Windows/WindowsHidDeviceFactory.cs#L20

The handle is being properly created so when it attempts to get the description for the Hid device, error 6 is returned. Error code 6 just means that the handle is invalid.

One potential reason why this might be happening is that the device is being opened in read/write mode with file sharing. Perhaps the device doesn't allow this. I have added some better error handling so that the problem is caught earlier and there's a description of the problem.

You can see it in the Develop branch:
https://github.com/MelbourneDeveloper/Device.Net/blob/e268db3e778fc79f26ec2645165e2825aee7914c/src/Hid.Net/Windows/WindowsHidDeviceFactory.cs#L27

So, a question comes... Is this device read only? Or could there be a problem with shared connections? If so, maybe that is why connection is failing. Perhaps the code needs to be changed 

Here:
https://github.com/MelbourneDeveloper/Device.Net/blob/e268db3e778fc79f26ec2645165e2825aee7914c/src/Hid.Net/Windows/WindowsHidDeviceFactory.cs#L25

And:
https://github.com/MelbourneDeveloper/Device.Net/blob/e268db3e778fc79f26ec2645165e2825aee7914c/src/Hid.Net/Windows/WindowsHIDDevice.cs#L72

I've logged this issue:
https://github.com/MelbourneDeveloper/Device.Net/issues/89

My advice to you is to clone the Develop branch, and fix the code for your device in those spots. Have a look at this documentation:
https://docs.microsoft.com/en-us/windows/win32/api/fileapi/nf-fileapi-createfilea

If you get it working, I can review a PR and get the fix in to the library.

srganeshram, 2019-07-28:
It seems like as you said, I am sure that device wont allow us to write, its a access card reader, it is used for reading card data only.

I took your updated code and found the error as 

[image]

and the call stack is

[image]

MelbourneDeveloper, 2019-07-28:
@srganeshram try changing the desired access, share mode, and createdisposition values.

Here is the documentation for CreateFile: https://docs.microsoft.com/en-us/windows/win32/api/fileapi/nf-fileapi-createfilea

https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/opening-hid-collections

MelbourneDeveloper, 2019-07-28:
@srganeshram the flags attributes argument could also be an issue...

Do you know the protocol for the card reader? My guess that a card reader would just show up as a device drive. If it just shows up as a drive, there's not much point in trying to use Hid.Net. You can just use standard file access methods to read/write files etc.

srganeshram, 2019-07-28:
@melbournedeveloper 
try changing the desired access, share mode, and createdisposition values.

I tried this but same issue.

The device type is "USB RFID Id Contactless Proximity Smart Card Reader 125KHZ Em4001 Em4100".

The reader is working fine, if I show the access card it reads it and displays in notepad or any textbox.

I am not sure about the protocol, and I don't think that it supports protocol. I just want to read any raw data from that device.
If you want the device details I have shared the link in my first comment.

Any help appreciate;

MelbourneDeveloper, 2019-07-28:
@srganeshram can I TeamViewer on to your computer right now?

MelbourneDeveloper, 2019-07-28:
@srganeshram and chance you could send a "USB RFID Id Contactless Proximity Smart Card Reader 125KHZ Em4001 Em4100" to Australia? I'd like to try to get it working.

MelbourneDeveloper, 2019-07-28:
@srganeshram please send me a TeamViewer link on Twitter or Slack

srganeshram, 2019-07-28:
@MelbourneDeveloper  I will share in a few minutes.

srganeshram, 2019-07-28:
I just twitted. Please check

MelbourneDeveloper, 2019-07-28:
@srganeshram I didn't get it

srganeshram, 2019-07-28:
your twitter id please.

MelbourneDeveloper, 2019-07-28:
@srganeshram cfdevelop

srganeshram, 2019-07-28:
Please check now.

MelbourneDeveloper, 2019-07-28:
@srganeshram you gave away your TeamViewer details publicly Please change your TeamViewer password immediately!

srganeshram, 2019-07-28:
I will reset no worries

MelbourneDeveloper, 2019-07-28:
@srganeshram please send me a message. Not tweet.

srganeshram, 2019-07-28:
@MelbourneDeveloper 
1 335 164 424
9v5b6y

MelbourneDeveloper, 2019-07-28:
No!!!!!

Delete this and change your password!!!!QUICKLY! This is a public forum.

MelbourneDeveloper, 2019-07-28:
@srganeshram you cannot give away your credentials in public like this.

MelbourneDeveloper, 2019-07-28:
I'm sorry. We cannot do TeamViewer at this point. Please regenerate your password and we can chat about this later.

srganeshram, 2019-07-28:
Ok. Not a problem. We can check later..

srganeshram, 2019-07-28:
If it is fine, please share your email I will send the teamview link.

MelbourneDeveloper, 2019-07-29:
@srganeshram where can I get a USB RFID Id Contactless Proximity Smart Card Reader 125KHZ Em4001 Em4100?

Is it possible to send one to Australia?

I will try to get this working if I can get one.

srganeshram, 2019-07-29:
@MelbourneDeveloper I will check and let you konw.

srganeshram, 2019-07-29:
@MelbourneDeveloper I just found one, in amazon.com. [RFID USB Card Reader](https://www.amazon.com.au/Reader-LANMU-125khz-Contactless-Proximity/dp/B07B7H6CQ2/ref=sr_1_fkmr0_1?keywords=USB+RFID+Id+Contactless+Proximity+Smart+Card+Reader+125KHZ+Em4001+Em4100&qid=1564399498&s=gateway&sr=8-1-fkmr0)

MelbourneDeveloper, 2019-07-29:
@srganesham I ordered one. It will take a couple of weeks.

srganeshram, 2019-07-29:
@MelbourneDeveloper I am very kind of you. Thanks for the support.

MelbourneDeveloper, 2019-08-03:
@srganeshram the package arrived but they couldn't find my letterbox or something... I think it will be delivered again Monday.

Could you please try the new 3.0 Beta 2 version to see if you are having the same issue?

srganeshram, 2019-08-03:
@MelbourneDeveloper, 
**Package:** I am waiting for the delivery.
**New 3.0 Beta 2 version** - I tried with the new beta version, the difference I found is earlier in my console it wont recognize the USB device itself, but now it can able to detect and list the devices.

But, I could not able to read the data still.I have attached the error details for your reference.

**Console Window - (Using New version 3.0 Beta 2)**
[image]
The above highlighted in red is the device I am trying to read, and it was listed out.

**Error Details  - (Using New version 3.0 Beta 2)**

[image]

**Call back**
[image]

Let me know if you need any further details..

Thanks

MelbourneDeveloper, 2019-08-03:
@srganeshram on the line where it tries to open the file handle change apicalls.genericread to 0 and see what happens.

srganeshram, 2019-08-03:
@MelbourneDeveloper  I changed as stated.
`_ReadSafeFileHandle = APICalls.CreateFile(DeviceId, APICalls.GenericRead, 3, IntPtr.Zero, APICalls.OpenExisting, 0, IntPtr.Zero);`
to
`_ReadSafeFileHandle = APICalls.CreateFile(DeviceId, 0, 3, IntPtr.Zero, APICalls.OpenExisting, 0, IntPtr.Zero);`

It seems it worked and passed the above line, I tried to read the data from the device by using "ReadReportAsync" method, but it throws me an error as **"System.UnauthorizedAccessException: 'Access to the path is denied.'"**

**Error details**
[image]

**Instance details of "_ReadFileStream"**
[image]

**Call back**
[image]

MelbourneDeveloper, 2019-08-03:
@srganeshram this is the same issue I get when I try to read data from my mouse. 

Please wait until Monday when I get the same device.

srganeshram, 2019-08-03:
Ok. Thanks for your support. We can wait.

MelbourneDeveloper, 2019-08-06:
@srganeshram , I got the device. The problem is that it doesn't seem to work... 

I put my card on it, but it doesn't beep or go green...

MelbourneDeveloper, 2019-08-06:
@srganeshram I tried multiple cards but nothing...

srganeshram, 2019-08-06:
You have to use HID Proximity Card. It should work.

Open a notepad and show the card, your card number should appear.

MelbourneDeveloper, 2019-08-06:
@srganeshram I used my credit card and it didn't work. I also used another card.

MelbourneDeveloper, 2019-08-06:
@srganeshram I tried three different credit cards, three different USB ports, and two different cables. Nothing works. The device beeps when I plug it in to the computer, the red LED appears, but nothing happens when I tap it with a credit card.

srganeshram, 2019-08-07:
@MelbourneDeveloper I am sure that you cannot use credit card to read the data.

You should use RFID Proximity Card ([RFID Proximity Id Card](https://www.amazon.com.au/compatible-EM4100-support-Attendance-System/dp/B07BGWV4XT/ref=sr_1_1?keywords=proximity+card&qid=1565162146&s=gateway&sr=8-1))

The above card can be read from the device.

Thanks

MelbourneDeveloper, 2019-08-07:
@srganeshram oh no! So, I have to order something else from Amazon?

Also, why can't you just use the device as a keyboard? The device is essentially a keyboard. The device sends keys to notepad like a regular keyboard. Is there some extra data if you read via USB?

srganeshram, 2019-08-07:
@melbournedeveloper I just showed the card. I have additional cards with me. If you need I can send the one to Australia.

MelbourneDeveloper, 2019-08-07:
@srganeshram send me a private message on Twitter or Slack. I will send you the address to send it to.

MelbourneDeveloper, 2019-08-07:
@srganeshram again:

Why can't you just use the device as a keyboard? The device is essentially a keyboard. The device sends keys to notepad like a regular keyboard. Is there some extra data if you read via USB?

MelbourneDeveloper, 2019-08-08:
@srganeshram there is a computer shop near my work. I will try to buy some cards today.

srganeshram, 2019-08-09:
Great. If don't work let me send one for you.

MelbourneDeveloper, 2019-08-19:
@srganeshram I haven't been able to find the cards here. You will need to send me one. But, I've tried all kinds of cards and nothing works. I also tried my passport. I'm pretty sure this device is useless.

srganeshram, 2019-08-22:
@MelbourneDeveloper 

No Problem. I have found a workaround for this. As you stated that this device is detected as a simple keyboard that gave me a hint of reading raw input from keyboard. I used the below article 
[Reading Raw Input from keyboard](https://www.codeproject.com/Articles/17123/Using-Raw-Input-from-C-to-handle-multiple-keyboard). 
Now, I am able to read the device data when I show the card. 

If you want to read the data from that device, you should use the proximity card. I can share the one from my end, if you wish you can share your address in a private message through twitter(@SRGaneshRam) or email(sr.ganeshram@hotmail.com)

Thanks.

closed 2019-08-22 by MelbourneDeveloper

AalapMistry, 2019-12-13:
'max430AI.ConfigurationTool.exe' (CLR v4.0.30319: DefaultDomain): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_64\mscorlib\v4.0_4.0.0.0__b77a5c561934e089\mscorlib.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'max430AI.ConfigurationTool.exe' (CLR v4.0.30319: DefaultDomain): Loaded 'F:\max430AI.ConfigurationTool\max430AI.ConfigurationTool\bin\Debug\max430AI.ConfigurationTool.exe'. Symbols loaded.
'max430AI.ConfigurationTool.exe' (CLR v4.0.30319: max430AI.ConfigurationTool.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Windows.Forms\v4.0_4.0.0.0__b77a5c561934e089\System.Windows.Forms.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'max430AI.ConfigurationTool.exe' (CLR v4.0.30319: max430AI.ConfigurationTool.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System\v4.0_4.0.0.0__b77a5c561934e089\System.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'max430AI.ConfigurationTool.exe' (CLR v4.0.30319: max430AI.ConfigurationTool.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Drawing\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Drawing.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'max430AI.ConfigurationTool.exe' (CLR v4.0.30319: max430AI.ConfigurationTool.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Configuration\v4.0_4.0.0.0__b03f5f7f11d50a3a\System.Configuration.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'max430AI.ConfigurationTool.exe' (CLR v4.0.30319: max430AI.ConfigurationTool.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Core\v4.0_4.0.0.0__b77a5c561934e089\System.Core.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'max430AI.ConfigurationTool.exe' (CLR v4.0.30319: max430AI.ConfigurationTool.exe): Loaded 'C:\WINDOWS\Microsoft.Net\assembly\GAC_MSIL\System.Xml\v4.0_4.0.0.0__b77a5c561934e089\System.Xml.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'max430AI.ConfigurationTool.exe' (CLR v4.0.30319: max430AI.ConfigurationTool.exe): Loaded 'F:\max430AI.ConfigurationTool\max430AI.ConfigurationTool\bin\Debug\Device.Net.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'max430AI.ConfigurationTool.exe' (CLR v4.0.30319: max430AI.ConfigurationTool.exe): Loaded 'F:\max430AI.ConfigurationTool\max430AI.ConfigurationTool\bin\Debug\Hid.Net.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
Line No:1
High:0
Low:8
'max430AI.ConfigurationTool.exe' (CLR v4.0.30319: max430AI.ConfigurationTool.exe): Loaded 'C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\Remote Debugger\x64\Runtime\Microsoft.VisualStudio.Debugger.Runtime.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
The thread 0x1ad8 has exited with code 0 (0x0).
The thread 0xae0 has exited with code 0 (0x0).
Status Bit:1
07FFFFFFFFFFFFFF
Line No:2
High:0
Low:16
Status Bit:1
070011D9FC800100
Line No:3
High:0
Low:24
Status Bit:1
07011213D7FFFFFF
Line No:4
High:0
Low:32
Status Bit:1
07029A0100001100
Line No:5
High:0
Low:40
Status Bit:1
0703000000000000
Line No:6
High:0
Low:48
Status Bit:1
0704C22200000000
Line No:7
High:0
Low:56
Status Bit:1
0705640000000C00
Line No:8
High:0
Low:64
Status Bit:1
0706000000000000
Line No:9
High:0
Low:72
Status Bit:1
0707112000000000
Line No:10
High:0
Low:80
Status Bit:1
0708960000001600
Line No:11
High:0
Low:88
Status Bit:1
0709000000000000
Line No:12
High:0
[... 784 lines cut ...]
High:5
Low:72
Status Bit:1
07ADFFFFFFFFFFFF
Line No:170
High:5
Low:80
Status Bit:1
07AE00000000401F
Line No:171
High:5
Low:88
Status Bit:1
07AF803EC05D007D
Line No:172
High:5
Low:96
Status Bit:1
07B0FFFFFFFFFFFF
Line No:173
High:5
Low:104
Exception thrown: 'System.IO.IOException' in mscorlib.dll

why Exception while reading data ?  please help error Exception occure during read data byte frame from hid USB device , using hid.net and device.net library

carl-qva, 2020-01-15:
Hi MelbourneDeveloper, im having an issue with Device.Net (and in general with all hid libraries that i've tested so far). The problem is that my app can communicate with my hid device only in administrator mode, when the app is not run in admin mode the device is not found. In admin mode i can read/write data without any problem. See this log from Device.Net and please help

region: ApiService, message: Calling CreateFile for DeviceId: \\?\hid#vid_0483&pid_a2ca&mi_00#7&1a31e08c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}. Desired Access: None. Share mode: 3. Creation Disposition: 3, LogLevel: Information 
Exception thrown: 'Device.Net.Exceptions.DeviceException' in Hid.Net.dll
region: WindowsHidDeviceFactory, message: GetDeviceDefinition error. Device Id: \\?\hid#vid_0483&pid_a2ca&mi_00#7&1a31e08c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}, LogLevel: Error, exception: CreateReadConnection call with Id of \\?\hid#vid_0483&pid_a2ca&mi_00#7&1a31e08c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} failed.
region: WindowsHidDeviceFactory, message: Device with path \\?\hid#vid_0483&pid_a2ca&mi_00#7&1a31e08c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} was skipped. See previous logs., LogLevel: Warning