MelbourneDeveloper/Device.Net #91: Working with .net core in Windows 10, but not working with full .NET Framewrok.

anandthirumala, 2019-07-29:
HidAPICall -> GetManufaturer 
GetHidString -> getString fails 

In WindowsDeviceBase -> Could not get Hid string

Then got Could not get Hid attribute.

anandthirumala, 2019-07-29:
We need this to be working on .NET framework. 

Please help if you could. 

Kindly respond ASAP

MelbourneDeveloper, 2019-07-29:
@anandthirumala please start with some details. 

- Ensure that you are using some logging
- Post your code so I can see what is happening. 
- Run the code and send me the results of the logger (copy paste the results)
-Turn on Break on All Exceptions (https://christianfindlay.com/2019/07/14/visual-studio-break-on-all-exceptions/)
- Run the code and send a screenshot of where the code is breaking for your device

Are running this on the same computer, with the same Device?

MelbourneDeveloper, 2019-07-29:
@anandthirumala maybe tonight in around 8 or 9 hours I can teamviewer on to your computer. 

Please DO NOT post any TeamViewer details in a public place. Please PM them to me on Twitter or Slack.

anandthirumala, 2019-07-29:
@MelbourneDeveloper thanks for responding Christian  (hope you don't mind calling me that). I am based in the UK. It's night time now. I am doing this project at work, so there is no mean for me to access the device from home. I would be very kind of you if you could do the same after 9 am UK time. My issue is looking very much like that of srganeshram, but I can get it working with .NET Core.  

Sreenshot will be the same us  srganeshram. That method is called from two places as mentioned before. FIrst one is unique to me

I have turned on all exception. 
-Turn on Break on All Exceptions (https://christianfindlay.com/2019/07/14/visual-studio-break-on-all-exceptions/)

And I am using the repos clone now, though started with lib in the beginning after reading some of your early posts.

MelbourneDeveloper, 2019-07-29:
@anandthirumala I will do an audit of the code around the connection area. This might mean that there are some bug fixes coming. Please be patient while I investigate.

anandthirumala, 2019-07-30:
Hello @MelbourneDeveloper, I am at work and could run teamviewer. If you have time, please do let me know. 

Thanks for all the help.

MelbourneDeveloper, 2019-07-30:
@anandthirumala  maybe in a couple of hours. I am investigating your issue.

anandthirumala, 2019-07-30:
@MelbourneDeveloper Thanks a lot. I don't know how to thank you enough.

anandst, 2019-07-30:
@MelbourneDeveloper There are two listing of the device 

(unnamed manufacturer) (unnamed product) (no serial number) (VID 1155, PID 22315, version 0.0) @ \\?\hid#vid_0483&pid_572b&mi_01#8&14df224e&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}

(unnamed manufacturer) (unnamed product) (no serial number) (VID 1155, PID 22315, version 0.0) @ \\?\hid#vid_0483&pid_572b&mi_00#8&280f3e8c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}

Hope this information is useful.

MelbourneDeveloper, 2019-07-30:
Please follow these instructions:

@anandthirumala please start with some details.

- Ensure that you are using some logging
- Post your code so I can see what is happening.
- Run the code and send me the results of the logger (copy paste the results)
- Turn on Break on All Exceptions (https://christianfindlay.com/2019/07/14/visual-studio-break-on-all-exceptions/)
- Run the code and send a screenshot of where the code is breaking for your device

Are running this on the same computer, with the same Device?

anandst, 2019-07-30:
@MelbourneDeveloper yes I am running on the same computer as the device.

manufaturerer is appearing as (Standard system devices).

string GetManufacturer(SafeFileHandle safeFileHandle) is the method which threw below error, 

In WindowsDeviceBase -> Could not get Hid string. Error code: 87.

anandst, 2019-07-30:
@MelbourneDeveloper  static HidAttributes GetHidAttributes(SafeFileHandle safeFileHandle)

threw Could not get Hid attribute. Error code 6.

MelbourneDeveloper, 2019-07-30:
@anandst please follow the instructions.

anandst, 2019-07-30:
using System;
using System.Collections.Generic;
using System.Linq;
using System.Threading.Tasks;
using Device.Net;
using Hid.Net.Windows;

namespace ToshibhaMsrDallasKeySO
{
    internal sealed class ToshibhaMsrDallasKeyServiceObject : IDisposable
    {
        #region Fields
        private const int PollMilliseconds = 3000;

        //Define the types of devices to search for. This particular device can be connected to via USB, or Hid
        private readonly List<FilterDeviceDefinition> _DeviceDefinitions = new List<FilterDeviceDefinition>
        {
            new FilterDeviceDefinition{ DeviceType= DeviceType.Usb, VendorId= 0x0483, ProductId=0x572b, Label="USB Input Device" }
        };

        #endregion

        #region Events
        public event EventHandler Initialized;
        public event EventHandler Disconnected;
        #endregion

        #region Public Properties
        public IDevice Device { get; private set; }
        public  DeviceListener DeviceListener { get;  }
        #endregion

        #region Constructor
        public ToshibhaMsrDallasKeyServiceObject()
        {
            var Logger = new DebugLogger();
            var Tracer = new DebugTracer();
            WindowsHidDeviceFactory.Register(Logger, Tracer);
            DeviceListener = new DeviceListener(_DeviceDefinitions, PollMilliseconds) { Logger = new DebugLogger() };
        }
        #endregion

        #region Event Handlers
        private void DeviceInitializedHandler(object sender, DeviceEventArgs e)
        {
            Device = e.Device;
            Initialized?.Invoke(this, new EventArgs());
        }

        private void DevicePoller_DeviceDisconnected(object sender, DeviceEventArgs e)
        {
            Device = null;
            Disconnected?.Invoke(this, new EventArgs());
        }
        #endregion

        #region Public Methods
        public void StartListening()
        {
            Device?.Close();
            DeviceListener.DeviceDisconnected += DevicePoller_DeviceDisconnected;
            DeviceListener.DeviceInitialized += DeviceInitializedHandler;
            DeviceListener.Start();
        }

        public async Task InitializeAsync()
        {
            //Get the first available device and connect to it
            var devices = await DeviceManager.Current.GetDevicesAsync(_DeviceDefinitions);
            Device = devices.FirstOrDefault();

            if (Device == null) throw new Exception("There were no devices found");

            await Device.InitializeAsync();
        }

        public async Task<byte[]> WriteAndReadFromDeviceAsync()
        {
            //Create a buffer with 3 bytes (initialize)
            var writeBuffer = new byte[64];
            writeBuffer[0] = 0x3f;
            writeBuffer[1] = 0x23;
            writeBuffer[2] = 0x23;

            //Write the data to the device
            return await Device.WriteAndReadAsync(writeBuffer);
        }

        public void Dispose()
        {
            DeviceListener.DeviceDisconnected -= DevicePoller_DeviceDisconnected;
            DeviceListener.DeviceInitialized -= DeviceInitializedHandler;
            DeviceListener.Dispose();
            Device?.Dispose();
        }
        #endregion
    }
}

anandst, 2019-07-30:
Type is HId not USB, I just changed to USB to see what happens.

anandst, 2019-07-30:
@MelbourneDeveloper calling code,

using System;
using System.Threading;
using System.Threading.Tasks;

namespace ToshibhaMsrDallasKeySO
{
    internal class Program
    {
        private static bool disconnected;
        private static ToshibhaMsrDallasKeyServiceObject ServiceObject;

        private static void Main(string[] args)
        {
            ServiceObject = new ToshibhaMsrDallasKeyServiceObject();

            ServiceObject.Initialized += ServiceObjectInitialized;
            ServiceObject.Disconnected += ServiceObjectDisconnected;

            Console.Clear();
            ServiceObject.StartListening();

            new ManualResetEvent(false).WaitOne();
        }

        #region Event Handlers
        private static void ServiceObjectDisconnected(object sender, EventArgs e)
        {
            Console.Clear();
            Console.WriteLine("Disconnected.");
            DisplayWaitMessage();

            disconnected = true;
        }

        private static async void ServiceObjectInitialized(object sender, EventArgs e)
        {
            try
            {
                disconnected = false;
                Console.Clear();
                await DisplayDataAsync();
            }
            catch (Exception ex)
            {
                Console.Clear();
                Console.WriteLine(ex.ToString());
            }
        }

        #endregion

        #region Private Methods

        private static async Task DisplayDataAsync()
        {
            while (!disconnected)
            {
                var bytes = await ServiceObject.WriteAndReadFromDeviceAsync();
                Console.Clear();
                Console.WriteLine("Device connected. Output:");
                DisplayData(bytes);
            }
        }

        private static void DisplayData(byte[] readBuffer)
        {
            Console.WriteLine(string.Join("", readBuffer));

            var result = System.Text.Encoding.UTF8.GetString(readBuffer);
            Console.WriteLine(result);
        }

        private static void DisplayWaitMessage()
        {
            Console.WriteLine("Waiting for device to be plugged in...");
        }

        #endregion
    }
}

MelbourneDeveloper, 2019-07-30:
@anandst results of the logger please.

MelbourneDeveloper, 2019-07-30:
@anandst OK. Let's try TeamViewer. Please send me the details on Twitter by DM

DO NOT send in public!!!!

https://twitter.com/CFDevelop

MelbourneDeveloper, 2019-07-30:
@anandst I will be turning my computer off soon, please contact...

anandst, 2019-07-30:
2019-07-30 10:52:19.0020|ERROR|ToshibhaMsrDallasKeySO.ToshibhaMsrDallasKeyServiceObject|GetDeviceDefinition error. Device Id: \\?\hid#vid_0483&pid_572b&mi_01#8&14df224e&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} Device.Net.Exceptions.ApiException
Could not get Hid string. Error code: 87
   at Device.Net.Windows.WindowsDeviceBase.HandleError(Boolean isSuccess, String message)
   at Hid.Net.Windows.HidAPICalls.GetHidString(SafeFileHandle safeFileHandle, GetString getString)
   at Hid.Net.Windows.HidAPICalls.GetManufacturer(SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId, SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId)
2019-07-30 10:52:19.1502|ERROR|ToshibhaMsrDallasKeySO.ToshibhaMsrDallasKeyServiceObject|GetDeviceDefinition error. Device Id: \\?\hid#vid_0483&pid_572b&mi_00#8&280f3e8c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} Device.Net.Exceptions.ApiException
Could not get Hid Attributes (Call HidD_GetAttributes). Error code: 6
   at Device.Net.Windows.WindowsDeviceBase.HandleError(Boolean isSuccess, String message)
   at Hid.Net.Windows.HidAPICalls.GetHidAttributes(SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId, SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId)
2019-07-30 10:52:21.9583|ERROR|ToshibhaMsrDallasKeySO.ToshibhaMsrDallasKeyServiceObject|GetDeviceDefinition error. Device Id: \\?\hid#vid_0483&pid_572b&mi_01#8&14df224e&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} Device.Net.Exceptions.ApiException
Could not get Hid string. Error code: 87
   at Device.Net.Windows.WindowsDeviceBase.HandleError(Boolean isSuccess, String message)
   at Hid.Net.Windows.HidAPICalls.GetHidString(SafeFileHandle safeFileHandle, GetString getString)
   at Hid.Net.Windows.HidAPICalls.GetManufacturer(SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId, SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId)
2019-07-30 10:52:21.9813|ERROR|ToshibhaMsrDallasKeySO.ToshibhaMsrDallasKeyServiceObject|GetDeviceDefinition error. Device Id: \\?\hid#vid_0483&pid_572b&mi_00#8&280f3e8c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} Device.Net.Exceptions.ApiException
Could not get Hid Attributes (Call HidD_GetAttributes). Error code: 6
   at Device.Net.Windows.WindowsDeviceBase.HandleError(Boolean isSuccess, String message)
   at Hid.Net.Windows.HidAPICalls.GetHidAttributes(SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId, SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId)
2019-07-30 10:52:24.9587|ERROR|ToshibhaMsrDallasKeySO.ToshibhaMsrDallasKeyServiceObject|GetDeviceDefinition error. Device Id: \\?\hid#vid_0483&pid_572b&mi_01#8&14df224e&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} Device.Net.Exceptions.ApiException
Could not get Hid string. Error code: 87
   at Device.Net.Windows.WindowsDeviceBase.HandleError(Boolean isSuccess, String message)
   at Hid.Net.Windows.HidAPICalls.GetHidString(SafeFileHandle safeFileHandle, GetString getString)
   at Hid.Net.Windows.HidAPICalls.GetManufacturer(SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId, SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId)
2019-07-30 10:52:24.9747|ERROR|ToshibhaMsrDallasKeySO.ToshibhaMsrDallasKeyServiceObject|GetDeviceDefinition error. Device Id: \\?\hid#vid_0483&pid_572b&mi_00#8&280f3e8c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} Device.Net.Exceptions.ApiException
Could not get Hid Attributes (Call HidD_GetAttributes). Error code: 6
   at Device.Net.Windows.WindowsDeviceBase.HandleError(Boolean isSuccess, String message)
   at Hid.Net.Windows.HidAPICalls.GetHidAttributes(SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId, SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId)
2019-07-30 10:52:27.9591|ERROR|ToshibhaMsrDallasKeySO.ToshibhaMsrDallasKeyServiceObject|GetDeviceDefinition error. Device Id: \\?\hid#vid_0483&pid_572b&mi_01#8&14df224e&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} Device.Net.Exceptions.ApiException
Could not get Hid string. Error code: 87
   at Device.Net.Windows.WindowsDeviceBase.HandleError(Boolean isSuccess, String message)
   at Hid.Net.Windows.HidAPICalls.GetHidString(SafeFileHandle safeFileHandle, GetString getString)
   at Hid.Net.Windows.HidAPICalls.GetManufacturer(SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId, SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId)
2019-07-30 10:52:27.9862|ERROR|ToshibhaMsrDallasKeySO.ToshibhaMsrDallasKeyServiceObject|GetDeviceDefinition error. Device Id: \\?\hid#vid_0483&pid_572b&mi_00#8&280f3e8c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} Device.Net.Exceptions.ApiException
Could not get Hid Attributes (Call HidD_GetAttributes). Error code: 6
[... cut ...]
2019-07-30 10:53:13.0173|ERROR|ToshibhaMsrDallasKeySO.ToshibhaMsrDallasKeyServiceObject|GetDeviceDefinition error. Device Id: \\?\hid#vid_0483&pid_572b&mi_00#8&280f3e8c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} Device.Net.Exceptions.ApiException
Could not get Hid Attributes (Call HidD_GetAttributes). Error code: 6
   at Device.Net.Windows.WindowsDeviceBase.HandleError(Boolean isSuccess, String message)
   at Hid.Net.Windows.HidAPICalls.GetHidAttributes(SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId, SafeFileHandle safeFileHandle)
   at Hid.Net.Windows.WindowsHidDeviceFactory.GetDeviceDefinition(String deviceId)

anandst, 2019-07-30:
Sorry didn't see that message I will ping

anandst, 2019-07-30:
Let me install team viewer

MelbourneDeveloper, 2019-07-30:
@anandst please hurry

anandst, 2019-07-30:
Says I can't message you in twitter

anandst, 2019-07-30:
I followed you now

MelbourneDeveloper, 2019-07-30:
@anandst  Try Slack

anandst, 2019-07-30:
Could you please follow me

anandst, 2019-07-30:
> @anandst Try Slack

Don't have slack installed

MelbourneDeveloper, 2019-07-30:
@anandst what's your handle

anandst, 2019-07-30:
@anandts

MelbourneDeveloper, 2019-07-30:
@anandst please DM me

anandst, 2019-07-30:
Send me your mail id please

MelbourneDeveloper, 2019-07-30:
@anandst send me your TeamView details.

MelbourneDeveloper, 2019-07-30:
@anandst I don't have any more time.

anandst, 2019-07-30:
1 343 736 322

anandst, 2019-07-30:
ng364e

MelbourneDeveloper, 2019-07-30:
NOOIOO!!!!!!

anandst, 2019-07-30:
Deleted

anandst, 2019-07-30:
Mail id please