MelbourneDeveloper/Device.Net #93: Trying to understand Library

Kyvex, 2019-08-02:
Private Shared Async Function InitializingDevices() As Task
        Dim logger = New DebugLogger
        Dim tracer = New DebugTracer
        WindowsUsbDeviceFactory.Register(logger, tracer)
        WindowsHidDeviceFactory.Register(logger, tracer)
        Dim deviceDefinitions = New List(Of FilterDeviceDefinition) From {
        New FilterDeviceDefinition With {
            .DeviceType = DeviceType.Hid,
            .VendorId = &H24A1,
            .ProductId = &HDA,
            .UsagePage = &H1
        }
    }
        Dim devices = Await DeviceManager.Current.GetDevicesAsync(deviceDefinitions)
        Dim moreDevices = devices.FirstOrDefault()
        Await moreDevices.InitializeAsync()
        Dim buffer = New Byte(63) {}
        buffer(0) = &H3F
        buffer(1) = &H23
        buffer(2) = &H23
        Dim readBuffer = Await moreDevices.WriteAndReadAsync(buffer)
    End Function

    Private Sub Button1_Click(sender As Object, e As RoutedEventArgs)
        Dim temp = InitializingDevices()
    End Sub

Hello! I recently started using your library and this is your Quick Start code. I am only using it to try and learn the library a little better. Every time I run the app, moreDevices always equals Nothing.

[image]

If anyone has some advice on how or what is wrong please help!

MelbourneDeveloper, 2019-08-03:
@Kyvex thanks for reporting. I have updated the wiki. There is nothing obviously wrong with your code. However, the first step it simply enumerate through all devices to get the device paths instead of using filter definitions. 

Please start with this article:
https://github.com/MelbourneDeveloper/Device.Net/wiki/Quick-Start#run-windows-sample-usbnetwindowssample

Then see enumerating devices:
https://github.com/MelbourneDeveloper/Device.Net/wiki/Enumerating-Connected-Devices

Then, if you still have issues, see this article:
https://github.com/MelbourneDeveloper/Device.Net/wiki/Debugging,-Logging,-and-Tracing

I have also updated the issues template, so please include all the information that is asked for there:
https://github.com/MelbourneDeveloper/Device.Net/blob/master/.github/ISSUE_TEMPLATE/issue-report.md

MelbourneDeveloper, 2019-08-06:
@Kyvex Hi. Have you made any progress on this? Are there any specific issues you need addressed?

MelbourneDeveloper, 2019-08-06:
@Kyvex I will close this in the next couple of days.

closed 2019-08-08 by MelbourneDeveloper