MelbourneDeveloper/Device.Net #98: Sample Code Only Enumerates HID Devices

Lotharyx, 2019-08-07:
Using the sample-code unmodified from develop branch, only HID devices are enumerated, despite other USB devices being attached.

No debug or trace output is generated for non-HID devices; they are simply not observed at all.

Here is the output from Usb.Net.WindowsSample, again the unmodified version straight from develop branch:

```
Currently connected devices:
\\?\hid#vid_1d50&pid_6122&mi_03#7&2436fed8&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
\\?\hid#vid_1d50&pid_6122&mi_02#7&1106e29a&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
\\?\hid#vid_1d50&pid_6122&mi_01#7&22939a4&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
\\?\hid#vid_1d50&pid_6122&mi_04#7&37671b16&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
\\?\hid#vid_1d50&pid_6122&mi_00#7&155955e2&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
\\?\hid#vid_046d&pid_c069#6&483d65f&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
```
As can be seen, all reported devices are HID, despite there being other USB devices attached to my system (a storage device, a virtual serial port, etc).  

Looking at the code in Visual Studio, it appears that LIBUSB is defined, so the factory that gets registered is `LibUsbUsbDeviceFactory`:

[image]

I would expect all USB devices to be enumerated, since the factory shouldn't be limited to HID devices.

Platform: Windows 10

MelbourneDeveloper, 2019-08-07:
@lotharyx, just quickly. Try dropping libusb and make sure windowsusbdevicefactory is getting registered.

Lotharyx, 2019-08-07:
I removed the conditional code for LIBUSB from the sample.  Although IntelliSense seems to think it can't find Usb.Net.Windows, the solution builds and runs fine.  The result is the same; only HID devices are enumerated.  The output is identical, as reported.  No exceptions are thrown.  Here is the debug output:
```
--------------------------------------
Message: Found device \\?\hid#vid_1d50&pid_6122&mi_03#7&2436fed8&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Time: 8/7/2019 4:30:57 PM
Section: WindowsHidDeviceFactory
Error: 
--------------------------------------
--------------------------------------
Message: Could not get Hid string. Caller: GetSerialNumber
Time: 8/7/2019 4:30:57 PM
Section: HidAPICalls
Error: 
--------------------------------------
--------------------------------------
Message: Found device \\?\hid#vid_1d50&pid_6122&mi_02#7&1106e29a&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Time: 8/7/2019 4:30:57 PM
Section: WindowsHidDeviceFactory
Error: 
--------------------------------------
--------------------------------------
Message: Could not get Hid string. Caller: GetSerialNumber
Time: 8/7/2019 4:30:57 PM
Section: HidAPICalls
Error: 
--------------------------------------
--------------------------------------
Message: Found device \\?\hid#vid_1d50&pid_6122&mi_01#7&22939a4&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Time: 8/7/2019 4:30:57 PM
Section: WindowsHidDeviceFactory
Error: 
--------------------------------------
--------------------------------------
Message: Could not get Hid string. Caller: GetSerialNumber
Time: 8/7/2019 4:30:57 PM
Section: HidAPICalls
Error: 
--------------------------------------
--------------------------------------
Message: Found device \\?\hid#vid_1d50&pid_6122&mi_04#7&37671b16&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Time: 8/7/2019 4:30:57 PM
Section: WindowsHidDeviceFactory
Error: 
--------------------------------------
--------------------------------------
Message: Could not get Hid string. Caller: GetSerialNumber
Time: 8/7/2019 4:30:57 PM
Section: HidAPICalls
Error: 
--------------------------------------
--------------------------------------
Message: Found device \\?\hid#vid_1d50&pid_6122&mi_00#7&155955e2&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Time: 8/7/2019 4:30:57 PM
Section: WindowsHidDeviceFactory
Error: 
--------------------------------------
--------------------------------------
Message: Could not get Hid string. Caller: GetSerialNumber
Time: 8/7/2019 4:30:57 PM
Section: HidAPICalls
Error: 
--------------------------------------
--------------------------------------
Message: Found device \\?\hid#vid_046d&pid_c069#6&483d65f&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Time: 8/7/2019 4:30:57 PM
Section: WindowsHidDeviceFactory
Error: 
--------------------------------------
--------------------------------------
Message: Could not get Hid string. Caller: GetSerialNumber
Time: 8/7/2019 4:30:57 PM
Section: HidAPICalls
Error: 
--------------------------------------
```
I went a little further and added this code after the initial call to `Console.Clear()` in `Menu()`:
```
Console.WriteLine("{0} DeviceFactories are known:", DeviceManager.Current.DeviceFactories.Count);
foreach(var factory in DeviceManager.Current.DeviceFactories) {
   Console.WriteLine("  {0} ({1})", factory.DeviceType, factory.GetType().ToString());
}
```
The output from that additional code is this:
```
2 DeviceFactories are known:
  Usb (Usb.Net.Windows.WindowsUsbDeviceFactory)
  Hid (Hid.Net.Windows.WindowsHidDeviceFactory)
```
I think that fairly clearly indicates that WindowsUsbDeviceFactory is indeed registered.

MelbourneDeveloper, 2019-08-07:
I'll look at this in more detail soon but I just ran the sample and I see a USB device:

[image]

Is it possible that your device is not a winusb device?

MelbourneDeveloper, 2019-08-07:
@Lotharyx can you please put a breakpoint in here: https://github.com/MelbourneDeveloper/Device.Net/blob/1a5c2073fe2ae2b50fe2bd9ab2fbeae413cf85aa/src/Device.Net/Windows/WindowsDeviceFactoryBase.cs#L43

MelbourneDeveloper, 2019-08-07:
@Lotharyx it should get called twice and the variable guidString should be different each time.

Lotharyx, 2019-08-07:
I don't know enough to know whether a given device is a winusb device or not.  How do I check?  I can't exactly argue with your results, but of the six or more non-HID peripherals I have connected, none are winusb?  That seems unlikely as well, but again my expertise is very lacking in this area.

As to your request: The breakpoint is hit twice, with different GUIDs:
guid dee824ef-729b-4a0e-9c14-b7117d33a817
guid 4d1e55b2-f16f-11cf-88cb-001111000030

MelbourneDeveloper, 2019-08-07:
@Lotharyx 

dee824ef-729b-4a0e-9c14-b7117d33a817 is the WinUSB guid. The API call SetupDiGetClassDevs should allow you to enumerate through any WinUSB devices when that guid is passed in.

The only thing I can think of is that the call that is being made is somehow excluding the device. This is the main code:

```cs
var devicesHandle = APICalls.SetupDiGetClassDevs(ref copyOfClassGuid, IntPtr.Zero, IntPtr.Zero, APICalls.DigcfDeviceinterface | APICalls.DigcfPresent);
```

I can only see that there these flags being passed in: 
APICalls.DigcfDeviceinterface | APICalls.DigcfPresent 

This is the documentation for the method:
https://docs.microsoft.com/en-us/windows/win32/api/setupapi/nf-setupapi-setupdigetclassdevsw

I can see the definition of the constants here:
https://www.pinvoke.net/default.aspx/setupapi.setupdigetclassdevs

```cs
        [Flags]
        public enum DiGetClassFlags : uint {
            DIGCF_DEFAULT       = 0x00000001,  // only valid with DIGCF_DEVICEINTERFACE
            DIGCF_PRESENT       = 0x00000002,
            DIGCF_ALLCLASSES    = 0x00000004,
            DIGCF_PROFILE       = 0x00000008,
            DIGCF_DEVICEINTERFACE   = 0x00000010,
        }
```

Could you please try a few different permutations? Perhaps try DIGCF_ALLCLASSES.

Most importantly, check that you see your device here:
[image]

If the device does not appear there in the device manager, it will not be listed.

MelbourneDeveloper, 2019-08-07:
@Lotharyx if you include DIGCF_ALLCLASSES you are likely to get red herrings. There's not much point in including USB devices that are not custom devices. If the device is not a custom device, there will probably be kernel level, or OS level drivers that deal with the device. I.e. there will be some other higher system level APIs that communicate with the device without the need for direct binary transfer of data with the device.

Lotharyx, 2019-08-07:
I'll look further into this within a few days.  Some of the unenumerated devices do show in the "Universal Serial Bus devices" node, but some do not.  Thanks again!

MelbourneDeveloper, 2019-08-19:
@Lotharyx any further information on this? Can you prove that there are Win USB devices connected that are not getting enumerated?

MelbourneDeveloper, 2019-08-20:
@Lotharyx this will issue will be closed in the next day or two

Lotharyx, 2019-08-20:
Oh, I apologize. I have not had any time at all to pursue this on my end. Please do close the issue, as I doubt I’ll be able to give it any attention in the near future.

MelbourneDeveloper, 2019-08-20:
@Lotharyx . Happy to reopen it if you hit up against the same problem in future.

closed 2019-08-20 by MelbourneDeveloper