MelbourneDeveloper/Device.Net #99: Implement class, subclass, and protocol

realivanjx, 2019-08-10:
Hi,

Im trying to implement [python-adb project by google](https://github.com/google/python-adb) in my c# project. I noticed that they use usb class, subclass, and protocol to filter out adb devices as described in [this file](https://github.com/google/python-adb/blob/master/adb/adb_commands.py). I cant see them in your `ConnectedDeviceDefinition` class. Are they not yet implemented or placed somewhere else?

Thanks.

MelbourneDeveloper, 2019-08-11:
@realivanjx great question! 

I have implemented minimal descriptors at this level. You can see where this information is being obtained [here](https://github.com/MelbourneDeveloper/Device.Net/blob/1a5c2073fe2ae2b50fe2bd9ab2fbeae413cf85aa/src/Usb.Net/Windows/WindowsUsbDeviceFactory.cs#L65). It wouldn't be too hard to implement so if you're willing to submit a PR to fix the issue, it would be great. Otherwise, I will look at this for the 3.1 release.

realivanjx, 2019-08-12:
How abt this?

https://github.com/MelbourneDeveloper/Device.Net/blob/1a5c2073fe2ae2b50fe2bd9ab2fbeae413cf85aa/src/Usb.Net/Windows/USB_DEVICE_DESCRIPTOR.cs

realivanjx, 2019-08-12:
Will this work?
https://github.com/MelbourneDeveloper/Device.Net/pull/100

realivanjx, 2019-08-12:
When i tried to build and use the dll from your project i still got these 0s

[image]

Did i miss something?

MelbourneDeveloper, 2019-08-12:
@realivanjx put a breakpoint [here](https://github.com/realivanjx/Device.Net/blob/ba9afbca6e3eea7ee0ac6db4093d33c3b8219d29/src/Usb.Net/Windows/WindowsUsbDeviceFactory.cs#L63) and inspect why the values aren't there:

Have a read of this:
https://docs.microsoft.com/en-us/windows/win32/api/winusb/nf-winusb-winusb_getdescriptor

And this:
https://www.pinvoke.net/default.aspx/Structures.USB_DESCRIPTOR_REQUEST

realivanjx, 2019-08-12:
this is weird. i created a new console app so i can debug your code and put breakpoint but it is not hitting anything inside `GetDeviceDefinition()`. Do you instead getting usb info somewhere else?

[image]

[image]

realivanjx, 2019-08-12:
Also here is breakpoint on `DeviceManager`:

[image]

MelbourneDeveloper, 2019-08-12:
@realivanjx  you are not awaiting the result. You need to await GetConnectedDeviceDefinitionAsync

realivanjx, 2019-08-12:
i used `.Result` already

realivanjx, 2019-08-12:
That field automatically resolves to task's result (replacement of await inside non async function).

[image]

realivanjx, 2019-08-12:
In `DeviceManager.GetConnectedDeviceDefinitionsAsync()` your code tries to call `WindowsUsbDeviceFactory.GetConnectedDeviceDefinitionsAsync()` but i dont see that function implementation in `WindowsUsbDeviceFactory` class.

realivanjx, 2019-08-13:
bingo...

[image]

realivanjx, 2019-08-13:
it seems that the function we are trying to debug is never called anywhere

MelbourneDeveloper, 2019-08-13:
@realivanjx good find. I'm snowed under right now, but should be able to look at this in a couple of days. Do you think you can get your head around the structure enough to fix this issue?

realivanjx, 2019-08-13:
im just stuck on how to get the usb handle and pass it around

realivanjx, 2019-08-13:
i have modified `WindowsUsbDeviceFactory` a bit:

```
using Device.Net;
using Device.Net.Exceptions;
using Device.Net.Windows;
using Microsoft.Win32.SafeHandles;
using System;
using System.Runtime.InteropServices;

namespace Usb.Net.Windows
{
    public class WindowsUsbDeviceFactory : WindowsDeviceFactoryBase, IDeviceFactory
    {
        public ushort? ReadBufferSize { get; set; }
        public ushort? WriteBufferSize { get; set; }

        #region Public Override Properties
        public override DeviceType DeviceType => DeviceType.Usb;
        #endregion

        #region Protected Override Methods

        /// <summary>
        /// The parent Guid to enumerate through devices at. This probably shouldn't be changed. It defaults to the WinUSB Guid. 
        /// </summary>
        protected override Guid GetClassGuid() => WindowsDeviceConstants.WinUSBGuid;
        #endregion

        #region Constructor
        public WindowsUsbDeviceFactory(ILogger logger, ITracer tracer) : base(logger, tracer)
        {
        }
        #endregion

        #region Public Methods
        public IDevice GetDevice(ConnectedDeviceDefinition deviceDefinition)
        {
            if (deviceDefinition == null) throw new ArgumentNullException(nameof(deviceDefinition));

            return deviceDefinition.DeviceType != DeviceType ? null : new UsbDevice(new WindowsUsbInterfaceManager(deviceDefinition.DeviceId, Logger, Tracer, ReadBufferSize, WriteBufferSize), Logger, Tracer);
        }
        #endregion

        #region Private Static Methods
        protected override ConnectedDeviceDefinition GetDeviceDefinition(string deviceId)
        {
            // Getting device interface handle.
            SafeFileHandle sfh = APICalls.CreateFile(deviceId, APICalls.GenericWrite | APICalls.GenericRead, APICalls.FileShareRead | APICalls.FileShareWrite, IntPtr.Zero, APICalls.OpenExisting, APICalls.FileAttributeNormal | APICalls.FileFlagOverlapped, IntPtr.Zero);

            if (sfh.IsInvalid)
            {
                //TODO: is error code useful here?
                int errorCode = Marshal.GetLastWin32Error();
                if (errorCode > 0) throw new ApiException($"Device handle no good. Error code: {errorCode}");
            }

            Logger?.Log(Messages.SuccessMessageGotWriteAndReadHandle, nameof(WindowsUsbInterfaceManager), null, LogLevel.Information);

            var isSuccess = WinUsbApiCalls.WinUsb_Initialize(sfh, out var defaultInterfaceHandle);
            WindowsDeviceBase.HandleError(isSuccess, Messages.ErrorMessageCouldntIntializeDevice);

            return GetDeviceDefinition(defaultInterfaceHandle, deviceId);
        }
        #endregion

        #region Public Static Methods
        /// <summary>
        /// Register the factory for enumerating USB devices in Windows.
        /// </summary>
        public static void Register(ILogger logger, ITracer tracer)
        {
            DeviceManager.Current.DeviceFactories.Add(new WindowsUsbDeviceFactory(logger, tracer));
        }

        public static ConnectedDeviceDefinition GetDeviceDefinition(SafeFileHandle defaultInterfaceHandle, string deviceId)
        {
            var deviceDefinition = new ConnectedDeviceDefinition(deviceId) { DeviceType = DeviceType.Usb };

            var bufferLength = (uint)Marshal.SizeOf(typeof(USB_DEVICE_DESCRIPTOR));
            var isSuccess2 = WinUsbApiCalls.WinUsb_GetDescriptor(defaultInterfaceHandle, WinUsbApiCalls.DEFAULT_DESCRIPTOR_TYPE, 0, WinUsbApiCalls.EnglishLanguageID, out var _UsbDeviceDescriptor, bufferLength, out var lengthTransferred);
            WindowsDeviceBase.HandleError(isSuccess2, "Couldn't get device descriptor");

            if (_UsbDeviceDescriptor.iProduct > 0)
            {
                deviceDefinition.ProductName = WinUsbApiCalls.GetDescriptor(defaultInterfaceHandle, _UsbDeviceDescriptor.iProduct, "Couldn't get product name");
            }

            if (_UsbDeviceDescriptor.iSerialNumber > 0)
            {
                deviceDefinition.SerialNumber = WinUsbApiCalls.GetDescriptor(defaultInterfaceHandle, _UsbDeviceDescriptor.iSerialNumber, "Couldn't get serial number");
            }

            if (_UsbDeviceDescriptor.iManufacturer > 0)
            {
                deviceDefinition.Manufacturer = WinUsbApiCalls.GetDescriptor(defaultInterfaceHandle, _UsbDeviceDescriptor.iManufacturer, "Couldn't get manufacturer");
            }

            deviceDefinition.VendorId = _UsbDeviceDescriptor.idVendor;
            deviceDefinition.ProductId = _UsbDeviceDescriptor.idProduct;
            deviceDefinition.WriteBufferSize = _UsbDeviceDescriptor.bMaxPacketSize0;
            deviceDefinition.ReadBufferSize = _UsbDeviceDescriptor.bMaxPacketSize0;
            deviceDefinition.DeviceClass = _UsbDeviceDescriptor.bDeviceClass;
            deviceDefinition.DeviceSubClass = _UsbDeviceDescriptor.bDeviceSubClass;
            deviceDefinition.DeviceProtocol = _UsbDeviceDescriptor.bDeviceProtocol;

            return deviceDefinition;
        }
        #endregion
    }
}
```

It gets better but i still cant get the fields i need:

[image]

[image]

MelbourneDeveloper, 2019-08-13:
@realivanjx can you please add this to the pull request so I can see the diff?

realivanjx, 2019-08-13:
let me try

realivanjx, 2019-08-13:
can i include the cli test project too? it is in the same folder as usb.net project

MelbourneDeveloper, 2019-08-13:
@realivanjx you should put extra projects like that in a separate PR.

realivanjx, 2019-08-13:
i will remove that once this is done. i will update my pr so u can see the diff.

MelbourneDeveloper, 2019-08-13:
@realivanjx I can only review small changes one at a time.

realivanjx, 2019-08-13:
please see this: https://github.com/MelbourneDeveloper/Device.Net/pull/100/commits/75ee22a28e7a15b7f58e1c815ba9417a9fe8da12

MelbourneDeveloper, 2019-08-19:
@realivanjx , I tried this code: 

```
            deviceDefinition.DeviceClass = _UsbDeviceDescriptor.bDeviceClass;
            deviceDefinition.DeviceSubClass = _UsbDeviceDescriptor.bDeviceSubClass;
            deviceDefinition.DeviceProtocol = _UsbDeviceDescriptor.bDeviceProtocol;
```
But, I just 0s on my device. Do you see anything different on your side?

What's the use case for this? Do you need to filter by these?

realivanjx, 2019-08-19:
Yes im getting zeroes too. They are for my adb implementation.

MelbourneDeveloper, 2021-01-26:
@realivanjx are you still interested in fixing this? Otherwise, I will close this.

realivanjx, 2021-01-26:
I found better implementation of detecting adb devices using class guid so this issue is irrelevant now. Thanks.

closed 2021-01-26 by realivanjx