MelbourneDeveloper/Device.Net #101: IPhone Support

michaelonz, 2019-08-12:
Hi,  Is there any plans to support IPHONE with this library? Pretty much want to use the usb cable on the iphone to communicate to a hardware device.

MelbourneDeveloper, 2019-08-12:
Would love to. Does iPhone even support connected USB devices?

MelbourneDeveloper, 2019-08-12:
@michaelonz , when I've looked in to this in the past, these are the kinds of things I see:

https://www.ios-blog.com/tutorials/iokit-an-introduction/

There is a framework called IO Kit, but the problem is that it doesn't really seem like you're meant to use it for iOS - or at least apps deployed in the store.

MelbourneDeveloper, 2019-08-12:
https://developer.apple.com/documentation/iokit

michaelonz, 2019-08-12:
This link implies it can be done (for ipad - i assume iphone is similar):

https://www.cultofmac.com/636290/every-usb-device-ive-plugged-into-an-ios-13-ipad-so-far/

michaelonz, 2019-08-12:
would be great if this could work :)

MelbourneDeveloper, 2019-08-12:
@michaelonz the first step is leveraging the API in C#. This article doesn't really talk about the API at all. 

Perhaps it's talking about the OS giving IO Kit the permission to actually use the device?

michaelonz, 2019-08-12:
Is this any help:
http://thecodewash.blogspot.com/2017/05/communicating-with-your-ios-app-over.html

MelbourneDeveloper, 2019-08-12:
@michaelonz ah yes, I've seen that one before. It's the only promising one I've come across.

If you are keen on building a POC - i.e. some app that talks to a device, I will find a way to integrate that code in to the Device.Net framework.

For now, this is something I definitely want to do, but all the other stuff in the pipeline also has to get done.

michaelonz, 2019-08-12:
Hi Christian,   Im very busy right now also - once the dust settles i will try and visit this and see if it can work.

MelbourneDeveloper, 2020-08-03:
http://thecodewash.blogspot.com/2017/05/communicating-with-your-ios-app-over.html

https://github.com/rsms/peertalk