MelbourneDeveloper/Device.Net #102: ViewController.Designer.cs missing

bryancostanich, 2019-08-15:
I just cloned the project and I'm trying to run the MacOS sample, but the `ViewController.Designer.cs` file seems to be missing:

[image]

MelbourneDeveloper, 2019-08-15:
Sorry about that @bryancostanich 

That's because of a bug in Visual Studio. It's supposed to be an automatically generated file, but for some reason, it never automatically generates it. I don't have access to a Mac right at this moment so I cannot generate the file.

I'd really appreciate it if you'd take the time to create a new Mac project in Visual Studio which would generate this file. If you could then post the file somewhere or submit a PR, that would help a lot of people.

bryancostanich, 2019-08-15:
hm. i generated the file, but there's a ton of build errors on that project around missing NetStandard stuff. I tried to add the netstandard nuget reference and also tried fiddling with build settings and could never get it to build.

in any case, for anyone in need, here's the content of the `ViewController.Designer.cs` file (you can just create a new .cs file of that name and copy the code in):

```csharp
// WARNING
//
// This file has been generated automatically by Xamarin Studio to store outlets and
// actions made in the UI designer. If it is removed, they will be lost.
// Manual changes to this file may not be handled correctly.
//
using Foundation;

namespace mactest
{
    [Register("ViewController")]
    partial class ViewController
    {
        void ReleaseDesignerOutlets()
        {
        }
    }
}
```

MelbourneDeveloper, 2019-08-15:
@bryancostanich thanks. Did you generate that by creating a fresh Mac Xamarin project?

bryancostanich, 2019-08-15:
Yes.

MelbourneDeveloper, 2019-08-15:
@bryancostanich 

> there's a ton of build errors on that project around missing NetStandard stuff. I tried to add the netstandard nuget reference and also tried fiddling with build settings and could never get it to build.

Sigh... I'm seeing the same thing now. Believe it or not, this did compile at some point in time.

Any chance you could migrate the code from this sample in to your blank Mac project? That's all I did last time, but it seems that something in Xamarin, or Visual Studio has changed. If you can get the sample running, and submit a PR, that would be awesome.

MelbourneDeveloper, 2019-08-15:
@bryancostanich otherwise this goes on the backlog of things I need to get done.

bryancostanich, 2019-08-15:
i'll see what i can do. right now, i can't seem to get notifications working, so this project may be a dead end for me. sleep time now.

MelbourneDeveloper, 2019-08-15:
@bryancostanich what do you mean by notifications?

bryancostanich, 2019-08-15:
re: notifications, i filed an issue.

btw, re: netstandard, it may be that we don't have it installed or something. or maybe vs4mac is just borked.

https://medium.com/@donblas/xamarin-mac-and-netstandard2-708a06890302

MelbourneDeveloper, 2019-08-19:
@bryancostanich I am closing this issue. At some point I will create a new sample from scratch to get it working.

Here is the new issue: https://github.com/MelbourneDeveloper/Device.Net/issues/108

closed 2019-08-19 by MelbourneDeveloper