MelbourneDeveloper/Device.Net #103: Mac: USB hot plug notifications don't seem to work.

bryancostanich, 2019-08-15:
I'm running the `Device.Net.LibUsbSample` console app on a mac, and the initial enumeration seems to find my USB device:

```bash
Currently connected devices: 
usbdev20.30

Console sample. This sample demonstrates either writing to the first found connected device, or listening for a device and then writing to it. If you listen for the device, you will be able to connect and disconnect multiple times. This represents how users may actually use the device.

1. Write To Connected Device
2. Listen For Device
```

But if I choose option `2` and unplug and then re-plug the device, i never get a notification. I also tried it unplugged to start and then chose 2, and then plugged in, and still no notification.

MelbourneDeveloper, 2019-08-15:
@bryancostanich thanks. Which version of the libraries are you using? Is the device Hid, or USB? 

Have you tried directly cloning the Develop (ready for release) branch and directly referencing the libraries?

bryancostanich, 2019-08-15:
USB. Master branch, cloned this evening. Not sure what you're asking re:version of the libs. the project i looked at was v0.1.

Should I be on the `develop` branch?

MelbourneDeveloper, 2019-08-15:
@bryancostanich v0.1 sounds odd. master should be on 3.0.0:
https://github.com/MelbourneDeveloper/Device.Net/blob/1a5c2073fe2ae2b50fe2bd9ab2fbeae413cf85aa/src/Device.Net/Device.Net.csproj#L9

Anyway, if you cloned master, you should pretty much have the latest. I will investigate soon.

bryancostanich, 2019-08-15:
Tried `develop` branch. same results.

MelbourneDeveloper, 2019-08-15:
@bryancostanich will hopefully investigate tomorrow.

bryancostanich, 2019-08-15:
yeah, version `3.0.0`. I forgot that the "version" in the project options dialogue in VS4Mac doesn't actually have any meaning. sigh.

MelbourneDeveloper, 2019-08-19:
@bryancostanich this is working fine with the same sample on Windows. I will test on Mac at some point. Meantime, feel free to debug and have a look at what is going wrong. The LibUsb bridge library is due for a refactor as well.

bryancostanich, 2019-08-19:
I think it's an issue with LibUsbDotNet. I discovered this [issue](https://github.com/LibUsbDotNet/LibUsbDotNet/issues/79) filed yesterday, that explains it.

Unfortunately, it doesn't look like LibUsbDotNet is being maintained anymore. So I suspect this will not be fixed any time soon.

MelbourneDeveloper, 2019-08-19:
@bryancostanich my aim is to create a new, lighter LibUsb wrapper library in the long run for Linux, and OSX. Unfortunately, this will be a big task. I welcome contributions if you're willing to spend time on it. I think that LibUsbDotNet would be a good reference library but the last commit was by last year, so it doesn't seem as though there is active development going on there. 

I did reach out to them for collaboration [here](https://github.com/LibUsbDotNet/LibUsbDotNet/issues/82). But, I never heard back.

bryancostanich, 2019-08-20:
huge task. in fact, probably a bigger task than the entirety of this library, honestly. LibUsb is a monster.

MelbourneDeveloper, 2019-08-20:
@bryancostanich, I've been through the code. Just because libusb has a lot of APIs doesn't mean a good wrapper can't be written. LibUsbDotNet isn't bad. It just needs to be finished. 

If someone is willing to help, it will get done faster.

bryancostanich, 2019-08-20:
I guess I was thinking about the underlying library; LibUsb. It's old, buggy, missing major pieces of functionality, and has a terrible license. Ultimately, a better wrapper is just polishing a turd.

MelbourneDeveloper, 2019-08-20:
@bryancostanich which C/C# cross platform library would you recommend for Linux/OSX?

bryancostanich, 2019-08-21:
I wouldn't. There isn't anything that I know of. That's why I was checking out Device.Net. ;) 

Honestly, at this point, I'm considering putting it into our project queue. USB is just broken.

MelbourneDeveloper, 2019-08-21:
@bryancostanich I see where you are coming from. The options weren't very good when I was looking for a cross platform solution for USB. I could have used LibUsb for the Windows implementation, but it didn't look too great.

But, again, I ask the question: **what is stopping you from using it?** Can you make a list of things that would stop you from using LibUsbDotNet? That library is open source, and can be fixed. Focus on the essentials, and the nice to haves can come later.

MelbourneDeveloper, 2019-08-21:
@bryancostanich are you able to send me one of the devices you are working on? If it's not too expensive, I can have a cracking at getting it to work.

bryancostanich, 2019-08-21:
Happy to send you something. I connected elsewhere. We'll sort that out.

Re: LibUsbDotNet; it's less of an issue of the .net binding, and more of an issue of the underlying library. Notifications, for instance, do not work. Also, it appears that there are a number of other broken things, and LibUsb is not an actively maintained project. It's also released under a bad license for commercial use.

MelbourneDeveloper, 2019-08-21:
@bryancostanich 

> Happy to send you something. I connected elsewhere. We'll sort that out.

Cool. I'll get you an address to send it to in Australia

> Re: LibUsbDotNet; it's less of an issue of the .net binding, and more of an issue of the underlying library.

Understood. Clearly this is an issue. The reason Device.Net even exists is that .NET libraries for connected devices are not well supported, have licensing issues, and are rarely stable. This framework is an attempt to remedy that. I have seen good implementations of LibUsb so it's still an option on the table. I want to take this project on board, but this is probably not something I'd do for free right now as there are so many other issues to sort out in the framework first.

MelbourneDeveloper, 2019-09-21:
@bryancostanich can confirm that this is not working on Mac. The same thing works fine with libusb on Windows. And, I can tell you that the version of libusbdotnet I'm wrapping literally crashes .NET Core. So, I'm looking at the option of building an OSX library from scratch as an option.