MelbourneDeveloper/Device.Net #104: Consider creating a non-"trezor" console device sample

bryancostanich, 2019-08-19:
Trying to test this project out, trying to run the console samples on a mac, but everything seems to be based around some "trezor" device and filters as such. 

Maybe consider pulling all the trezor stuff out of a generic console sample? 

I've been trying to do it myself, but am getting stuck on the filtering API and and device enumeration methods.

MelbourneDeveloper, 2019-08-19:
@bryancostanich wanna send me a device?

I ordered a card reader online so I could create a new sample. When it finally arrived, it did nothing.

bryancostanich, 2019-08-19:
I didn't realize that method was trying to connect.

I think a simple sample like LibUsbDotNet's `Show.Info` would be good to add. That's what I thought that sample basically was, but with some additional read/write.

MelbourneDeveloper, 2019-08-19:
@bryancostanich unfortunately, you're working with the least developed library. Device.Net.LibUsb is a wrapper for LibUsbDotNet, but it is not as developed as the other libraries. This is partly because last time I checked, LibUsbDotNet was in the middle of a major refactor, and neither the released version at the time, nor the develop branch were working properly. 

The long term plan is to implement more information when enumerating devices with `GetConnectedDeviceDefinitionsAsync`. There is a stub [here](https://github.com/MelbourneDeveloper/Device.Net/issues/12) for doing so. However, not all USB, or Hid information is obtained when enumerating devices. There are trade offs because getting information about USB devices requires creating and disposing of handles. But, as more requests for this functionality are made, it will become a higher priority.

Lastly, what you have to understand about Device.Net is that it is a framework with a collection of libraries. It is not just a USB library. The framework aims at putting a seamless layer across all platforms and all device types (e.g. Hid, USB, possibly Bluetooth in future). This requires far more maintenance than one library and that is why the issues list must be rationalized and divided in to versioned releases. In your case, if you do not need cross platform support, and you do not need Hid access, then I recommend using LibUsbDotNet directly until there is time to write a Linux/OSX C# LibUsb wrapper library from scratch. 

However, if you're evaluating Device.Net because you need cross platform support, then you should let me know what your showstopper issues are. If you can tell me what would stop you from using Device.Net, I can focus fixing those issues to get you over the line. Making suggestions is good, but I won't be in any rush to change anything unless I know that you're actually blocked by some specific missing functionality or bug.

MelbourneDeveloper, 2019-08-25:
@bryancostanich I've ordered a GPS device which I'm pretty sure will use USB. I'll try to get that working and add that as a sample. BTW: There are other samples for filtering devices here:

https://github.com/MelbourneDeveloper/KeepKey.Net/blob/14fca783f8e108e75fad2b48a67d4ce5481f6421/src/KeepKey.Net/KeepKeyManagerBroker.cs#L20

https://github.com/MelbourneDeveloper/Ledger.Net/blob/d289426b167c9e852aea60ba5a17545416feabc3/src/Ledger.Net/LedgerManagerBroker.cs#L14

Unfortunately, both these devices are Hid only though.

MelbourneDeveloper, 2021-02-09:
There is now a thermometer 🌡️ sample

closed 2021-02-09 by MelbourneDeveloper