MelbourneDeveloper/Device.Net #105: API inconsistency

bryancostanich, 2019-08-19:
The call `GetDevicesAsync` requires a `List<FilterDeviceDefinition>` in order to make it work, but if you want to get all the devices, it looks like you have to use the `GetConnectedDeviceDefinitionsAsync` method, which, confusingly, also requires the same parameter, but allows a `null` to be passed. The weird thing about this, is that they seemingly do the same thing, but the first returns a `List<IDevice>` and the second returns `IEnumerable<ConnectedDeviceDefinition>`.

Why not just allow `GetDevicesAsync` to take a `null` for all devices? 

Why does GetConnectedDeviceDefinitionsAsync return a different type? I see that `IDevice` contains a `ConnectedDeviceDefinition` anyway?

MelbourneDeveloper, 2019-08-19:
@bryancostanich 

`GetConnectedDeviceDefinitionsAsync` and `GetDevicesAsync` do two different things. 

`GetConnectedDeviceDefinitionsAsync` gets a list of device definitions - not device connections. It lists what is connected, but doesn't attempt to connect to them. This is part of the core framework in that it belongs to `IDeviceFactory`. I believe that the design of this method and interface is good.

`GetDevicesAsync` tries to actually connect to the devices. In most cases you wouldn't want to connect to multiple devices. You will probably only want to connect to one device, and that's how this class is designed. But, more importantly, `GetDevicesAsync` is not part of the core framework interfaces. It's a layer on top that handles enumeration and connection/disconnection. I'm not sure if this method is designed well or not, but the whole `DeviceManager` class needs to some refactoring. I agree it is confusing. I have put a TODO in for refactoring this in [Version 4](https://github.com/MelbourneDeveloper/Device.Net/issues/107). 

> Why not just allow GetDevicesAsync to take a null for all devices?

Is there a real life use case where you'd want to connect to all devices at the same time?

> Why does GetConnectedDeviceDefinitionsAsync return a different type?

See the first point.

bryancostanich, 2019-08-19:
I don't necessarily need to connect, I just need to enumerate all devices. I'm guess that's likely a pretty common use case. Especially given that's the what one of the main samples does in LibUsb. In that case, it sounds like I need `GetConnectedDeviceDefinitionsAsync`. I suggest adding an overload that doesn't require a filter.

I was confused by the `GetDevicesAsync` because there's no "connect" verb in the name, or any indication that it does anything other than get the devices. So it wasn't readily apparent that I was connecting. I'm not sure I even understand the use case in which you would filter and connect all at once, but it sounds like you're using it with your crypto stuff. In that case, I suggest a method rename. Just my 2¢. :)

MelbourneDeveloper, 2019-08-19:
> I just need to enumerate all devices.

That's simple. You don't need `GetDevicesAsync`. [This article](https://github.com/MelbourneDeveloper/Device.Net/wiki/Enumerating-Connected-Devices) explains how to do it with `GetConnectedDeviceDefinitionsAsync`

> I was confused by the GetDevicesAsync because there's no "connect" verb in the name

Fair call.

> I'm not sure I even understand the use case in which you would filter and connect all at once, but it sounds like you're using it with your crypto stuff.

There is no use case as far as I know. No. I don't do that with any crypto stuff. I was answering your question:

> Why not just allow GetDevicesAsync to take a null for all devices?

So, I think we can say that it doesn't make sense to have a method that accepts no filters that would connect to all devices at once.

@bryancostanich

MelbourneDeveloper, 2019-08-19:
@bryancostanich 

> I suggest a method rename

Fair call. Please make your suggestion for a new name [here](https://github.com/MelbourneDeveloper/Device.Net/issues/107). I've already created a stub for it. Just bear in mind that what it is doing is actually initializing the device. So, a less misleading name would probably include "Initialize"

AalapMistry, 2019-12-13:
Status Bit:1
07FFFFFFFFFFFFFF
Line No:2
High:0
Low:16
Status Bit:1
070011D9FC800100
Line No:3
High:0
Low:24
Status Bit:1
07011213D7FFFFFF
Line No:4
High:0
Low:32
Status Bit:1
07029A0100001100
Line No:5
High:0
Low:40
Status Bit:1
0703000000000000
Line No:6
High:0
Low:48
Status Bit:1
0704C22200000000
Line No:7
High:0
Low:56
Status Bit:1
0705640000000C00
Line No:8
High:0
Low:64
Status Bit:1
0706000000000000
Line No:9
High:0
Low:72
Status Bit:1
0707112000000000
Line No:10
High:0
Low:80
Status Bit:1
0708960000001600
Line No:11
High:0
Low:88
Status Bit:1
0709000000000000
Line No:12
High:0
Low:96
Status Bit:1
070A121A00000000
Line No:13
High:0
Low:104
Status Bit:1
070B640000000C00
Line No:14
High:0
Low:112
Status Bit:1
070C000000000000
Line No:15
High:0
Low:120
[... 771 lines cut ...]
07ADFFFFFFFFFFFF
Line No:170
High:5
Low:80
Status Bit:1
07AE00000000401F
Line No:171
High:5
Low:88
Status Bit:1
07AF803EC05D007D
Line No:172
High:5
Low:96
Status Bit:1
07B0FFFFFFFFFFFF
Line No:173
High:5
Low:104
Exception thrown: 'System.IO.IOException' in mscorlib.dll

Reading small frame lines from device it work but reading more frame lines it throw exception or device disconnected .

Help  i am reading data from above address High Low and random throw exeption while reading if more data frame line , else works fine if small lines frame data. 
Help or suggest me what to do ?

st-arnold, 2020-01-14:
I was confused because of this as well. Good to hear there is improvement.

I found another API inconsistency in WinUsbApiCalls.cs:
Some P/Invoke signatures have IntPtr type for InterfaceHandle parameter while others have SafeFileHandle.

MelbourneDeveloper, 2020-01-15:
@AalapMistry  could you please tell me what that log is from? I don't really understand...

@st-arnold , yes there are some APIs that are not designed perfectly. I'm very much open to code-design suggestions if you've got some ideas. If you want to test, you could try changing everything to SafeFileHandle.

AalapMistry, 2020-01-15:
Please close that ticket , the issue is from firmware side ,Thanks for
replay

MelbourneDeveloper, 2020-11-21:
Hi @bryancostanich 

Just letting you know that this issue is being closed because the API has been completely redesigned. V4 will be out soon.

closed 2020-11-21 by MelbourneDeveloper