MelbourneDeveloper/Device.Net #109: Always get SEHException with visual studio debug

realivanjx, 2019-08-21:
**Describe the issue**
I always got `System.Runtime.InteropServices.SEHException` if i run my app (test console app) with visual studio debugger. If i run without debugger, this error never appear. I suspect that it is thrown when im trying to dispose the `IDevice` object:

```
m_deviceService.Device.Dispose();
```

or even with this code:

```
await DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(null);
```

Whereas `m_deviceService` is an instance of `BulkIOService`. I dont know which part of my code causes this error because visual studio says that it is executing external code (see screenshot below). I dont know if this is a real issue but it is quite annoying to always get it when debugging. Any idea?

**Screenshots**

[image]

Platform: Windows 10
Version: 3.0.0 beta 2

MelbourneDeveloper, 2019-08-21:
@realvanjx thanks for reporting. Yes, I've seen this one before. I'm not sure if it's a real issue myself...

Could you try including the device.net projects in your solution directly? Instead of the NuGet. I need to see the exact line.

realivanjx, 2019-08-21:
I dont think that it will show the line either because i think it errors in the c++ api you are using.

MelbourneDeveloper, 2019-08-21:
@realivanjx you're probably right. I had the same issue. Do you have "enable just my code" turned on? Check that it debugging options.

realivanjx, 2019-08-21:
Yes it is on already

MelbourneDeveloper, 2019-08-21:
@realivanjx the only thing I can recommend is to turn off the exception in the debugging options. I will look out for it and try to find out which API is causing the problem.

realivanjx, 2019-09-05:
i have tried to disable the exception in debugging options but that doesnt work. the app also keeps closing when i hit continue on visual studio after the error. this makes me impossible to debug the app now

realivanjx, 2019-09-05:
never got this issue when running without debugging though

realivanjx, 2019-11-23:
is there any updates on this issue?

MelbourneDeveloper, 2019-11-23:
@realivanjx I probably won't fix this issue because it doesnt cause any serious problems that I know about. 

It's only an annoyance. As far as I know, it only causes an issue for debugging right?

realivanjx, 2019-11-23:
yea, but it affects me so that i cant debug the app at all now

realivanjx, 2019-11-23:
so for me it kinda serious

realivanjx, 2019-11-23:
maybe this could be the issue with visual studio itself? because this only happens while debugging though

realivanjx, 2019-11-23:
[image]

st-arnold, 2020-01-14:
Same problem here... probably it has something to do with SafeFileHandles, as VS was trying to locate source code for that (if Just My Code in debugger settings wasn't enabled).

MelbourneDeveloper, 2020-01-15:
@st-arnold you're probably right. I'd prefer that everything were converted to safe handles if possible

datoml, 2020-03-09:
I can confirm that this only happens during a debugger is attached.
You can easily reproduce the problem when you start for instance a reading process from a device and unplug the USB during that.
Reattach the USB and start again a reading process.
Then this SEHException occurs.
As said, when a debugger is attached to the app crashes.
Without one it running without any issue.

MelbourneDeveloper, 2021-01-26:
@st-arnold @datoml @realivanjx 

This issue may have been because of an issue with dispose. Any chance you could try the latest version to see if you still get the issue?

https://www.nuget.org/packages/Usb.Net/4.0.8-alpha

NorbiPeti, 2021-06-17:
I still have this issue using Usb.Net 4.0.9-alpha and Rider for debugging. The stack trace references SafeHandle.Finalize() (and none of my code). Disabling external code debugging seems to solve the issue.

MelbourneDeveloper, 2021-06-17:
@NorbiPeti yep. No idea what this or how to fix it. Suggestions welcome.

MelbourneDeveloper, 2021-06-17:
@NorbiPeti have you disabled "Just My Code" in visual studio debugging options?