MelbourneDeveloper/Device.Net #120: ConnectedDeviceProperties like Manufactrurer, ProductName, SerialNumber contain wrong characters (Windows 7)

JuryOberst, 2019-09-27:
**Issue**
I had a problem on some Windows 7 computers. Everytime I tried to receive the ConnectedDeviceDefinitions like ProductName, Manufacturer or SerialNumber the first char of the property was replaced with charecters that were not part of the original string.

The first char changed on every request of the property.

**Example**
For example a manufacturer original string was "OHLO HID Connector" but after I received the string it changed to "µHLO HID Connector" or something similar.

**Thoughts**
I think the problem could be, that the memory space is released before calling the method "PtrToStringAuto", so an other task have time to override a part of the memory.

**Fix**
I  think the changes in the pull request #119 should fix  the issue.

MelbourneDeveloper, 2019-09-27:
@JuryOberst thanks. I noticed that you also switched from the unicode version to the Auto version of the method. I tested with both the old way and your new way and both had the same result. I can't say I've ever tested on Windows 7 so I'll have to take your word for it. I'm not sure what the correct code should be but I hope this fixes your issue. I probably won't be doing a release for a while so let me know how you go.

MelbourneDeveloper, 2019-09-27:
Note: https://github.com/MelbourneDeveloper/Device.Net/projects/12

closed 2019-10-08 by MelbourneDeveloper