MelbourneDeveloper/Device.Net #121: When enable HID and Serial Port in windows,  GetConnectedDeviceDefinitionsAsync return error

fengqiangboy, 2019-11-01:
**Your Code**

```c#
// enable HidDevice and SerialPort here
var debugLogger = new DebugLogger();
var debugTracer = new DebugTracer();
WindowsHidDeviceFactory.Register(debugLogger, debugTracer);
WindowsSerialPortDeviceFactory.Register(debugLogger, debugTracer);

var hidFilterDeviceDefinition = new FilterDeviceDefinition
                {DeviceType = DeviceType.Hid, VendorId = 0x0483, ProductId = 0x5750};

// here, hidDevices  will include SerialPortDevice
var hidDevices = await DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(hidFilterDeviceDefinition);

var serialPortFilterDeviceDefinition = new FilterDeviceDefinition {DeviceType = DeviceType.SerialPort};

// serialPortDevices' deviceType is null
var serialPortDevices = await DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(serialPortFilterDeviceDefinition);
```

**Info**
 - Platform: Windows,
 - Device Type: Hid,  SerialPort
 - Version: 3.1.0

MelbourneDeveloper, 2019-11-01:
@fengqiangboy focus one device first. Remove the code for serial ports and try to use the Hid device.

MelbourneDeveloper, 2019-11-01:
Try passing null to GetConnectedDeviceDefinitionsAsync so you get all Hid devices.

MelbourneDeveloper, 2019-11-01:
@fengqiangboy please read this and include the log in your post: 

https://github.com/MelbourneDeveloper/Device.Net/wiki/Debugging,-Logging,-and-Tracing

MelbourneDeveloper, 2019-11-01:
@fengqiangboy 

See this one:
https://github.com/MelbourneDeveloper/Device.Net/wiki/Debugging,-Logging,-and-Tracing#i-cant-see-my-device

fengqiangboy, 2019-11-01:
> @fengqiangboy focus one device first. Remove the code for serial ports and try to use the Hid device.

It's ok when only use Hid device.

MelbourneDeveloper, 2019-11-01:
@fengqiangboy so the issue is just serial port access?

fengqiangboy, 2019-11-01:
> @fengqiangboy so the issue is just serial port access?

Yes, it only happen when I add serial port access.

MelbourneDeveloper, 2019-11-01:
@fengqiangboy please run the windows sample and check to see if the COM port is displayed

fengqiangboy, 2019-11-01:
> @fengqiangboy please run the windows sample and check to see if the COM port is displayed

Yes, the COM port is displayed.

[image]

MelbourneDeveloper, 2019-11-01:
@fengqiangboy ok . Maybe it's a bug...

Please post your log

MelbourneDeveloper, 2019-11-01:
@fengqiangboy 

Please try this code
https://github.com/MelbourneDeveloper/Device.Net/blob/740659228fece1a0d17b2ebf2541e0c697b0df5e/src/Device.Net.UnitTests/IntegrationTests.cs#L109

MelbourneDeveloper, 2019-12-09:
@fengqiangboy is this still an issue?

Can you please post your log and try the code above?

carl-qva, 2020-01-15:
Hi MelbourneDeveloper, im having an issue with Device.Net (and in general with all hid libraries that i've tested so far). The problem is that my app can communicate with my hid device only in administrator mode, when the app is not run in admin mode the device is not found. In admin mode i can read/write data without any problem. See this log from Device.Net and please help

region: WindowsHidDeviceFactory - GetConnectedDeviceDefinitionsAsync, message: About to call SetupDiGetClassDevs for class Guid 4d1e55b2-f16f-11cf-88cb-001111000030. Flags: 18, LogLevel: Information
region: ApiService, message: Calling CreateFile for DeviceId: \?\hid#vid_0483&pid_a2ca&mi_00#7&1a31e08c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}. Desired Access: None. Share mode: 3. Creation Disposition: 3, LogLevel: Information
Exception thrown: 'Device.Net.Exceptions.DeviceException' in Hid.Net.dll
region: WindowsHidDeviceFactory, message: GetDeviceDefinition error. Device Id: \?\hid#vid_0483&pid_a2ca&mi_00#7&1a31e08c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}, LogLevel: Error, exception: CreateReadConnection call with Id of \?\hid#vid_0483&pid_a2ca&mi_00#7&1a31e08c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} failed.
region: WindowsHidDeviceFactory, message: Device with path \?\hid#vid_0483&pid_a2ca&mi_00#7&1a31e08c&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030} was skipped. See previous logs., LogLevel: Warning

MelbourneDeveloper, 2020-01-15:
@carl-qva I get a lot of messages like this. I think Microsoft has started putting security around devices so that some calls don't work. It makes sense. Users who are not admins shouldn't really have access to devices. UWP has a better system where the user is forced to accept that the app is going to access the device. This problem has never really been solved at the Windows level but I think that Microsoft are working on it. Meantime, I think your app will have to specify that it needs those privileges.

carl-qva, 2020-01-15:
> @carl-qva I get a lot of messages like this. I think Microsoft has started putting security around devices so that some calls don't work...

Thank you very much for your quick reply. I think that in several cases a program running without privileges should be able to use devices. I´ll try to find some workaround

Greetings from HavanaDeveloper ;)

carl-qva, 2020-01-15:
> Thank you very much for your quick reply. I think that in several cases a program running without privileges should be able to use devices. I´ll try to find some workaround
>
> Greetings from [...]

I was thinking right now that i could install a service along with my app. The service will be running with privileges and can communicate with the device, my app will be running without privileges and can communicate with the service through a socket. That way, at installation time the elevated privileges will be needed, and after installation, any user without privileges running a non-elevated instance of my app can communicate with the device through the socket using the service that will be running with privileges... that could do the job right??

MelbourneDeveloper, 2020-12-13:
@carl-qva I expect that is true.

@fengqiangboy I am closing this issue. If you still have issues in the latest version, please let me know and I will reopen the issue.

closed 2020-12-13 by MelbourneDeveloper