MelbourneDeveloper/Device.Net #126: USB Control transactions

networkfusion, 2020-01-14:
I am trying to create a .netcoreapp (2.1) to update firmware on ST devices i.e. http://eliasoenal.com/st%20website/17068.pdf. All of the examples I have seen use control transactions e.g.
```
            UsbSetupPacket setup = new UsbSetupPacket()
            {
                RequestType = (byte)(DFU_RequestType | USB_DIR_IN),
                Request = (byte)DFU_UPLOAD,
                Value = (short)blockNum,
                Index = 0
            };
            usb.ControlTransfer(ref setup, data, length, out int len);
```

Is this possible with this lib? If so, are there any examples?

Thanks.

MelbourneDeveloper, 2020-01-15:
@networkfusion I wish I could give you a clearer answer but the truth is I can't answer this right now. I know that the main types I've worked with a bulk and interrupt. If it's possible, it will be in this area of theo code: https://github.com/MelbourneDeveloper/Device.Net/blob/740659228fece1a0d17b2ebf2541e0c697b0df5e/src/Usb.Net/Windows/WindowsUsbInterfaceManager.cs#L40

If you can point to other code in another repo I will consider grabbing it and merging it in.

networkfusion, 2020-01-16:
Okay, I have tried the latest preview release of https://github.com/LibUsbDotNet/LibUsbDotNet nuget and it seems to offer everything I need. I will close this issue for now, but will add a link to my project once I have a working example of how I am using it, so you can choose to add support in future if interested. Thanks for your swift response.

MelbourneDeveloper, 2020-01-16:
@networkfusion just some points about LibUsb. It's been a defacto standard across platforms for a long time, but a lot of people have trouble with licensing, and LibUsbDotNet hasn't been updated since 2018.

You could try this wrapper: https://github.com/MelbourneDeveloper/Device.Net/tree/master/src/Device.Net.LibUsb . It's not complete, but I would welcome a PR to upgrade it to do what you need.

The advantage would be that you keep the Device.Net abstraction layer. Device.Net is designed for dependency injection and I plan to continue to follow best practice to enable usage with IoC containers and so on for modern usage. If you ever need something that's compatible with HID or COM port access, these abstractions are your best bet. 

Thanks for letting me know and I will try to implement the functionality you need in future.

networkfusion, 2020-01-16:
Thanks for this, the deeper I get into the rabbit hole, the harder it gets... I have uploaded what I want to achieve here (https://github.com/networkfusion/St-Dfu), but it is still causing issues due to the need to install a filter driver on windows. I will also be looking at what is needed to get it working with this lib instead anyway... as such, I am going to reopen the issue...

MelbourneDeveloper, 2020-01-21:
@networkfusion I hope to take a look at this this weekend.

MelbourneDeveloper, 2020-07-28:
@networkfusion I would like to take a look at this sometime soon. Is it easy to get hold of these devices? Is there another type of device I could test this functionality on?

networkfusion, 2020-07-30:
@MelbourneDeveloper https://www.st.com/en/evaluation-tools/nucleo-f746zg.html is a nice cheap board that nanoFramework can provide a DFU package for. The nanoFramework project is also willing to provide a board (probably different from above) free of charge if it helps. If you would like to take up the offer, let me know, and we can arrange it (but would somehow need you to let me know your address).

MelbourneDeveloper, 2020-07-30:
@networkfusion if I get the device, is there a clear test I can do? 

I.e. I send it some bytes, and there is a set array of bytes I can expect back?

networkfusion, 2020-07-30:
I will have to remember how far I got with the original code (it has been a while). I will try and do that over the next couple of days and let you know.

MelbourneDeveloper, 2020-07-30:
If you can write a clear integration test that fails, that would be really good. I.e. specify the data in and specify the expected result, I should be able to do the rest. 

You can send the device to a post office box. Can you hit me up on Twitter for a dm?

Cfdevelop

networkfusion, 2020-07-30:
Thanks, I will bare that in mind. BTW, here is the protocol spec document https://www.st.com/resource/en/application_note/cd00264379-usb-dfu-protocol-used-in-the-stm32-bootloader-stmicroelectronics.pdf, and here is some example code I threw up: ```        private void GetStatus(DfuStatus status) //throws Exception
        {
            byte[] buffer = new byte[6];
            //int length = usb.controlTransfer(DFU_RequestType | USB_DIR_IN, DFU_GETSTATUS, 0, 0, buffer, 6, 500);
            LibUsbDotNet.Main.UsbSetupPacket setup = new LibUsbDotNet.Main.UsbSetupPacket()
            {
                RequestType = (byte)(DFU_RequestType | USB_DIR_IN),
                Request = (byte)DFU_GETSTATUS,
                Value = 0,
                Index = 0
            };
            int length = usb.ControlTransfer(setup, buffer, 0, 6);

            if (length < 0)
            {
                throw new Exception("USB Failed during getStatus");
            }
            status.bStatus = buffer[0]; // state during request
            status.bState = buffer[4]; // state after request
            status.bwPollTimeout = (buffer[3] & 0xFF) << 16;
            status.bwPollTimeout |= (buffer[2] & 0xFF) << 8;
            status.bwPollTimeout |= (buffer[1] & 0xFF);
        }``` from https://github.com/networkfusion/St-Dfu/blob/master/ST-DFU/Dfu.cs

piwi1263, 2020-08-26:
@MelbourneDeveloper Hi Christian can you DM me on Twitter (@piwi1263) with the address I can send the above mentioned board to ?

Thanks, Peter.

MelbourneDeveloper, 2020-08-26:
@piwi1263 I've responded on Twitter. Are you working with @networkfusion ?

piwi1263, 2020-08-27:
Yeah, we both are into [nanoFramework](https://github.com/nanoframework) an effort to have c# run on little, resource constrained MCUs

MelbourneDeveloper, 2020-08-27:
@piwi1263 cool. I guess that's the other end of what I do.

networkfusion, 2020-08-28:
FYI, Here is the necessary implementation code from libUsbDotNet https://github.com/LibUsbDotNet/LibUsbDotNet/blob/ba4af0121ac8259368e47bc33f0f3021322317f8/src/LibUsbDotNet/LibUsb/UsbDevice.DeviceHandle.cs#L153

I am still struggling to find some code that has fixed bytes for send and receive, but I will try and create a PR that works against my DFU program (if it helps)...

MelbourneDeveloper, 2020-12-13:
@networkfusion  Control Transactions are now available for Windows in Usb.Net

closed 2020-12-13 by MelbourneDeveloper