MelbourneDeveloper/Device.Net #127: Alternate interfaces

st-arnold, 2020-01-14:
I suggest adding support for alternate interfaces.

I was trying out this library for connecting to Silicon Labs CP2615 chip and using its "I/O protocol".
After much investigation, trial and error I got it working, but it required some "dirty" modifications to the lib code. One of them was adding P/Invoke to WinUsb_SetCurrentAlternateSetting which is required to enable the I/O protocol endpoints in the chip by selecting alternative interface.
Also, enumerating interfaces and endpoints of this device didn't work as expected in WindowsUsbInterfaceManager.Initialize(). The default interface returned by WinUsb_Initialize was interface number 1 and WinUsb_GetAssociatedInterface didn't give the other interfaces (there are 5 in total), just returned ERROR_NO_MORE_ITEMS. This problem I couldn't solve.
For that interface 1 the endpoints were not enumerated because alternate 0 (the only one checked by GetInterface) doesn't have them while alt 1 and 2 do...

Here's device info from USBTreeView if you're interested in the descriptor structure:

[cp2615_UsbTreeView.txt](https://github.com/MelbourneDeveloper/Device.Net/files/4062108/cp2615_UsbTreeView.txt)

MelbourneDeveloper, 2020-01-15:
@st-arnold can you submit a PR or point me to the code?

HID devices should show up if you register the windows hid factory. The Windows sample should list them.

st-arnold, 2020-01-15:
I'm not so experienced GitHub user and I manually downloaded the repo so I don't know how to make a PR... And my changes are quick&dirty hard coded for my proof-of-concept testing, breaking functionality for other uses.
But here are attached the two Usb.Net/Windows files I modified. My changes are marked with //d comments.

[Windows.zip](https://github.com/MelbourneDeveloper/Device.Net/files/4065368/Windows.zip)

For simplicity's sake I left out HID in my project in which I only need to access the CP2615 I/O protocol endpoints. Probably I could get the HID interface listed if I included HID lib & factory.

MelbourneDeveloper, 2020-12-13:
@st-arnold I am closing this issue for now. If you are still experiencing issues with the new version which is 4.0.2-alpha, please let me know and I will investigate.

closed 2020-12-13 by MelbourneDeveloper