MelbourneDeveloper/Device.Net #128: WinUSB GUID cannot get device found.

Halry, 2020-02-15:
**Describe the issue**
I made a USB device which is using raw USB,then I can get it enumerated in Windows 10 by installing driver using zadig.But no luck in Windows 7,device cannot be found in "GetConnectedDeviceDefinitionsAsync"->"SetupDiGetClassDevs",then I change USB descriptor to make it become a WCID device which compatible with Microsoft OS Description 1.0 but after changes,I can't get it enumerated both in Windows 10 and Windows 7.I then change the 
GetClassGuid() from WindowsDeviceConstants.WinUSBGuid to WindowsDeviceConstants.GUID_DEVINTERFACE_USB_DEVICE,now everything is working.

- Screenshot of where the exception is occurring (See [Break On All Exceptions](https://christianfindlay.com/2019/07/14/visual-studio-break-on-all-exceptions/) and [this](https://github.com/MelbourneDeveloper/Device.Net/wiki/Debugging,-Logging,-and-Tracing) )
no exception occurred.
- Screenshot of the Windows Sample. I.e. run the windows sample and post the console window here.
[image]
[image]
**Your Code**
Post your code here. Please use markdown so that the code is formatted.
```
public class USB:IDisposable
    {
        private readonly List<FilterDeviceDefinition> _DeviceDefinitions = new List<FilterDeviceDefinition>
        {
            new FilterDeviceDefinition{DeviceType=DeviceType.Usb,VendorId=0xffff,ProductId=0xf0f0}
        };

        public IDevice USB_Device { get; private set; }
        public DeviceListener DeviceListener { get; }
        public USB()
        {
            DeviceListener = new DeviceListener(_DeviceDefinitions, 1000) { Logger = new DebugLogger() };
        }

        public event EventHandler DeviceConnected;
        public event EventHandler DeviceDisconnected;
        //event handler
        private void DevicePoller_DeviceConnected(object sender,DeviceEventArgs e)
        {
            USB_Device = e.Device;
            DeviceConnected?.Invoke(this, new EventArgs());
        }
        private void DevicePoller_DeviceDisconnected(object sender,DeviceEventArgs e)
        {
            USB_Device = null;
            DeviceDisconnected?.Invoke(this, new EventArgs());
        }
        public void StartListening()
        {
            USB_Device?.Close();
            DeviceListener.DeviceDisconnected += DevicePoller_DeviceDisconnected;
            DeviceListener.DeviceInitialized += DevicePoller_DeviceConnected;
            DeviceListener.Start();
        }
        public void Dispose()
        {
            DeviceListener.DeviceDisconnected -= DevicePoller_DeviceDisconnected;
            DeviceListener.DeviceInitialized -= DevicePoller_DeviceConnected;
            DeviceListener.Dispose();
            USB_Device?.Dispose();
        }
        public async Task InitDeviceAsync()
        {
            var device = await DeviceManager.Current.GetDevicesAsync(_DeviceDefinitions);
            USB_Device = device.FirstOrDefault();
            if (USB_Device == null) throw new Exception("There were no devices found");
            await USB_Device.InitializeAsync();
        }
````
**Log / Stack Trace**
Please include the stack trace of where the problem occurred as text or a screenshot. You *must* use logging and include the output here. Please see [this](https://github.com/MelbourneDeveloper/Device.Net/wiki/Debugging,-Logging,-and-Tracing)
WinUSB:
```
--------------------------------------
Message: About to call SetupDiGetClassDevs for class Guid dee824ef-729b-4a0e-9c14-b7117d33a817. Flags: 18
Level: Information
Time: 2/15/2020 19:58:03
Section: WindowsUsbDeviceFactory - GetConnectedDeviceDefinitionsAsync
Error: 
--------------------------------------
--------------------------------------
Message: The call to SetupDiEnumDeviceInterfaces returned ERROR_NO_MORE_ITEMS
Level: Information
Time: 2/15/2020 19:58:03
Section: WindowsUsbDeviceFactory - GetConnectedDeviceDefinitionsAsync
Error: 
--------------------------------------
--------------------------------------
Message: Poll complete
Level: Information
Time: 2/15/2020 19:58:03
Section: DeviceListener - CheckForDevicesAsync
Error: 
--------------------------------------
```
GUID_DEVINTERFACE_USB_DEVICE:
```
--------------------------------------
Message: About to call SetupDiGetClassDevs for class Guid a5dcbf10-6530-11d2-901f-00c04fb951ed. Flags: 18
Level: Information
Time: 2/15/2020 19:59:01
Section: WindowsUsbDeviceFactory - GetConnectedDeviceDefinitionsAsync
Error: 
--------------------------------------
--------------------------------------
Message: The call to SetupDiEnumDeviceInterfaces returned ERROR_NO_MORE_ITEMS
Level: Information
Time: 2/15/2020 19:59:01
Section: WindowsUsbDeviceFactory - GetConnectedDeviceDefinitionsAsync
Error: 
--------------------------------------
--------------------------------------
Message: Attempting to initialize with DeviceId of \\?\usb#vid_ffff&pid_f0f0#1111111111111#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Level: Information
Time: 2/15/2020 19:59:01
Section: DeviceListener - CheckForDevicesAsync
Error: 
--------------------------------------
--------------------------------------
Message: Successfully opened handle on device for reading and writing
Level: Information
Time: 2/15/2020 19:59:01
Section: WindowsUsbInterfaceManager
Error: 
--------------------------------------
Message: Device connected
Level: Information
Time: 2/15/2020 19:59:01
Section: DeviceListener - CheckForDevicesAsync
Error: 
--------------------------------------
--------------------------------------
Message: Poll complete
Level: Information
Time: 2/15/2020 19:59:01
Section: DeviceListener - CheckForDevicesAsync
Error: 
--------------------------------------
```
**Info**
 - Platform: [e.g. Android] Windows(Not UWP)
 - Device Type: [e.g. USB/Hid]WinUSB
 - Version: [e.g. 3.0]
My Device:
[image]

MelbourneDeveloper, 2020-02-16:
@Halry thanks for your documentation. I will try to look at this soon. I'm currently busy with several projects, but Device.Net is next on my list.

Halry, 2020-02-16:
I think you should left a option to let people select which guid should be used:)

MelbourneDeveloper, 2020-02-16:
@Halry it's a good idea in theory but that guid can distinguish between USB and hid for example. If I open it up, there needs to be way too make sure that the values can still be filtered be the type.

Halry, 2020-02-19:
Maybe you can use "UsagePage" to determined whether it is a USB device or HID device.

MelbourneDeveloper, 2020-07-12:
@Halry any chance you could submit a PR for this?

Halry, 2020-07-20:
> @Halry any chance you could submit a PR for this?

Sorry, but I don't know how to use "UsagePage" to determine whether it is an USB type or HID. I only familiar with C. ;)

MelbourneDeveloper, 2020-12-13:
@Halry you can now specify the class guid for factories like this:

https://github.com/MelbourneDeveloper/Device.Net/blob/13b7f8104b411a95e7e279aa5dbab604cd1689b2/src/Device.Net.UnitTests/IntegrationTests.cs#L48

I am closing this issue for now, but if it doesn't fix your problem, please let me know.

closed 2020-12-13 by MelbourneDeveloper