MelbourneDeveloper/Device.Net #135: System.DllNotFoundException

sbogutzky, 2020-07-29:
I use macOS 10.15.5 and Visual Studio for Mac Community 8.6.8 (build 2). I installed libusb-1.0.23. 

"libusb-1.0 library not found.  This is often an indication that libusb-1.0 was installed to '/usr/local/lib' and mono.net is not looking for it there. To resolve this, add the path '/usr/local/lib' to '/etc/ld.so.conf' and run 'ldconfig' as root. (http://www.mono-project.com/DllNotFoundException)"

How can I fix this problem on my Mac?

MelbourneDeveloper, 2020-07-29:
It usually just means that you need to install LibUsb. 

You can do it with homebrew

sbogutzky, 2020-07-29:
I installed libusb via homebrew. 

/usr/local/Cellar/libusb/1.0.23: 26 files, 474.2KB

How can I add the path of libusb to the mono search path on my Mac.

MelbourneDeveloper, 2020-07-29:
@sbogutzky OK. So, I'd recommend running the LibUsb sample in the MacOS samples. I tried it out from the master branch on a clean mac. This is the error I get. It's look like the same error as yours.

[image]

The error points to this resource:
https://www.mono-project.com/docs/advanced/pinvoke/dllnotfoundexception/

I installed 1.0.23 with homebrew, and then I still get the error. 

When I look in the folder that it mentions, I see this:
[image]

The etc folder is missing ld.so.conf
[image]

I'm thinking that perhaps an update to macOS has disturbed something. I tried using the terminal tool that the documentation suggests called ldconfig but I couldn't find that tool.

I suggest reading through mono resource mentioned above. I also suggest getting the latest version of LibUsbDotNet and trying out their latest samples. If their sample works, I will pull the latest from there in to Device.Net and get it working.

sbogutzky, 2020-07-30:
The mono resource above only describe how to fix it on a linux system. Ldconfig is not available on macOS. The installation of libusb is not the problem. But how can we configure mono on macOS to search in the path /usr/local/bin for libusb?

MelbourneDeveloper, 2020-07-30:
@sbogutzky I don't know. I would ask on the LibUsbDotNet GitHub page. Please let me know if you get it working.

MelbourneDeveloper, 2021-01-26:
@sbogutzky Hi. I'm closing this issue because I believe that it is LibUsbDotNet related.

Still, if you have fixed this issue, a PR would be welcome.

closed 2021-01-26 by MelbourneDeveloper