MelbourneDeveloper/Device.Net #155: More/Better Documentataion

carcarjg, 2020-09-24:
The current documentation is good. But I feel it could be better.

The main points are
https://github.com/MelbourneDeveloper/Device.Net/blob/master/src/Usb.Net.WindowsSample/Program.cs
- On line 22 your example where did TrezorExample come from? this seems like an important thing to know when you need to read and write data from different tasks other than the initialization one.

- PDF or the Like documentation, I would prefer class/method documentation. as the wiki articles, you have currently are more suited as Getting Started articles. Run Doxygen and have what that spits out somewhere.

Other than those two big points everything looks great! Don't take this negatively, as I would much rather have what you have posted over the nothingness that exists on so many other packages.

MelbourneDeveloper, 2020-09-24:
@carcarjg thanks. That's good feedback. I will try to include this information in the next upgrade which is coming reasonably soon.

Also, there is this [issue](https://github.com/MelbourneDeveloper/Device.Net/issues/134). At the very least, I will try to get the existing XML doc into the NuGet package this time around. I will also look into Doxygen. Is it possible to host the output of Doxygen on Github pages? Why Doxygen? Is it better than other systems?

carcarjg, 2020-09-24:
Doxygen can output HTML, LaTeX, RTF, and XML. HTML can be run on github pages, and I also think there are a few apps or github actions on the github marketplace that can update the Repo's WIki pages based off files in a directory.

and Why Doxygen.... well, its free, easy to use, fas, and is configrable. There are better options(I cant remember the name of them off the top of my head)  that are paid, but Doxygen is the best of the free ones.

MelbourneDeveloper, 2020-09-24:
@carcarjg cool. I have used DocFX in the past. Have you compared the two? Is there a reason to choose one over the other?

carcarjg, 2020-09-24:
I would say Doxygen is easier to use and thats realy the reason I went with it over DocFX. I dont want to have to deal with CMD line everytime I need to update docs. I've used both, and Doxygen at the time was the better option, I havent Used newer versions of DocFX

MelbourneDeveloper, 2020-09-24:
@carcarjg great. Doxygen sounds good. I'll look into it.

MelbourneDeveloper, 2020-10-07:
@carcarjg I'm just looking at moving the doco to Github pages...

Do you know if it is possible to use Doxygen to export to a format that works with Github pages?

carcarjg, 2020-10-07:
I use actions to publish a site based off my Documentation folder.
https://github.com/marketplace/actions/publish-to-github-pages

MelbourneDeveloper, 2020-10-07:
@carcarjg brilliant! I was looking at that yesterday and couldn't get it working. Your example will be helpful.

MelbourneDeveloper, 2021-01-26:
@carcarjg the official documentation is [here](https://melbournedeveloper.github.io/Device.Net/index.html). 

It's still a work in progress but there is a decent amount of documentation and API documentation now.

closed 2021-01-26 by MelbourneDeveloper

carcarjg, 2021-01-26:
> @carcarjg the official documentation is [here](https://melbournedeveloper.github.io/Device.Net/index.html).
> 
> It's still a work in progress but there is a decent amount of documentation and API documentation now.

Wow!! Looks great! Props to you for not just saying there are comments somewhere in the code then telling me to screw off XD