MelbourneDeveloper/Device.Net #161: DeviceManager.Current.GetDevicesAsync() does not enumerate non HID USB devices

flyingoverclouds, 2020-10-15:
When you try to enumerate USB device after registering WindowsUsbDeviceFactory, you get an empty list. 

The **GetClassGuid** implementation in **WindowsUsbDeviceFactory** class must return **WindowsDeviceConstants.GUID_DEVINTERFACE_USB_DEVICE** instead of *WindowsDeviceConstants.WinUSBGuid*

With this correction, you are able to enumerate HID & pure USB device.

MelbourneDeveloper, 2020-10-15:
@flyingoverclouds thanks for pointing this out. 

Currently, the code does enumerate non-HID devices. I know this because if you register WindowsUsbDeviceFactory without WindowsHidDeviceFactory, it picks up USB devices that WindowsHidDeviceFactory doesn't pick up. These devices don't work with Hid. This [line](https://github.com/MelbourneDeveloper/Device.Net/blob/bfc3d1c0e22d7d7a22cf4841f9bee651d656bbce/src/Hid.Net/Windows/WindowsHidApiService.cs#L49) is where Hid.Net picks the Guid up from. This guid is different from WinUSBGuid.

Do you think it should enumerate on both constants? Or, does GUID_DEVINTERFACE_USB_DEVICE contain WinUSBGuid as a subset of devices? Also, can you please give me some examples of some devices that might show up as a GUID_DEVINTERFACE_USB_DEVICE so I can build some tests around this?

Perhaps if you could elaborate about your scenario, I can try to make sure that the issue is fixed permanently. What device are you trying to connect to?

flyingoverclouds, 2020-10-16:
I am currently adding some USB feature on hid & non hid usb device (control transfer for example), and i understood that WindowsUsbDeviceFactory should enumerate UsbDevice :)

MelbourneDeveloper, 2020-10-16:
@flyingoverclouds firstly, it sounds like you should talk to the nanoFramework people. There is already a pull request for control transfer [here](https://github.com/MelbourneDeveloper/Device.Net/pull/146). I strongly recommend joining the [discord channel](https://discord.gg/ZcvXARm) so you can chat with us about that. The reason this PR is stalled is that I haven't been able to figure out how to test it with a real device yet.

>  i understood that WindowsUsbDeviceFactory should enumerate UsbDevice

Yes. of course. My question: 

> Do you think it should enumerate on both constants? Or, does GUID_DEVINTERFACE_USB_DEVICE contain WinUSBGuid as a subset of devices?

MelbourneDeveloper, 2020-11-06:
@flyingoverclouds have you made any progress with control transfer?

What is the device that you are using? I need to get a device that I can test control transfer with. Did you join discord by any chance?

MelbourneDeveloper, 2020-11-06:
@flyingoverclouds I got a DFU mode device to connect. Please have a look at the latest code in the develop branch. You can see how to use WindowsDeviceConstants.GUID_DEVINTERFACE_USB_DEVICE to get a list of connected devices.

https://github.com/MelbourneDeveloper/Device.Net/blob/fc000a28c93e7843371ac2cfddced9440beeb1b6/src/Device.Net.UnitTests/IntegrationTests.cs#L45

[image]

Can this issue be closed?

MelbourneDeveloper, 2020-11-21:
This has been fixed. You have to specify the class guid  when listing devices

closed 2020-11-21 by MelbourneDeveloper