MelbourneDeveloper/Device.Net #162: Incorrectly implemented dispose pattern

GMMan, 2020-10-19:
**Describe the issue**
In a few of the classes I have had to touch to add some additional features, I've noticed that disposable classes that also use finalizers are not implementing the dispose pattern correctly. Namely, `UsbDevice` and `DeviceBase`, and also `WindowsUsbInterface` which lacks a finalizer but has unmanaged resources.

The dispose pattern is supposed to be implemented with both `void Dispose()` and `void Dispose(bool disposing)`, with `disposing` determining whether it's being called from the `Dispose()` function or the finalizer. The point of separating out the two is to avoid accessing managed objects in the finalizer, which may have already been destroyed. So far I haven't experienced crashes due to this, but there's a good chance it could happen.

For reference: https://docs.microsoft.com/en-us/dotnet/standard/garbage-collection/implementing-dispose

**Info**
 - Platform: Windows
 - Device Type: USB
 - Version: 3.1.0

MelbourneDeveloper, 2020-10-19:
@GMMan yep. PRs welcome.

MelbourneDeveloper, 2020-12-13:
@GMMan any chance you could submit a PR for places where you think this can be fixed?

MelbourneDeveloper, 2021-01-26:
@GMMan this is fixed now right?

Closing this, but let me know if there is still an issue.

closed 2021-01-26 by MelbourneDeveloper