MelbourneDeveloper/Device.Net #163: SafeFileHandle from WinUsb_Initialize() being released multiple times

GMMan, 2020-10-19:
**Describe the issue**
In my program I reobtain the same USB device after certain operations because the device disconnects and reconnects. I have the device in a `using` block each time. At some point after acquiring the device a second time, I get an `SEHException` tracing it to a `SafeFileHandle`.

An analysis of the issue indicated that `SafeFileHandle` will own the handle by default when used as an argument in a P/Invoke declaration. Combining that with `WinUsb_Free()` in the `Dispose()` function appears to double release the handle when the `SafeFileHandle` is eventually finalized. Changing the `InterfaceHandle` argument to `IntPtr` and manually constructing `defaultInterfaceHandle` without owning the handle seems to have alleviated the issue.

**Screenshots**
[image]

- Screenshot of where the exception is occurring
[image]

**Your Code**
Basically calling this (added to `WindowsUsbInterface`) with a large buffer

```cs
        public async Task ControlTransfer(byte requestType, byte request, ushort value, ushort index, byte[] buffer)
        {
            if (buffer == null)
            {
#if NETSTANDARD
                buffer = Array.Empty<byte>();
#else
                buffer = new byte[0];
#endif
            }

            await Task.Run(() =>
            {
                WINUSB_SETUP_PACKET setupPacket = new WINUSB_SETUP_PACKET
                {
                    RequestType = requestType,
                    Request = request,
                    Value = value,
                    Index = index,
                };

                for (int i = 0; i < buffer.Length || i == 0; i += 4096)
                {
                    int currentChunkLength = Math.Min(buffer.Length - i, 4096);
                    byte[] chunk = new byte[currentChunkLength];
                    Buffer.BlockCopy(buffer, i, chunk, 0, chunk.Length);
                    setupPacket.Length = (ushort)chunk.Length;
                    uint lengthTransferred = 0;
                    var isSuccess = WinUsbApiCalls.WinUsb_ControlTransfer(_SafeFileHandle, setupPacket, chunk, (uint)chunk.Length, ref lengthTransferred, IntPtr.Zero);
                    WindowsDeviceBase.HandleError(isSuccess, "Couldn't transmit control data");
                    Tracer?.Trace(true, chunk);
                }
            });
        }
```

**Log / Stack Trace**
```
--------------------------------------
Message: About to call SetupDiGetClassDevs for class Guid dee824ef-729b-4a0e-9c14-b7117d33a817. Flags: 18
Level: Information
Time: 2020-10-18 8:59:49 PM
Section: WindowsUsbDeviceFactory - GetConnectedDeviceDefinitionsAsync
Error: 
--------------------------------------
--------------------------------------
Message: The call to SetupDiEnumDeviceInterfaces returned ERROR_NO_MORE_ITEMS
Level: Information
Time: 2020-10-18 8:59:49 PM
Section: WindowsUsbDeviceFactory - GetConnectedDeviceDefinitionsAsync
Error: 
--------------------------------------
--------------------------------------
Message: Successfully opened handle on device for reading and writing
Level: Information
Time: 2020-10-18 8:59:49 PM
Section: WindowsUsbInterfaceManager
Error: 
--------------------------------------
<snip read/write>
--------------------------------------
Message: About to call SetupDiGetClassDevs for class Guid dee824ef-729b-4a0e-9c14-b7117d33a817. Flags: 18
Level: Information
Time: 2020-10-18 8:59:54 PM
Section: WindowsUsbDeviceFactory - GetConnectedDeviceDefinitionsAsync
Error: 
--------------------------------------
--------------------------------------
Message: The call to SetupDiEnumDeviceInterfaces returned ERROR_NO_MORE_ITEMS
Level: Information
Time: 2020-10-18 8:59:54 PM
Section: WindowsUsbDeviceFactory - GetConnectedDeviceDefinitionsAsync
Error: 
--------------------------------------
--------------------------------------
Message: Successfully opened handle on device for reading and writing
Level: Information
Time: 2020-10-18 8:59:54 PM
Section: WindowsUsbInterfaceManager
Error: 
--------------------------------------
<snip read/write>
An unhandled exception of type 'System.Runtime.InteropServices.SEHException' occurred in System.Private.CoreLib.dll
External component has thrown an exception.
```

**Info**
 - Platform: Windows
 - Device Type: USB
 - Version: 3.1.0

GMMan, 2020-10-19:
According to [this](https://stackoverflow.com/a/54276779/1180879) using `SafeFileHandle` for `InterfaceHandle` is flat-out wrong since it's not a file handle.

MelbourneDeveloper, 2020-10-19:
@GMMan have you tried replacing safe handle with InterfaceHandle? Did that solve your problem?

GMMan, 2020-10-19:
I changed the signature of `WinUsb_Initialize()` to
```cs
public static extern bool WinUsb_Initialize(SafeFileHandle DeviceHandle, out IntPtr InterfaceHandle);
```
and the calling code to
```cs
var isSuccess = WinUsbApiCalls.WinUsb_Initialize(_DeviceHandle, out var interfaceHandle);
WindowsDeviceBase.HandleError(isSuccess, Messages.ErrorMessageCouldntIntializeDevice);

var defaultInterfaceHandle = new SafeFileHandle(interfaceHandle, false);
```

Seems to work so far.

MelbourneDeveloper, 2020-10-20:
@GMMan amazing.

I'm not against changing it, but I need to see some evidence that SafeFileHandle is causing an issue. What type of device is this?

@GMMan ps: control transfers are coming. Someone else has already submitted a [PR](https://github.com/MelbourneDeveloper/Device.Net/pull/146)

MelbourneDeveloper, 2020-10-20:
@GMMan you should join the discord channel [here](https://discord.gg/ZcvXARm). Others are discussing this.

I just need a device I can test control transfers with. I can't accept any PR until I have an integration test that passes.

GMMan, 2020-10-20:
The project is a custom firmware upload tool for a device powered by a Rockchip SoC. It uses control transfers to upload boot code when the device is to be initialized when rebooted to maskrom mode. I'll try making a somewhat generic example; it's probably not tied to control transfers directly, just the device reacquisition that causes the previous `IUsbDevice` instance to be garbage collected.

MelbourneDeveloper, 2020-10-20:
@GMMan please jump on Discord. We are working on a firmware upload for an STM32 device currently. I think you can help.

MelbourneDeveloper, 2020-11-07:
@GMMan we are starting to work on control transfer [here](https://github.com/MelbourneDeveloper/Device.Net/blob/92e66dd97b2d437a6aaff4618fc278cc314e435f/src/Device.Net.UnitTests/IntegrationTests.cs#L74). Have you joined discord yet? I'd really like to chat there.

MelbourneDeveloper, 2020-12-13:
@GMMan it seems like you have done useful research here. I want to address your issue if possible. Do you want to submit a PR so I can see what you are suggesting?

brianavid, 2021-01-19:
I get a very similar exception on exit: 
```
Unhandled Exception: System.Runtime.InteropServices.SEHException: External component has thrown an exception.
   at Microsoft.Win32.Win32Native.CloseHandle(IntPtr handle)
   at Microsoft.Win32.SafeHandles.SafeFileHandle.ReleaseHandle()
   at System.Runtime.InteropServices.SafeHandle.InternalFinalize()
   at System.Runtime.InteropServices.SafeHandle.Dispose(Boolean disposing)
   at System.Runtime.InteropServices.SafeHandle.Finalize()
The program '[33524] PushWhacker.exe' has exited with code 0 (0x0).
```

In my case, the usage of Usb.Net is very simple:
Open:

```
        public static async Task<bool> OpenAsync()
        {
            var logger = new DebugLogger();
            var tracer = new DebugTracer();
            WindowsUsbDeviceFactory.Register(logger, tracer);
            var deviceDefinitions = new FilterDeviceDefinition { DeviceType = DeviceType.Usb, VendorId = AbletonVendorID, ProductId = Push2ProductId };
            var deviceDefinition = DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(deviceDefinitions).Result;
            
            if (deviceDefinition == null || deviceDefinition.Count() < 1)
            {
                return false;
            }

            PushDeviceId = deviceDefinition.First().DeviceId;

            //This is the only platform specific part. Each platform has a UsbInterfaceManager
            usbInterfaceManager = new WindowsUsbInterfaceManager
            (
                PushDeviceId,
                logger,
                tracer,
                null,
                null
            );

            usbDevice = new UsbDevice(PushDeviceId, usbInterfaceManager, logger, tracer);

            await usbDevice.InitializeAsync();

```

Close:

```
            if (usbDevice == null) return;

            usbDevice.Close();
            usbDevice.Dispose();
            usbInterfaceManager.Close();
            usbInterfaceManager.Dispose();
            usbDevice = null;
            usbInterfaceManager = null;
```

It's only at the end on exit and isn't visible when not debugging. But it may be more evidence ...

This is version 3.1.0 via NuGet.

MelbourneDeveloper, 2021-01-19:
@brianavid this is very old code. Have ytou tried the newer 4.X alpha version?

MelbourneDeveloper, 2021-01-19:
@GMMan same question. Have you tried the newer version? Control Transfer now works.

If you are still having trouble with SafeFileHandle, I can look in to it.

GMMan, 2021-01-19:
I haven't touched control transfer stuff yet, though I put up a few changes I had here: https://github.com/GMMan/Device.Net/commit/d2ce336167ea728107b6ce78e5315d279d6c24f2

brianavid, 2021-01-20:
> @brianavid this is very old code. Have ytou tried the newer 4.X alpha version?

I much prefer to use NuGet packages in my project to avoid the hassle of publishing source that depends on unreleased private builds of other projects. That was why I was running 3.1.0

Anyway, it isn't clear where is the newer 4.0 alpha. I have built and run against your latest master branch that seems pretty recent. And I still get the same problem.

So I attempted a cut-down application to demonstrate the problem. Interestingly it didn't fail with a Console app. But I created the most trivial Windows Form app (from the VS wizard) and added the above code to Program.cs and I get the error when running as Debug.

The Windows Form VS solution is attached as a Zip 
[Usb.Net.Win.zip](https://github.com/MelbourneDeveloper/Device.Net/files/5841497/Usb.Net.Win.zip)
- see the Program.cs in it.

MelbourneDeveloper, 2021-01-20:
@brianavid you can get the 4.x alpha versions from nuget.org if you check the prerelease checkbox. 

The master branch is extremely old.

brianavid, 2021-01-20:
OK - I hadn't spotted that pre-release option checkbox. And it took me a while to construct the necessary very simple library calls without documentation. But I am now using 4.0.7alpha and my initialisation is now ...

```
            var deviceDefinitions = new FilterDeviceDefinition (AbletonVendorID, Push2ProductId); 
            var deviceFactories = WindowsUsbDeviceFactoryExtensions.CreateWindowsUsbDeviceFactory(deviceDefinitions);
            var deviceManager = new DeviceManager(new List<IDeviceFactory> { deviceFactories });
            var deviceDefinition = deviceManager.GetConnectedDeviceDefinitionsAsync().Result;

            if (deviceDefinition == null || deviceDefinition.Count() < 1)
            {
                return false;
            }

            PushDeviceId = deviceDefinition.First().DeviceId;
```

But I still get the same exception on exit!!

Updated zipped solution (without the NuGet packages) is here: 
[Usb.Net.Win.zip](https://github.com/MelbourneDeveloper/Device.Net/files/5842045/Usb.Net.Win.zip)

brianavid, 2021-01-20:
I have further reduced the failing test case to a single (non-console) Program.cs Main():

```
using System.Collections.Generic;
using System.Linq;
using Device.Net;
using Usb.Net.Windows;

namespace Usb.Net.Win
{
    static class Program
    {
        private const int AbletonVendorID = 0x2982;
        private const int Push2ProductId = 0x1967;

        static void Main()
        {
            var deviceDefinitions = new FilterDeviceDefinition (AbletonVendorID, Push2ProductId); 
            var deviceFactory = WindowsUsbDeviceFactoryExtensions.CreateWindowsUsbDeviceFactory(deviceDefinitions);
            var deviceManager = new DeviceManager(new List<IDeviceFactory> { deviceFactory });
            var deviceDefinition = deviceManager.GetConnectedDeviceDefinitionsAsync().Result;

            if (deviceDefinition != null && deviceDefinition.Count() >= 1)
            {
                var PushDeviceId = deviceDefinition.First().DeviceId;

                //This is the only platform specific part. Each platform has a UsbInterfaceManager
                var usbInterfaceManager = new WindowsUsbInterfaceManager(PushDeviceId);

                var usbDevice = new UsbDevice(PushDeviceId, usbInterfaceManager);

                usbDevice.InitializeAsync().Wait();

                usbDevice.Close();
                usbDevice.Dispose();
                usbInterfaceManager.Close();
                usbInterfaceManager.Dispose();
            }
        }
    }
}

```

This is about the limit while remaining readable. Skipping the Close() and Dispose() calls makes no difference and still causes the Win32 CloseHandle SEHException in the Finalizers.

MelbourneDeveloper, 2021-01-21:
@brianavid wow! You must really hate async/await! Just brutally calling `Wait()` on those tasks!

I have seen the exception that you mention happening in the past, but I don't see it on the devices I use. I'll try using your code to see if I get it.

MelbourneDeveloper, 2021-01-21:
@GMMan suggested a solution. Any chance we can get a PR for that?

brianavid, 2021-01-21:
> @brianavid wow! You must really hate async/await! Just brutally calling `Wait()` on those tasks!
> 
> I have seen the exception that you mention happening in the past, but I don't see it on the devices I use. I'll try using your code to see if I get it.

No - I don't really hate it. They are still all there in my "live" code. But I was trying to reduce the "test" code to be as short as possible, but still causing the problem and still readable. I knew you had had the exception in the past, because one of the links I found while researching it was https://stackoverflow.com/questions/54189056/c-sharp-winusb-cant-call-closehandle-on-interface which seems very familiar!

Good hunting!

GMMan, 2021-01-21:
> @GMMan suggested a solution. Any chance we can get a PR for that?

Filed as #187, though it seems your pipeline's not working quite right.

GMMan, 2021-01-21:
BTW, as more context for the problem, usually if you only open the device once and close it at the end of program execution, you don't really see the exception because the finalizer would be running after the program's exiting, but if you dispose the device and it gets garbage collected, the finalizer will kick in and the safe file handle will try to close the handle, either one that `CloseHandle()` don't know about or already closed via WinUSB's free function.

MelbourneDeveloper, 2021-01-21:
@GMMan thankyou very much! I tested this on three devices and it seems to at least be working the same as before.

@brianavid does this solve your issue?

MelbourneDeveloper, 2021-01-21:
@GMMan unfortunately, I still have an issue...

The second time I try to initialize one of my devices, it hangs. I can't tell if this is a problem with the unit testing framework or a bug...

brianavid, 2021-01-21:
> @brianavid does this solve your issue?

Thank you. Yes ... It seems to fix my issue. I have fetched your develop branch, built it locally and linked with my test code. I now no longer see the exception in the finalizers. My main concern was that I had evidence of a bug that may be more catastrophic for other users that need to close and then re-open devices with the GC wreaking havoc after the close. As I do nothing after the close, my live project had no known consequential issues.

MelbourneDeveloper, 2021-01-21:
@GMMan thanks! Are you happy to close this issue now?

MelbourneDeveloper, 2021-01-22:
@GMMan I am closing this one. If you have further issues, let me know.

Thanks again.

closed 2021-01-22 by MelbourneDeveloper