MelbourneDeveloper/Device.Net #164: ReadReportAsync gets blocked

MPapus, 2020-10-27:
**Describe the issue**
So I used the console [Usb.Net.WindowsSample](https://github.com/MelbourneDeveloper/Device.Net/blob/master/src/Usb.Net.WindowsSample/Program.cs) project to play around with a soundcard ([this one](https://www.cmedia.com.tw/products/USB20_FULL_SPEED/CM6206)) which accepts writing and reading data to/from its EEPROM via HID interface, and my company's HW team programmed via MPLAB. Issue appears when trying to read from device's EEPROM (tried with register 3) execution gets blocked when awaiting [`WindowsHIDDevice`](https://github.com/MelbourneDeveloper/Device.Net/blob/bfc3d1c0e22d7d7a22cf4841f9bee651d656bbce/src/Hid.Net/Windows/WindowsHIDDevice.cs) `_ReadFileStream`'s `ReadAsync` method ([here](https://github.com/MelbourneDeveloper/Device.Net/blob/bfc3d1c0e22d7d7a22cf4841f9bee651d656bbce/src/Hid.Net/Windows/WindowsHIDDevice.cs#L222)).

**Screenshots**
- Here shows how EEPROM has to be accessed, and EEPROM has addresses from 0 to 63.
[image]

- Here the discovered devices and mine marked down
[image]

**Your Code**
I modified [TrezorExample](https://github.com/MelbourneDeveloper/Device.Net/blob/master/src/Usb.Net.UWP.Sample/TrezorExample.cs)'s `_DeviceDefinitions` to only include desired filter:
```
private readonly List<FilterDeviceDefinition> _DeviceDefinitions = new List<FilterDeviceDefinition>
{
    new FilterDeviceDefinition{ DeviceType= DeviceType.Hid, VendorId= 0x153C, ProductId= 0x000C }
};
```
And here the modified version of the `WriteAndReadFromDeviceAsync` so the buffer fits it, setting the bits 6 and 4 to one to indicate we want to access the EEPROM and read data from it. I'm assuming bytes 1 and 2 (DATAL and DATAH in the picture above) will be filled with the desired info from the specified register when reading the output. Byte 3 includes which EEPROM's register we want to access, 3 in this case.
```
public async Task<byte[]> WriteAndReadFromDeviceAsync()
{
    //Create a buffer with 4 bytes (initialize)
    var firstByte = new BitArray(8);
    firstByte[6] = true; // bit to indicate we want to access EEPROM
    firstByte[4] = true; // bit to indicate we want to read from EEPROM
    var writeBuffer = new byte[4];
    firstByte.CopyTo(writeBuffer, 0); // copy bit array to byte array
    writeBuffer[3] = 0x3; // set the register number we want to access

    //Write the data to the device
    return await TrezorDevice.WriteAndReadAsync(writeBuffer);
}
```

So after doing all this and run the sample, it find the desired device, it successfully initializes it and it even gets to write the data I just mentioned before (see the logs below) but when calling [this](https://github.com/MelbourneDeveloper/Device.Net/blob/bfc3d1c0e22d7d7a22cf4841f9bee651d656bbce/src/Hid.Net/Windows/WindowsHIDDevice.cs#L222) (in 3.1.0 it's inside a `Task.Run()`) it never finishes.

**Log / Stack Trace**
Please find attached the relevant logging:
```
--------------------------------------
Message: Found device \\?\hid#vid_153c&pid_000c&mi_03#8&72d0618&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Level: Information
Time: 27/10/20 13:08:48
Section: WindowsHidDeviceFactory
Error: 
--------------------------------------
--------------------------------------
Message: Could not get Hid string. Caller: GetManufacturer
Level: Warning
Time: 27/10/20 13:08:48
Section: WindowsHidApiService
Error: 
--------------------------------------
--------------------------------------
Message: Could not get Hid string. Caller: GetSerialNumber
Level: Warning
Time: 27/10/20 13:08:48
Section: WindowsHidApiService
Error: 
--------------------------------------
--------------------------------------
Message: The call to SetupDiEnumDeviceInterfaces returned ERROR_NO_MORE_ITEMS
Level: Information
Time: 27/10/20 13:08:48
Section: WindowsHidDeviceFactory - GetConnectedDeviceDefinitionsAsync
Error: 
--------------------------------------
--------------------------------------
Message: Calling CreateFile for DeviceId: \\?\hid#vid_153c&pid_000c&mi_03#8&72d0618&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}. Desired Access: GenericRead. Share mode: 3. Creation Disposition: 3
Level: Information
Time: 27/10/20 13:08:48
Section: ApiService
Error: 
--------------------------------------
--------------------------------------
Message: Calling CreateFile for DeviceId: \\?\hid#vid_153c&pid_000c&mi_03#8&72d0618&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}. Desired Access: GenericWrite, GenericRead. Share mode: 3. Creation Disposition: 3
Level: Information
Time: 27/10/20 13:08:48
Section: ApiService
Error: 
--------------------------------------
--------------------------------------
Message: Could not get Hid string. Caller: GetManufacturer
Level: Warning
Time: 27/10/20 13:08:48
Section: WindowsHidApiService
Error: 
--------------------------------------
--------------------------------------
Message: Could not get Hid string. Caller: GetSerialNumber
Level: Warning
Time: 27/10/20 13:08:48
Section: WindowsHidApiService
Error: 
--------------------------------------
--------------------------------------
Message: Read file stream opened successfully
Level: Information
Time: 27/10/20 13:08:48
Section: WindowsHidDevice
Error: 
--------------------------------------
--------------------------------------
Message: Write file stream opened successfully
Level: Information
Time: 27/10/20 13:08:48
Section: WindowsHidDevice
Error: 
--------------------------------------
Write: (80,0,0,3) (4)
```

**Info**
 - Platform: [Windows]
 - Device Type: [Hid]
 - Version: [3.1.0]

MelbourneDeveloper, 2020-10-27:
@MPapus Nice! I always wanted to work with audio interfaces. I'm a musician myself.

> it successfully initializes it and it even gets to write the data I just mentioned before (see the logs below) but when calling this (in 3.1.0 it's inside a Task.Run()) it never finishes.

Yes, this is a common problem. This usually happens when the data you sent it is the wrong size, or you didn't send the 0 at the start.

Firstly, check the detected report size. ReadBufferSize after initialization. If it's 64, you should send an array that is 64 bytes long.

Another thing you should try is putting a 0 in front. E.g:

0,80,0,0,3 [...]

MPapus, 2020-10-28:
Thanks @MelbourneDeveloper it works! I can get everything I needed! I'm closing this

closed 2020-10-28 by MPapus