MelbourneDeveloper/Device.Net #169: Control Transfer API Design

MelbourneDeveloper, 2020-11-13:
The control transfer method(s) read and write data from the device. I would like to try to keep some consistency across the APIs. The original APIs are called `WriteAsync` and `ReadAsync`. When used in a sequence, they become `WriteAndReadAsync`.  The key about write and read is that there is a semaphore lock around the calls so that they happen in sequence.

I'm not particularly in love with the current naming and would consider changing the names. However, I want to try to keep V4 with a reasonably stable public API. I hope to finish off the public API by V5. 

The control transfer methods come in two flavours: in and out, or read and write. As far as I understand, one sends data to the device, and one reads from it - in a very similar way to the methods above. The major difference is that when we do control transfer, we also sent a `SetupPacket` with every call. When we write to the device, there is no need for a return value. And, when we read from the device, there is no need to send a buffer. So, the first question, is **_should we split the method in to two_**?

Secondly, are there cases where we will need to do transfers in a request/response style? I.e. send and then await the response? If so, we're probably going to need a call like `WriteAndReadAsync` to put a semaphore around the call so the sequence is guaranteed. **_Do we need this?_**

Lastly, **_what should the control transfer method(s) be called?_**

MelbourneDeveloper, 2020-11-13:
@flyingoverclouds @GMMan @networkfusion @josesimoes

josesimoes, 2020-11-13:
> So, the first question, is **_should we split the method in to two_**?

Nope. The lower layer call is the same. The validations/handling are basically the same. So I would keep it in a single method.

josesimoes, 2020-11-13:
> Secondly, are there cases where we will need to do transfers in a request/response style? I.e. send and then await the response? If so, we're probably going to need a call like `WriteAndReadAsync` to put a semaphore around the call so the sequence is guaranteed. **_Do we need this?_**

I don't think we need this... Whenever there is a need to send something and read something those are "atomic" operations and I don't see any advantage on having a combined one.

MelbourneDeveloper, 2020-11-13:
@josesimoes , when you say "atomic", do you mean that each call is meaningful by itself? I.e. It's not a request/response?

josesimoes, 2020-11-13:
Regarding naming... I don't dislike what is there now... 
Maybe the SendControlTransferAsync is *not exactly* a send... more like an Execute... or more Perform... 
And definitely stay away from "write" or "read".

MelbourneDeveloper, 2020-11-13:
@josesimoes I don't like "Send" because it implies that we're always sending data when we might not.

> Execute... or more Perform...
both sound good

> stay away from "write" or "read"
I think I agree

Still, it feels a bit disjointed because the main methods are called `ReadAsync` and `WriteAsync`. Control transfers are kind of analogous to those because we are writing and reading data - it's just that we also send the setup packet.

josesimoes, 2020-11-13:
> Still, it feels a bit disjointed because the main methods are called `ReadAsync` and `WriteAsync`. Control transfers are kind of analogous to those because we are writing and reading data - it's just that we also send the setup packet.

I got "inspiration" from the WinUSB API that we are calling in the end. 😉 

Plus this is a new *main* method, doing another thing, so no worries there. At least the way I see it.

MelbourneDeveloper, 2020-11-13:
I'm thinking `PerformControlTransferAsync` is in front.

However, I'm also considering removing the `Async` suffix from everything...

MelbourneDeveloper, 2020-11-13:
What's the consensus on removing `Async` from everything?

It's frustrating because Microsoft originally recommending suffixing everything with `Async` but newer APIs don't necessarily have it.

josesimoes, 2020-11-13:
> What's the consensus on removing `Async` from everything?
> 
> It's frustrating because Microsoft originally recommending suffixing everything with `Async` but newer APIs don't necessarily have it.

Curious... I haven't read anything about that. Or even notice it.
My mindset is still on the async suffix. 😁

MelbourneDeveloper, 2020-11-13:
@josesimoes I think I'll leave it for now, but if things go in the other direction soon, I might change it in V5.

closed 2020-12-13 by MelbourneDeveloper