MelbourneDeveloper/Device.Net #177: Stuck on Getting Devices

Mxater, 2020-11-20:
**Describe the issue**
I try to get Devices using 
`var devicesList =  DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(new FilterDeviceDefinition()).Result;;`
But Stuck loading.
Works in Console Application, but not in Desktop Application

**Screenshots**
[image]
[image]

**Your Code**
 ```
using System;
using System.Collections.Generic;
using System.Linq;
using System.Threading.Tasks;
using System.Xml;
using Device.Net;
using Hid.Net.Windows;
using Usb.Net.Windows;

namespace DualSenseSupport
{
    public class Devices
    {
        private static readonly DSProject.Logger _logger = new DSProject.Logger();
        private static readonly Tracer _tracer = new Tracer();
        
        
        private static List<DeviceInfo> devicesList= new List<DeviceInfo>();
        
        public static void Init()
        {
            WindowsHidDeviceFactory.Register(_logger, _tracer);
            WindowsUsbDeviceFactory.Register(_logger, _tracer);
            
            SearchDS5Controller();
        }

        public static int GetDeviceCount()
        {
            return devicesList.Count;
        }

        public static DeviceInfo GetDevice(int index)
        {
            return devicesList[index];
        } 
        

        public static IEnumerable<DeviceInfo> EnumerateDevices()
        {
            // var devicesx = DeviceManager.Current.GetDevicesAsync(new List<FilterDeviceDefinition>()
            // {
            //     new FilterDeviceDefinition{  VendorId= 1356 }
            // }).Result;
            var devicesList =  DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(new FilterDeviceDefinition()).Result;
            var devices = new List<DeviceInfo>();
            foreach (var deviceDefinition in devicesList)
            {
                if (deviceDefinition.VendorId == 1356)
                {
                    var device = new DeviceInfo
                    {
                        Connection = deviceDefinition,
                        Device = DeviceManager.Current.GetDevice(deviceDefinition),
                        Uid= Guid.NewGuid().ToString()
                    };
                    if (device.Connection.WriteBufferSize == 48)
                    {
                        device.ConnectionMethod = DeviceInfo.ConnectionType.Usb;
                    }
                    else if(device.Connection.WriteBufferSize == 547)
                    {
                        device.ConnectionMethod = DeviceInfo.ConnectionType.Bluetooth;
                    }
                    else
                    {
                        device.ConnectionMethod = DeviceInfo.ConnectionType.Unknow;
                    }
                    devices.Add(device);
                }
            }

            return devices;
        }

        public static void SearchDS5Controller()
        {
            var devices=EnumerateDevices();
            foreach (var deviceInfo in devicesList.Where(deviceInfo => deviceInfo.Device.IsInitialized))
            {
                deviceInfo.Close();
            }

            foreach (var deviceInfo in devices)
            {
                devicesList.Add(deviceInfo);
                deviceInfo.Init();
            }
            
        }

        public static void Close()
        {
            foreach (var deviceInfo in devicesList.Where(deviceInfo => deviceInfo.Device.IsInitialized))
            {
                deviceInfo.Close();
            }
        }
        
    }
    
   
}
```

**Info**
 - Platform: Windows c# .net 4.7.2
 - Device Type: USB
 - Version: 3

MelbourneDeveloper, 2020-11-20:
Thankyou @Mxater 

Could you please post the log?

Also, when you say desktop app, what is the platform? .net framework? .net core?

MelbourneDeveloper, 2020-11-20:
DSProject.Logger(); 

should log to the console

Mxater, 2020-11-20:
.net framework, Windows Forms Application.

I get this.
[image]

Mxater, 2020-11-20:
I using DSProject.Logger here
[image]

Mxater, 2020-11-20:
```
About to call SetupDiGetClassDevs for class Guid 4d1e55b2-f16f-11cf-88cb-001111000030. Flags: 18
Calling CreateFile for DeviceId: \\?\hid#vid_1532&pid_022a&mi_01&col02#8&3b20310f&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}. Desired Access: None. Share mode: 3. Creation Disposition: 3
Found device \\?\hid#vid_1532&pid_022a&mi_01&col02#8&3b20310f&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
Could not get Hid string. Caller: GetSerialNumber
Calling CreateFile for DeviceId: \\?\hid#vid_1532&pid_022a&mi_01&col03#8&3b20310f&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}. Desired Access: None. Share mode: 3. Creation Disposition: 3
Found device \\?\hid#vid_1532&pid_022a&mi_01&col03#8&3b20310f&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
Could not get Hid string. Caller: GetSerialNumber
Calling CreateFile for DeviceId: \\?\hid#vid_1532&pid_022a&mi_02#8&29d2fd21&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}. Desired Access: None. Share mode: 3. Creation Disposition: 3
Found device \\?\hid#vid_1532&pid_022a&mi_02#8&29d2fd21&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Could not get Hid string. Caller: GetSerialNumber
Calling CreateFile for DeviceId: \\?\hid#vid_1532&pid_022a&mi_01&col04#8&3b20310f&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}. Desired Access: None. Share mode: 3. Creation Disposition: 3
Found device \\?\hid#vid_1532&pid_022a&mi_01&col04#8&3b20310f&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}
Could not get Hid string. Caller: GetSerialNumber
Calling CreateFile for DeviceId: \\?\hid#vid_1532&pid_022a&mi_01&col05#8&3b20310f&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}. Desired Access: None. Share mode: 3. Creation Disposition: 3
Found device \\?\hid#vid_1532&pid_022a&mi_01&col05#8&3b20310f&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}
Could not get Hid string. Caller: GetSerialNumber
Calling CreateFile for DeviceId: \\?\hid#vid_1e71&pid_2002#7&cf57595&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}. Desired Access: None. Share mode: 3. Creation Disposition: 3
Found device \\?\hid#vid_1e71&pid_2002#7&cf57595&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Calling CreateFile for DeviceId: \\?\hid#vid_046d&pid_c231#2&227aec24&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}. Desired Access: None. Share mode: 3. Creation Disposition: 3
Found device \\?\hid#vid_046d&pid_c231#2&227aec24&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Could not get Hid string. Caller: GetManufacturer
Could not get Hid string. Caller: GetSerialNumber
Could not get Hid string. Caller: GetProduct
Calling CreateFile for DeviceId: \\?\hid#vid_1532&pid_022a&mi_01&col01#8&3b20310f&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd. Desired Access: None. Share mode: 3. Creation Disposition: 3
Found device \\?\hid#vid_1532&pid_022a&mi_01&col01#8&3b20310f&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
Could not get Hid string. Caller: GetSerialNumber
Calling CreateFile for DeviceId: \\?\hid#vid_1532&pid_022a&mi_00#8&1748f34d&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd. Desired Access: None. Share mode: 3. Creation Disposition: 3
Found device \\?\hid#vid_1532&pid_022a&mi_00#8&1748f34d&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
Could not get Hid string. Caller: GetSerialNumber
Calling CreateFile for DeviceId: \\?\hid#vid_1532&pid_022a&mi_00&col03#9&2338c189&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}. Desired Access: None. Share mode: 3. Creation Disposition: 3
Found device \\?\hid#vid_1532&pid_022a&mi_00&col03#9&2338c189&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Could not get Hid string. Caller: GetManufacturer
Could not get Hid string. Caller: GetSerialNumber
Could not get Hid string. Caller: GetProduct
Calling CreateFile for DeviceId: \\?\hid#vid_046d&pid_c232#2&113d570e&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd. Desired Access: None. Share mode: 3. Creation Disposition: 3
Found device \\?\hid#vid_046d&pid_c232#2&113d570e&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
Could not get Hid string. Caller: GetManufacturer
Could not get Hid string. Caller: GetSerialNumber
Could not get Hid string. Caller: GetProduct
Calling CreateFile for DeviceId: \\?\hid#vid_054c&pid_0ce6&mi_03#9&1e6d4b54&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}. Desired Access: None. Share mode: 3. Creation Disposition: 3
Found device \\?\hid#vid_054c&pid_0ce6&mi_03#9&1e6d4b54&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Could not get Hid string. Caller: GetSerialNumber
Calling CreateFile for DeviceId: \\?\hid#vid_1532&pid_0016#7&23089577&2&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}. Desired Access: None. Share mode: 3. Creation Disposition: 3
Found device \\?\hid#vid_1532&pid_0016#7&23089577&2&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
Could not get Hid string. Caller: GetSerialNumber
The call to SetupDiEnumDeviceInterfaces returned ERROR_NO_MORE_ITEMS

```

Mxater, 2020-11-20:
I find some, All Async task not work in main thread.

If use new Thread this work.
```
IEnumerable<ConnectedDeviceDefinition> devicesList = new List<ConnectedDeviceDefinition>();
            var thread = new Thread(o =>
            {
                devicesList = DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(new FilterDeviceDefinition())
                    .Result;
            });
            thread.Start();
            thread.Join();
```

MelbourneDeveloper, 2020-11-21:
@Mxater in the new version (4.0) [this code](https://github.com/MelbourneDeveloper/Device.Net/blob/e50b4fdb22d4357f8f7053571ec26cafaf71bba1/src/Device.Net/Windows/WindowsDeviceEnumerator.cs#L35) has completely changed. It will always wrap the task in a thread. You can see it here. 

I recommend working with the develop branch. It is pretty stable at the moment and the 4.0 alpha release will be any day now.

MelbourneDeveloper, 2020-12-13:
@Mxater I am closing this issue. If you still have the same issue with the latest version 4.0.2-alpha, please let me know and I will reopen.

closed 2020-12-13 by MelbourneDeveloper