MelbourneDeveloper/Device.Net #180: What links a ConnectedDeviceDefinition to an IDevice

zlittell, 2020-11-23:
**Describe the issue**
When calling DeviceManager.Current.GetDevicesAsync() sometimes ConnectedDeviceDefinition is null and sometimes it isn't. However, if I call DeviceManager.Current.GetConnectedDeviceDefinitionsAsync() it always retrieves all the ConnectedDeviceDefinitions everytime. I use the definition to display a devices serial and sometimes it works and sometimes it doesn't.

I am not sure if maybe I am just not understanding something correctly or there is some other issue.

**Your Code**
```
using Device.Net;
using Hid.Net.Windows;
using System;
using System.Collections.Generic;
using System.Threading.Tasks;

namespace USBHIDConsoleDemo
{
    class Program
    {
        static async Task<int> Main()
        {
            DebugLogger dLogger = new DebugLogger();
            DebugTracer dTracer = new DebugTracer();

            Console.WriteLine("Hello World!");
            WindowsHidDeviceFactory.Register(dLogger, dTracer);

            var deviceDefinitions = new List<FilterDeviceDefinition>
            {
                new FilterDeviceDefinition {DeviceType=DeviceType.Hid, VendorId= 0x1209, ProductId=0x9001}
            };

            while (true)
            {
                var devices2 = await DeviceManager.Current.GetDevicesAsync(deviceDefinitions);
                var devices3 = await DeviceManager.Current.GetConnectedDeviceDefinitionsAsync(deviceDefinitions[0]);
                foreach (var temp in devices2)
                {
                    if (temp.ConnectedDeviceDefinition != null)
                    {
                        Console.WriteLine("GetDevices Device: " + temp.ConnectedDeviceDefinition.SerialNumber);
                    }
                }
                foreach (var temp in devices3)
                {
                    Console.WriteLine("GetConnectedDevice Device:" + temp.SerialNumber);
                }
            }
        }
    }
}
```

**Info**
 - Platform: Windows
 - Device Type: HID
 - Version: 3.1

MelbourneDeveloper, 2020-11-23:
@zlittell Hi. Yes. This call is a little confusing and has been cleaned up in V4.

If you specify no filter definitions to `GetConnectedDeviceDefinitionsAsync` it should return every device definition. However, `GetDevicesAsync` doesn't because it might return too many devices. 

If the result is inconsistent, that sounds like a bug.

zlittell, 2020-11-23:
Sorry I don't think I described it correctly in the first part. I am using filters on both functions. When getting a list of IDevices with GetDevicesAsync(ListofFilters), sometimes the object, ConnectedDeviceDefinition, of the IDevice is set correctly and sometimes its just null. If I get a list of ConnectedDeviceDefinitions using GetConnectedDeviceDefinitionsAsync it always shows the ConnectedDeviceDefinitions correctly.

One solution I have been toying with is just keeping a list of ConnectedDeviceDefinitions and then using GetDevice() with the selected ConnectedDeviceDefinition passed as the argument. It just would be easier to use the list of IDevices, but I have to have the serial so the user can select a device if more than one is plugged in.

[image]
[image]

MelbourneDeveloper, 2020-11-23:
@zlittell ah yes. I think I understand what you are saying. This would be a bug. I would hope this is fixed in version 4 (develop branch) but if not, I will keep this issue open to remind me that it is a problem.

This kind of ties in to some design issues.

Really, this property is not set until you initialize the device. That's a kind of design problem that I will deal with long term.

zlittell, 2020-11-23:
Thats fine. I will use the workaround its simple enough. I just wanted to make sure I wasn't doing something wrong. Thanks for the quick response and the amazing work!