MelbourneDeveloper/Device.Net #181: "Device.Net - Windows Samples.sln" does not include SerialPort.Net project

LiorBanaiKama, 2020-12-01:
so fresh build fails since the SerialPort.Net is not compiled

MelbourneDeveloper, 2020-12-01:
@LiorBanaiKama it builds for me. Are you talking about develop branch?

Please post the error

LiorBanai, 2020-12-01:
when I download the code (from master) and open the sln I get the following project in solution
[image]

so then I'm unable to compile:
[image]

If I add the missing project. for example to the device folder it does compile and run:
[image]

MelbourneDeveloper, 2020-12-01:
@LiorBanai yep. I there are too many solutions. I think you're looking at a broken one. I only tend to work on the all solution. 

Make sure you using the develop branch

LiorBanai, 2020-12-01:
ok, thanks

LiorBanai, 2020-12-01:
I suggest making develop the default branch since people unaware that it is more updated and github landing page points to master so downloading code will fetch old code:

[image]

MelbourneDeveloper, 2020-12-01:
@LiorBanai that's how most repos work. The master or main branch represents the released code (Nuget package) and the develop branch represents what will be released next. 

You can read about it here
https://www.atlassian.com/git/tutorials/comparing-workflows/gitflow-workflow

MelbourneDeveloper, 2020-12-13:
@LiorBanai closing this issue. The develop branch will become the master branch when the full version is released.

closed 2020-12-13 by MelbourneDeveloper