MelbourneDeveloper/Device.Net #186: Overloads appear to be missing for `GetReportAsync(ushort reportId);`

rquackenbush, 2021-01-19:
**Describe the issue**
Overloads appear to be missing for `GetReportAsync(ushort reportId);`

**Screenshots**
[image]

**Your Code**
```c#
//I'm expecting to pass in '1' here
var report = await hidDevice.ReadReportAsync(cancellationToken);
```

In the sample code I've been looking at (which uses the UWP `Windows.Devices.HumanInterfaceDevice.HidDevice`, the code is doing the following:

```c#
var  report = Device.GetInputReportAsync(1)
```

This is possible because the UWP version of the USB classes has more overloads:

```c#
public IAsyncOperation<HidInputReport> GetInputReportAsync();
public IAsyncOperation<HidInputReport> GetInputReportAsync(ushort reportId);
public IAsyncOperation<HidFeatureReport> GetFeatureReportAsync();
public IAsyncOperation<HidFeatureReport> GetFeatureReportAsync(ushort reportId);
```

Is there a way to add these overloads to Hid.Net? Or should I construct my own messages to make this happen?

**Log / Stack Trace**
N/A

**Info**
 - Platform: Windows
 - Device Type: Hid
 - Version: 4.0.7-alpha

MelbourneDeveloper, 2021-01-19:
@rquackenbush Hi. Where did you get the sample code from? The documentation is out of date. I will update it once version 4 is finished.

This is the current implementation for UWP: https://github.com/MelbourneDeveloper/Device.Net/blob/4618f5b3811ee02b6b15ce0f7b1457aab52a41d7/src/Hid.Net.UWP/UWPHidDeviceHandler.cs#L139

It will be merged in to develop soon and I will do another alpha release.

MelbourneDeveloper, 2021-01-19:
@rquackenbush there is no way to read by a specifc Report Id. You just read, and then you get back whatever result the device gives you which will include the Report Id.

MelbourneDeveloper, 2021-01-19:
If you want to filter Report Ids you need to do it on your side.

The DataStreamer class is designed for this kind of thing. You can turn it in to an IObservable and use `Select` on the ReportId.

MelbourneDeveloper, 2021-01-26:
@rquackenbush unless I'm mistaken, this is not a bug so I will close this. Let me know if I'm wrong.

closed 2021-01-26 by MelbourneDeveloper