MelbourneDeveloper/Device.Net #188: CancellationTokens Don't Work

MelbourneDeveloper, 2021-01-22:
Some operations simply hang. An example is FileStream.ReadAsync on Windows. I assumed passing a cancellation token in wouldstop the operation with an exception but it doesn't. This is a pretty serious flaw and might be inherent in .NET.

MelbourneDeveloper, 2021-01-22:
This probably happens because we need to open file streams as async but when we do, we get an error.

ionothanus, 2021-07-08:
Opening the streams with `async: true` provokes an error because the underlying file handle was not created for asynchronous I/O. The documentation states [FILE_FLAG_OVERLAPPED](https://docs.microsoft.com/en-us/windows/win32/api/fileapi/nf-fileapi-createfilea#FILE_FLAG_OVERLAPPED) is required to create a handle for asynchronous I/O. I don't know about everything everywhere, but I can confirm with .NET 5.0 that once the appropriate flag (0x40000000) is passed to CreateFile through the dwFlagsAndAttributes parameter, `async: true` succeeds and cancellation tokens work.

i.e., here, the `0` (sixth) argument should be `0x40000000`. 
https://github.com/MelbourneDeveloper/Device.Net/blob/a7991515b5c6e45dd3b65b20b820512f42636d4b/src/Device.Net/Windows/ApiService.cs#L41

In my testing this forces `async: true` when opening the FileStream, i.e., you cannot set FILE_FLAG_OVERLAPPED and leave `async: false`. 

I haven't created a PR because I don't know about the broader effects of making this change (are async I/O handles appropriate everywhere? Does this work on .NET Framework?). Happy to discuss further - if you want a PR for just the two changes described above, let me know.

MelbourneDeveloper, 2021-07-08:
I reckon you should try it out. If it works, awesome. I should be able to accept the pr.

MelbourneDeveloper, 2021-07-20:
@ionothanus thanks

PR #215 (2021-07-18): Enable async I/O handles to support cancellation tokens
Resolves #188 by opening handles in asynchronous (overlapped) mode.
closed 2021-07-20 by MelbourneDeveloper