MelbourneDeveloper/Device.Net #193: What is it for LoggerFactory ?

HotYearKit, 2021-02-09:
Microsoft.Extensions.Logging.Abstractions  Can't let me use Device.Net   on the Unity Engine.

MelbourneDeveloper, 2021-02-09:
More detail please. I will close this issue otherwise.

HotYearKit, 2021-02-10:
at System.IO.FileStream.Init (Microsoft.Win32.SafeHandles.SafeFileHandle safeHandle, System.IO.FileAccess access, System.Boolean ownsHandle, System.Int32 bufferSize, System.Boolean isAsync, System.Boolean isConsoleWrapper) [0x0006e] in <9577ac7a62ef43179789031239ba8798>:0 
  at System.IO.FileStream..ctor (Microsoft.Win32.SafeHandles.SafeFileHandle handle, System.IO.FileAccess access, System.Int32 bufferSize, System.Boolean isAsync) [0x00011] in <9577ac7a62ef43179789031239ba8798>:0 
  at (wrapper remoting-invoke-with-check) System.IO.FileStream..ctor(Microsoft.Win32.SafeHandles.SafeFileHandle,System.IO.FileAccess,int,bool)
  at Hid.Net.Windows.WindowsHidApiService.OpenRead (Microsoft.Win32.SafeHandles.SafeFileHandle readSafeFileHandle, System.UInt16 readBufferSize) [0x00000] in <6aa716917c4045a18ded6a6892a77cbf>:0 
  at Hid.Net.Windows.WindowsHidHandler.<InitializeAsync>b__33_0 () [0x001b1] in <6aa716917c4045a18ded6a6892a77cbf>:0 
  at System.Threading.Tasks.Task.InnerInvoke () [0x0000f] in <9577ac7a62ef43179789031239ba8798>:0 
  at System.Threading.Tasks.Task.Execute () [0x00010] in <9577ac7a62ef43179789031239ba8798>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <9577ac7a62ef43179789031239ba8798>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0003e] in <9577ac7a62ef43179789031239ba8798>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x00028] in <9577ac7a62ef43179789031239ba8798>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x00008] in <9577ac7a62ef43179789031239ba8798>:0 
  at System.Runtime.CompilerServices.ConfiguredTaskAwaitable+ConfiguredTaskAwaiter.GetResult () [0x00000] in <9577ac7a62ef43179789031239ba8798>:0 
  at Hid.Net.HidDevice+<InitializeAsync>d__18.MoveNext () [0x00116] in <6aa716917c4045a18ded6a6892a77cbf>:0 
--- End of stack trace from previous location where exception was thrown ---
  at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw () [0x0000c] in <9577ac7a62ef43179789031239ba8798>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Threading.Tasks.Task task) [0x0003e] in <9577ac7a62ef43179789031239ba8798>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Threading.Tasks.Task task) [0x00028] in <9577ac7a62ef43179789031239ba8798>:0 
  at System.Runtime.CompilerServices.TaskAwaiter.ValidateEnd (System.Threading.Tasks.Task task) [0x00008] in <9577ac7a62ef43179789031239ba8798>:0 
  at System.Runtime.CompilerServices.ConfiguredTaskAwaitable+ConfiguredTaskAwaiter.GetResult () [0x00000] in <9577ac7a62ef43179789031239ba8798>:0 
  at MothTest+<GetAndShowDevices>d__3.MoveNext () [0x003cb]

HotYearKit, 2021-02-10:
[image]

MelbourneDeveloper, 2021-02-10:
I need more context

HotYearKit, 2021-02-10:
public static async void GetAndShowDevices(Action<string> DebugText)
    {
        //Create logger factory that will pick up all logs and output them in the debug output window
        var loggerFactory = LoggerFactory.Create((builder) =>
                            {
                                _ = builder.SetMinimumLevel(LogLevel.Trace);
                            });

        //----------------------

        // This is Windows specific code. You can replace this with your platform of choice or put this part in the composition root of your app

        //Register the factory for creating Hid devices. 
        var hidFactory = new FilterDeviceDefinition(vendorId: 0x0483, productId: 0x5750).CreateWindowsHidDeviceFactory(loggerFactory);

        //Register the factory for creating Usb devices.
        var usbFactory = new FilterDeviceDefinition(vendorId: 0x0483, productId: 0x5750).CreateWindowsUsbDeviceFactory(loggerFactory);

        //----------------------

        //Join the factories together so that it picks up either the Hid or USB device
        var factories = hidFactory.Aggregate(usbFactory);

        //Get connected device definitions
        var deviceDefinitions = (await factories.GetConnectedDeviceDefinitionsAsync().ConfigureAwait(false)).ToList();

        if (deviceDefinitions.Count == 0)
        {
            //No devices were found
            DebugText?.Invoke("No devices were found");
            Debug.Log("No devices were found");
            return;
        }
        foreach (var item in deviceDefinitions)
        {
            DebugText?.Invoke($"VID:{item.VendorId},PID:{item.ProductId},DeviceID:{item.DeviceId}");
            Debug.Log($"VID:{item.VendorId},PID:{item.ProductId},DeviceID:{item.DeviceId}");
        }

        //Get the device from its definition
        var trezorDevice = await hidFactory.GetDeviceAsync(deviceDefinitions.First()).ConfigureAwait(false);

        //Initialize the device
        try
        {
            await trezorDevice.InitializeAsync().ConfigureAwait(false);
        }
        catch (System.Exception e)
        {
            DebugText?.Invoke(e.Message + ";" + e.Source + ";" + e.StackTrace);
            Debug.Log(e.Message + ";" + e.Source + ";" + e.StackTrace);
            return;
        }
}

MelbourneDeveloper, 2021-02-11:
This is the last chance. I will delete this problem if I don't get a reasonable response.

What is the question?